dresden-elektronik/deconz-rest-plugin #1478: Keen Vent Not Found by Phoscon as light

tenholde, 2019-04-28:
[image]
[image]

I am trying to learn the ConBee/deCONZ product, with mixed success.  I am linux illiterate, running WIN10, and most posts refer to solving problems in linux?  I have been able to connect to a vent, and have  at times had phoscon find and control it.  In attempting to add other vents, everything stopped working so I reset everything a started over.  My repeater (Keen Hub) and vent were found right away, and I can control the vent from deCONZ with no problem.  Phoscon will NOT detect it as a light (while it did before I reset everything, intermittently).  I am using a plugin running under HomeSeer3 that connects to deCONZ, and it has found the pressure and temperature sensors and is updating the temperature sensor properly, but it does not find the vent as a Light (phoscon did not either), so I cannot control the light.  I've tried resetting and rescanning multiple times with the same result.

Also, I'd like update to newest 2.05.64 release, but do not know how to do such under windows.  If I accomplished that, I would then try to update firmware to 26330500.

Currently:   Original Conbee
                    2.05.63
                   26320500

Any help would be appreciated.

tenholde, 2019-04-28:
successfully updated to 2.05.64 and 26330500 and still phoscon will not find the vent

tenholde, 2019-04-28:
After another 4 tries at scanning and resetting, all of a sudden the vent appears correctly in phoscon.

LQIs all look good.  Not sure I understand what is going on.  I will move forward with adding more vents and see what happens.

wvuyk, 2019-04-29:
I don’t understand either. In the file you send me there was indeed no clue for the on/off functions in the REST API. Somehow – also in github – getting the REST API complete by scanning is having difficulties. I also saw your post on github, let’s hope they can recheck this, so it stabilizes.

 

Thanks for letting me know!

 

Wim

closed 2019-04-29 by tenholde