dresden-elektronik/deconz-rest-plugin #1479: IKEA CWS bulbs don't respond to ct group commands

olemr, 2019-04-28:
[image]

In PWA, it is not possible to enter the Color Temperature menu. Tried both Single and Group.
(This group has 4x CWS bulbs only)

From REST API, sending a ct command to the group returns success, but does nothing.
(This group has a mix of CT and CWS bulbs)

Sending REST ct command to a single bulb works OK.

A known issue?

olemr, 2019-05-01:
Update: this is probably just an PWA issue.
CT control works fine from OWLC.

sfalkman, 2019-05-01:
Do you mean that you can't see the Temperature slider i PWA? Because I can't either.
However, if I remove the `sl_hide` class from the `<div id="sl-ct" ...>` div, the slider is shown and is working as expected.

Also, I don't get any temperature slider in Home Assistant for this light either.

The light I'm using is TRÅDFRI bulb E27 CWS opal 600lm with version 1.3.002

manup, 2019-05-01:
The Problem is that the IKEA CWS lights don't support Zigbee color temperature commands.

When sending a `ct` command to a single CWS light a little conversion is made to transform `ct` into `xy` color space which the light understands.

When using the group controls the conversion isn't applied and that's the reason the light just ignores the command. The reasoning behind this is when a group contains plain color temperature lights they would ignore the xy color command.

olemr, 2019-05-01:
> The Problem is that the IKEA CWS lights don't support Zigbee color temperature commands.
But mine do. After the RGB party is over, I can send a ct: command over REST in order to set the exact same CT as my CT bulbs.

> Do you mean that you can't see the Temperature slider i PWA?
For groups, I can see the slider, but only CT bulbs are responding.
For a single CWS bulb, the slider dous not appear at all in PWA, but in OWLC color map it works fine.
Trying OWLC on a mixed CT/CWS group makes it sort of hang, as long as I'm in that menu.

olemr, 2019-05-01:
I even tried making a test group of with just 4 CWS bulbs in it.
Behaviour is the same as for a mixed group. In PWA the 
[image]
button does not respond at all (as for single CWS bulbs)
In OWLC the group doesn't respond to either brightness or color temperature commands.

olemr, 2019-05-01:
> The reasoning behind this is when a group contains plain color temperature lights they would ignore the xy color command.

Could you send both?
CT would ignore xy
CWS would ignore ct
All would be dandy?

ebaauw, 2019-05-02:
And extended colour lights, supporting both, would react to both, leading to undeterministic behaviour.

Personally, I think the decision to support `ct` for a colour light is wrong.  The light doesn’t support colour temperature - that should be clear to the API consumer.

In previous API v2 discussions, I suggested to have a two-layer setup: a “raw” layer where you interact very closely to the ZigBee communication, and a “cooked” layer for applications.  In the “cooked” layer, you would expose brightness from 0 to 100%, whereas the “raw” layer exposes current level from 0 to 254.  Similarly, the “raw” layer exposes CIE 1939 `xy`, where the “cooked” layer exposes RGB hue and saturation (or r, g, and b).  Any translation from colour temperature (preferably in K instead of mired) to `xy` should happen in the “cooked” layer.

olemr, 2019-05-02:
> And extended colour lights, supporting both, would react to both, leading to undeterministic behaviour.

There are of course other bulbs than Ikea out there and you guys see the whole picture.
There is still an issue in PWA though, not showing Color temperature for a single CWS light.

closed 2019-09-06 by stale[bot]