dresden-elektronik/deconz-rest-plugin #1480: Meaning of LQI numbers on link lines

tenholde, 2019-04-28:
When LQI button is selected, LQI numbers appear next to most lines on the chart.

I think I understand what the LQI is telling me, but there are two numbers separated by a slash.

Some are something like:   252/245   and some are line   254/0

A few have no numbers next to the line.

What does the second number vs the first number mean, and should I be concerned if there is no number next to one or more lines?

manup, 2019-04-29:
Thee yellow router nodes are polled periodically for their neighbor tables. The notion of 252/245 between two nodes represents how they see each other. A value of 0 is shown if either a node is a end-device, like a switch or sensor, or if the node wasn't queried yet or doesn't have a neighbor entry for the other node in its table.

>What does the second number vs the first number mean, and should I be concerned if there is no number next to one or more lines?

No need to worry the lines and LQI values are just a view which might not be 100% accurate or complete in some cases.

closed 2019-05-05 by tenholde

melslenstra, 2019-12-26:
@manup thanks for this explanation, I was also wondering about this. How do I tell which (left or right) value is the LQI reported by which of the two nodes in each link, for links between router(s) and coordinator?

Might I suggest to add an explanation of this to the deCONZ user manual? I apologize if this isn't the correct place for such a suggestion.