dresden-elektronik/deconz-rest-plugin #1484: KEEN Vent Battery Reporting

tenholde, 2019-04-30:
The value for battery percentage appears to be reported incorrectly in the REST information.

Comparing the Power Configuration value to REST:

100% appears as 50% in REST
75% appears as 37% in REST
50% appears as 25% in REST

See attached for example of 75%/37%

Let me know if I can supply anything additional.
[image]

[image]

[image]

tenholde, 2019-04-30:
[image]

version information

ebaauw, 2019-04-30:
The ZCL spec states that _Battery Percentage Remaining_ is in 0.5%.  Did you check what value theKeen Vent reports for a fresh battery?

tenholde, 2019-04-30:
I put in new batteries, and the Power Configuration Cluster reads as 100.  The REST info contains 50.

ebaauw, 2019-04-30:
Apparently Keen Home don't adhere to the standard, just like IKEA and iCasa.  I'll add them to the appropriate exception list.

tenholde, 2019-05-01:
Thank you.  How does this get accomplished?  Will this be included in the next release of deCONZ?

Ed Tenholder
ed@tenholder.net

closed 2019-09-05 by stale[bot]