dresden-elektronik/deconz-rest-plugin #1485: Losing connection with OSRAM Smart+ motion sensor

masonevans, 2019-04-30:
I have an OSRAM Smart+ motion sensor which I've been experiencing a lot of issues with.  It seems to work for awhile and then suddenly I stop getting updates from it.  This happens even without anything else on the network changing or losing power.  Usually if I take the battery out of the sensor and put it back in it reconnects and all is well but obviously this is not a long-term solution for a motion sensor.

A few things I have noticed (0x000D6F000E05952D is the motion sensor)

- When it's working, I see entries in my deCONZ log that look like this

```
08:23:22:919 no button map for: Motion Sensor-A ep: 0x01 cl: 0x0500 cmd: 0x0A pl[0]: 002  
08:23:22:919 ZCL attribute report 0x000D6F000E05952D for cluster 0x0500, ep 0x01
```

- When it's not working, I see lots entries like this

`08:25:19:228 APS-DATA.confirm id: 190 status: transaction expired (8 seconds)`

Here is the whole log from the last entry where it was working to the first entry after it stopped

```
08:23:22:919 no button map for: Motion Sensor-A ep: 0x01 cl: 0x0500 cmd: 0x0A pl[0]: 002
08:23:22:919 ZCL attribute report 0x000D6F000E05952D for cluster 0x0500, ep 0x01
08:24:01:976 Current channel 15
08:24:02:007 Device TTL 6720 s flags: 0x7
08:24:24:053 Beacon src: 0xF7FF ch: 15 updateId: 0
08:24:26:420 reuse dead link (dead link container size now 6)
08:24:30:912 binding for cluster 0x0000 of 0xD0CF5EFFFEDA8910 exists (verified by reporting)
08:24:30:913 binding for cluster 0x0006 of 0xD0CF5EFFFEDA8910 exists (verified by reporting)
08:24:30:913 binding for cluster 0x0008 of 0xD0CF5EFFFEDA8910 exists (verified by reporting)
08:24:31:080 ZCL configure reporting rsp seq: 44 0xD0CF5EFFFEDA8910 for cluster 0x0006 attr 0x0000 status 0x00
08:24:31:136 ZCL configure reporting rsp seq: 45 0xD0CF5EFFFEDA8910 for cluster 0x0008 attr 0x0000 status 0x00
08:24:31:312 verified group capacity: 255 and group count: 0 of LightNode 0xd0cf5efffeda8910
08:24:33:023 delay sending request 244 dt 0 ms to 0xD0CF5EFFFEDA8910, cluster 0x0006
08:24:33:112 delay sending request 244 dt 0 ms to 0xD0CF5EFFFEDA8910, cluster 0x0006
08:24:33:152 	0xD0CF5EFFFEDA8910 force poll (2)
08:24:33:287 	0xD0CF5EFFFEDA8910 force poll (2)
08:25:01:976 Current channel 15
08:25:02:007 Device TTL 6660 s flags: 0x7
08:25:06:945 GW firmware version: 0x26330500
08:25:06:945 GW firmware version is up to date: 0x26330500
08:25:19:228 APS-DATA.confirm id: 190 status: transaction expired (8 seconds)
```

Currently the end node is showing as connected to both deCONZ and another Zigbee router (the other nodes are expected to be offline)

[image]

Running latest deCONZ and Firmware, running in Docker

[image]

I'm not sure what else to do to debug this issue.  Sometimes the sensor will connect to just the control outlet that it's next to but then it reconnected itself to the gateway, at which point it worked for awhile and then randomly stopped working.  I know that it's detecting motion because I see the light flash but no commands actually make it to the gateway.

Philje123, 2019-04-30:
I too am experiencing this issue.

But strangely I have 2x Osram Motion sensors. The one in my kitchen has never done this.

But the one in my hallway randomly looses it's connection.

masonevans, 2019-04-30:
And now without me doing anything (no devices were power cycled nor was deCONZ restarted) the sensor is sending data again

```
12:54:01:249 APS-DATA.confirm id: 64 status: transaction expired (8 seconds)
12:54:01:978 Current channel 15
12:54:02:010 Device TTL 3841 s flags: 0x7
12:54:35:869 no button map for: Motion Sensor-A ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
12:54:35:869 ZCL attribute report 0x000D6F000E05952D for cluster 0x0402, ep 0x01
12:54:35:870 discard double entry in binding queue (size: 2) for for 0x000D6F000E05952D, cluster 0x0001
12:54:35:972 no button map for: Motion Sensor-A ep: 0x01 cl: 0x0500 cmd: 0x0A pl[0]: 002
12:54:35:972 ZCL attribute report 0x000D6F000E05952D for cluster 0x0500, ep 0x01
12:54:36:106 delay sending request 45 dt 0 ms to 0xD0CF5EFFFEDA8910, cluster 0x0006
12:54:36:111 delay sending request 45 dt 0 ms to 0xD0CF5EFFFEDA8910, cluster 0x0006
12:54:36:147 	0xD0CF5EFFFEDA8910 force poll (2)
12:54:36:205 ZCL attribute report 0xD0CF5EFFFEDA8910 for cluster 0x0006, ep 0x01
12:54:36:283 	0xD0CF5EFFFEDA8910 force poll (2)
12:54:36:371 APS-DATA.confirm id: 249 status: transaction expired (8 seconds)
12:54:36:412 Set sensor check interval to 100 milliseconds
```

And I see updates in Phoscon & Home Assistant again

SeraphimSerapis, 2019-05-02:
I've had a number of issues with these. Went with Hue Motion Sensors and they're rock solid.

Philje123, 2019-05-02:
I have 2x IKEA motion sensors a a Hue and they're rock solid.

I got the Osram Sensors as they were cheap (paid €10 each) but they're not as reliable and have odd batteries.

Ah well. It's been fine since I reset, deleted and re-added it the other day.

closed 2019-09-06 by stale[bot]

Philje123, 2019-09-27:
I am still having these issues. Pretty much everyday the motion sensors will stop working.

I usually have to reboot to get them working again. Pulling the battery out doesn't work.

I notice in Phoscon the software version is reported as:

����������������

I tried pairing one to my Lightify bridge to see if there was a software update for them but they're already up to date.