dresden-elektronik/deconz-rest-plugin #1487: Aqara Wall switch QBKG04LM - Wrong section

lenisko, 2019-04-30:
Hey there!

Can this device get somehow moved to `Switch` section?

It's Xiaomi `QBKG04LM` Aqara Wall Switch / one key version / can act as router.

I've managed to pair it using `Sector` section, after it it has landed in `Light` section and should be in `Switch`. 
Funny thing btw :-)

There are also other wall switches from Xiaomi, I can provide details later if needed.

[image]

ebaauw, 2019-04-30:
No, it shouldn’t.  Stateful wired wall switches and dimmers, used to control traditional lights, behave like smart lights, and are exposed as such.  Same for plugs.  Stateless wireless switches, used to control smart lights (or wired switches or plugs), are exposed as switches.

lenisko, 2019-04-30:
In this case, it might be wise to add some note about this particular case in "Add new" menu.

I wasn't able to connect to this Switch from "Switches" section, it was confusing as hell.

Staying at this topic, this device is represented with 2 devices marked as light, where one of them is unused? Also both of them are marked as `Unknown`.

[image]

ebaauw, 2019-04-30:
What version of deCONZ are you on?  Could you read the attributes of the  _Basic_ cluster on endpoint 01 in the GUI and post a screenshot?

lenisko, 2019-04-30:
```
Product RaspBee
Version 2.05.63 / 4/3/2019
Firmware 26330500
```

Sure, here's screen

[image]

ebaauw, 2019-05-01:
Did you pair the switch recently (i.e. using this version of deCONZ)?  Does the API show any `/sensors` resources for the switch?  There should be a ZHASwitch for endpoint 04 (to control smart lights that aren’t wired to the switch), but I don’t think Phoscon yet supports that.

lenisko, 2019-05-01:
Yes it was paired yesterday.

```
  "2": {
    "config": {
      "on": true,
      "reachable": true,
      "temperature": 2600
    },
    "ep": 4,
    "etag": "565d666746c1af23aba5b74b02a6d071",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.ctrl_neutral1",
    "name": "lumi.ctrl_neutral1",
    "state": {
      "buttonevent": 1002,
      "lastupdated": "2019-04-30T23:03:06"
    },
    "swversion": "04-25-2018",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:02:9c:2c:e5-04-0006"
  },
```

`lastupdated` is changing, but event is staying the same for `single, double, hold`.

ebaauw, 2019-05-01:
Checking the code:
- We did solve the copying of the `manufacturername` and `modelid` from the _Basic_ cluster on endpoint 01 to the other endpoints, but that is whitelisted for manufacturer code 0x115F.  This switch uses 0x1037 instead.
- The `type` of `"Smart plug"` for endpoint 02 is probably because the simple descriptor hadn't been read in time and was defaulted to 0xffff.
- The one-button `lumi.ctrl_neutral1` has the same ZigBee fingerprint as the two-button `lumi.ctrl_neutral2`.  We need to check the `modelid` and suppress the creation of the second light resource for endpoint 03 for the one-button variant.

> lastupdated is changing, but event is staying the same for single, double, hold

The switch only reports press/release (1002).

am1ter, 2019-09-01:
Hello!

Could you please help me with disconnecting button from relay for this switch (Aqara QBKG04LM, 1 line, no neutral)? 

In other thread realted to 2 line switch Aqara QBKG12LM https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1420#issuecomment-483057308 @ebaauw clarified that Aqara switches can work in decoupled mode and all you need to do is write new values (`0xFE`) in attributes `0xFF22` and `0xFF23`. Unfortunatelly, deconz don't show this parameters for my switch QBKG04LM with default general.xml, despite in this files there is relevant lines.

From this topic https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1487#issuecomment-488347147 I discovered that my switch QBKG04LM uses another manufacturing code - `0x1037` instead of `0x115F`. When I changed `mfcode` to `0x137` in the file general.xml to correct value I discovered that disconnecting parameters appeared in deconz interface with a `0x00` value. But when I try to read or write a new value, for example `0xFE`, I get an error:

> failed UNSUP_MANUF_GENERAL_COMMAND

I read as many topics realted to my case as I could, but didn't find anything what could help me. Moreover I use Conbee II so I can't use ZSHARK for sniffing my xiaomi zigbee gateway and can't look out a command that disconnect switch button from relay.

I would really appreciate if anyone can give me advice how I can try to fix my problem.

VivantSenior, 2019-11-08:
> It's Xiaomi `QBKG04LM` Aqara Wall Switch / one key version / can act as router.

The Xiaomi wired switches without neutral can't work as routers / repeaters.

closed 2020-03-14 by stale[bot]