dresden-elektronik/deconz-rest-plugin #1488: Aqara sensors are disconnected after some time when connected through router

imortkz, 2019-05-01:
Aqara sensor connected to the router disconnects and became unavailable after some time. Other sensors, connected to the coordinator, are stable.

Router: Sylvania SMART+ Smart Plug (72922-A)
Sensors: Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)

When connected to the router keeps online for hour or so, then didn't send any update.

Connect sensor to the router and wait for a while.
Aqara sends 5 updates in 41 minutes after rejoin and connected to the router, and no more updates after.

ConBee is upgraded to the last firmware.
Should I expect any messages in the logs while debug is on?

manup, 2019-05-01:
>Should I expect any messages in the logs while debug is on?

I'm afraid the logs won't show the required details. Since the direct connection with coordinator is stable I assume the plug does one of the following:
* Throw the sensor out of the child table and doesn't forward the messages to coordinator
* Forces the sensor to leave the network

Only sniffer logs can tell what happens exactly.

Related issues:
https://github.com/Koenkk/zigbee2mqtt/issues/1474
https://github.com/Koenkk/zigbee2mqtt/issues/1485

imortkz, 2019-05-01:
Yeah, I tried zigbee2mqtt to find out that I have the same problem as with ConBee stick.
Will try to get another router to check if it's happened because of smartplug.

Thank you

pergolafabio, 2019-05-02:
i had the same exact issue with my xiaomi pir sensors, now as test i removed all osram smart plugs, to the xiaomi are paired directly with stick... seems more stable now

imortkz, 2019-05-02:
I need to have a routers present in the network, so I'll try to build and add a CC2530-based router next week and test it.

sragas, 2019-05-10:
Do have exact the same issue. Using two IKEA Tradfri repeater as router and since a few days all paired  xiaomi end devices (door and multi sensors) are not reachable any more. The ones paired with conbee stick are still functional. Could also not managed to rejoin them till now.

aodro, 2019-05-13:
The same issue here. CC2531, CC2530+CC2591 and an ikea bulb as routers. Xiaomi devices do not work, while a ikea remote does. Maybe is only a problem with xioami devices (they do not want to use routers)?

pergolafabio, 2019-05-13:
well, i have only 4 x xiaomi pir's paired with Conee, no routers at this moment
today 1  xiaomi lost connection too , after about 3 weeks
dont know why , but i need to pair it again, verry frustating

all connections are good, green lines

sragas, 2019-05-13:
Also got only issues with Xiaomi end devices, all other work properly (till now).
The Xiamoi end devices paired with conbee are still working at this point (about 2 weeks now) but the other ones are unreachable after 2-3 days from pairing.

pergolafabio, 2019-05-13:
yeah , same experience :(

imortkz, 2019-05-14:
> The same issue here. CC2531, CC2530+CC2591 and an ikea bulb as routers. Xiaomi devices do not work, while a ikea remote does. Maybe is only a problem with xioami devices (they do not want to use routers)?

Have CC2530-based router and looks like Xiaomi sensor is stable connected for the day, and survive a 12h coordinator downtime (connected to the router).

Will see how it goes for the next couple of days.

sragas, 2019-05-14:
@imortkz do you working with the reworked version from manup or an official release?

imortkz, 2019-05-14:
Official Docker image

sragas, 2019-05-15:
@manup I build the plugin from the actual master branch source including your commit to test. Unfortunately the plugin didn't get loaded if I use my self compiled version. Build on Ubuntu and used in x86 Docker Image. Are there any plans for a new official release in short term, or do you have any hints to get the self build one running?

P1Rebo, 2019-05-25:
Any updates? Experiencing the same issues using Osram Smart Plug als repeaters. Updated ConBee (1) to the latest Firmware, using latest Deconz release. These damn Xiaomi Sensors... cheap but so much effort to get / keep them running...

sragas, 2019-05-26:
I didn‘t manage to get the self compiled plugin to run, so as soon as v 2.05.65 is released i will give it a try. And as you said, cheap but sometimes pain in the...

imortkz, 2019-05-29:
I've a working setup with zigbee2mqtt and two kinds of router: DIY cc2530 routers, SmartThings Outlet (IM6001-OTP05). Sensors are steady connected to the router, no issues.

Didn't tried them with deCONZ stick, but I have a strong feeling that it will be the same.

I has exactly the same constant disconnects with Osram both for deCONZ and zigbee2mqtt.

hgelpke, 2019-05-30:
Hopefully it works with Deconz. I'm having the exact same issue and I think it might be the Tradfri repeater. I've taken it out for now and can can add a cc2530 if that helps.

sragas, 2019-05-30:
Just updated deCONZ yesterday to v2.05.65, which may fix the issue, and repaired lost Xiaomi Aqara sensors. Till now (14h) all sensors are up and running. All paired with conbee stick or one of two tradfri repeater. Keep you posted

sragas, 2019-06-06:
Still all sensors up and running. New release seems to fix the issue on my site.

closed 2019-10-11 by stale[bot]

hawkefly, 2020-04-12:
It seems that 20200312 firmware in dev branch fix this issue

cagnulein, 2020-12-15:
i'm still having this issue with 
version: 2.05.84 / 14/9/2020
firmware: 26580700

maybe a regression? do you have still problems?
thanks

thueske, 2020-12-15:
I have this issue, too.

My temperature sensor works fine for several weeks, but since some days it continuously disappears / disconnects.

[image]

(Last time where repairing worked there were a red line before...)

Version: 2.07.01 / 8.12.2020
Firmware: 26660700

Any ideas @sragas?

cagnulein, 2020-12-16:
@thueske i opened a new one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3929

sragas, 2020-12-17:
@thueske Moved my place and setup all devices from scratch. Only have one device (temp sensor) which is disappearing continuously, but this may be a hardware fault. No idea how to solve this and it will be replaced soon. But my old setup runs for months without any problems.

IgnacioHR, 2021-05-06:
I've the same or similar issue with a Xiaomi temperature sensor.
See the picture all other devices works!
In my net, there are two OSRAM bubles close to the Xiaomi sensor but the sensor it is unable to send data anymore.
If I take the Xiaomi temperature and humidity sensor closer to the RaspBee and start the sensor discovery in the gateway the sensor works (connected to the RaspBee) but once I move it to its location it disappears again. It is not discovered in its location. It was operative for about 3 months and connected to the RaspBee.
It might be:
- The battery went down a bit after 3 months of operation and now, it has no power to reach the RaspBee anymore?
- The Xiaomi sensors are unable to use a router and require direct connection to the RaspBee?
- Maybe this is NOT a RaspBee problem but a Xiaomi problem. 

[image]