dresden-elektronik/deconz-rest-plugin #1489: No Reaction for Busch & Jäger switch

lepatrick, 2019-05-01:
Hey guys,

I have connected Busch Jaeger 6736-84 using Phoscon App. After connection it works fine. Light can be switched on and of using phoscon. After A while it's impossible. Phoscon GUI switches state but nothing happens at the real light. I did reconnect the switch, replaced it, after some hours it stops working. But deCONZ shows connection.

Any ideas how I can solve this problem?

I'm using deCONZ with ConBee on Ubuntu System:

Version 2.05.64 / 22.4.2019
Firmware 26330500

esseki, 2019-05-16:
Hi,

don't want to hijack this issue, so i'm not sure if i should report my problem in this same ticket as i'm facing the exact same kind of problem but with different switches even if these are still battery powered but from a different brand : [Jung module 4-gang](https://www.jung.de/en/online-catalogue/374871630/)

I indeed replaced all the Xiaomi Mija round cheap switches i had in my house for those expensive Jung switches that i am happy about. Except from the fact that as @lepatrick reported it, after a while they kind of loose the route to the Conbee and the Conbee does not seem to receive commands anymore preventing associated lights from being controlled. When this happens (it can be once a day) i have to repair the switch in "Phoscon / Switches / Add a new switch". Hopefully by doing so the id associated to the corresponding sensor in the REST API remains the same, allowing all the rules i have created manually to function, even after a new pairing. But this is still very annoying to have to repair those switches constantly.

I have tested several releases and firmware updates with no luck from V2_05_58. 

Just would like to mention that as @manup adviced me in another thread, i paid attention not to having any Ikea lights powered when adding the Jung switch to the network. 

My network is about 17 Philips Hue bulbs covering the house + 8 Ikea Tradfri bulb + a dozen of Xiaomi Aqara sensors (temperature and door mainly). And i have to say that i have no problem with any other devices in my network. Not even with any of the Xiaomi battery powered once. The problem is happening only with the Jung switches.

Is there any chance my problem is the same as the one reported by @lepatrick ? As it seems to be very similar.
In this case is it worse i share any log/info to help debugging (just let me know what would be helpful and i will provide).

Thank you in advance for any help.

lepatrick, 2019-05-17:
Hey @esseki,

since 10 days I haven‘t any problem. What I did? I remove all Busch-Jaeger switches from deCONZ and also removed all TRÅDFRI Lights, because this were devices with problems. All Xiaomi I let stay in the network. I removed conbee stick for a while (in my head it was kind of clearing cache) and updated all TRÅDFRIs to newest firmware using the ikea gateway. After adding Conbee back to deCONZ, I first connect all wall switches using phoscon as lights and all battery wall switches as switch. Next day I add TRÅDFRIs back to network. Test some restarts and it’s working fine right now. I don‘t know exact what helps, but maybe it helps you. :)

esseki, 2019-05-17:
That's an excellent news. Thank you @lepatrick for sharing. Will test and report.

esseki, 2019-05-18:
Update as of tonight. I did the following :  

- [X] remove all ~~Busch-Jaeger~~ JUNG switches from deCONZ 
- [X] and also removed all TRÅDFRI Lights, because this were devices with problems. 
- [X] All Xiaomi I let stay in the network. 
- [X] I removed conbee stick for a while (in my head it was kind of clearing cache) 
- [X] and updated all TRÅDFRIs to newest firmware using the ikea gateway. 
- [X] After adding Conbee back to deCONZ, I first connect ~~all wall switches using phoscon as lights and~~ all battery wall switches as switch. 

Will test a few days and report back before continuing with the following : 

- [ ] Next day I add TRÅDFRIs back to network. 
- [ ] Test some restarts and it’s working fine right now.

esseki, 2019-08-04:
So it took me a while before i feel like adding the TRÅDFRIs lights back to the network. And during those weeks with no TRÅDFRIs lights, the Jung switches have been working perfectly.
Problem is since i added the TRÅDFRIs lights back to the network, (even after following carefully @lepatrick procedure ;)

There is definitively something going wrong between Jung switches and Ikea TRÅDFRIs lights.

closed 2019-12-09 by stale[bot]

alex030, 2020-01-28:
@esseki: did you ever solved the problem with the Jung switches? I only had one single Ikea light. So I removed it. But I still see this Jung switches loosing connectivity. Do I really need to reset the hole system and first add the switches and then my non Ikea lights? Could you solve this strange behaviour?

esseki, 2020-01-31:
Hi @alex030 ,

i finally decided to get rid of all my Ikea bulbs and to replace them by Philips Hue only.
I could not find a solution to make the Jung switch work correctly as soon as there is a single Ikea bulb in the Zigbee network.

You don't need to reset the whole system. Following my post https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1489#issuecomment-493704562 you just need to Jung switches and the Ikea bulbs but the rest of the network should not be changed. This was sufficient for me to get the network stable after adding again the switches.

Regards