dresden-elektronik/deconz-rest-plugin #1490: Conbee II with Mijia temperature/humidity sensor

alex3305, 2019-05-02:
I've just received my Conbee II and it is working great with both Phoscon and Home Assistant. The only thing is I cannot seem to connect to the old style Xiaomi Mijia temperature / humidity sensor. Doesn't matter how long I press the button, it will just not connect. Things I've tried:

 * Press the button shortly
 * Press the button for 3 - 4 seconds to reset
 * Press the button for 10+ seconds for a complete reset

None of these appear to work. My newer Xiaomi Aqara temperature/humidity sensors work fine. Is there anything else I can test or perhaps a bug in the Conbee II implementation?

edwork, 2019-05-07:
I have the same issue right now on .64, I think it actually started with .63. I bought 4 more Aqara PIR sensors and couldn't pair them. Thinking my network was too congested I purchased 2 Tradfri repeaters and was unable to pair them too. 

The one major change in made was switching from installing .deb files to the official Apt repo. Maybe permissions changed making certain files unwritable? Thinking about it, the last time I tried to backup my configuration it failed to generate the file. 

Seeing any similarities?

manup, 2019-05-07:
Are you running latest ConBee II firmware version 0x26490700?

It's shown in the settings:

 *Phoscon App &rarr; Menu &rarr; Settings &rarr; Gateway*

edwork, 2019-05-08:
I'm iffy now on the relevance of my issue now, but I resolved my problem by going to the "Zigbee Configuration" page (hold down Alt while pressing Advanced on the Gateway page of PWA, and loading the most recent profile available. 

I was instantly able to pair one of my troublesome sensors. The backup option also started working again.

alex3305, 2019-05-10:
> Are you running latest ConBee II firmware version 0x26490700?
> 
> It's shown in the settings:
> 
> _Phoscon App → Menu → Settings → Gateway_

Yes I am:

**Vendor** dresden elektronik
**Product** ConBee II
**Version** 2.05.63 / 3-4-2019
**Firmware** 26490700

alex3305, 2019-05-26:
This issue still persists with the latest (firmware) update. Anything else I can try or do to fix this?

cc @manup

manup, 2019-05-27:
The version is fine, one other important factor can be the signal. We have seen some cases where devices where hard to pair due to interference. Do you use an USB extension cable? It can mitigate such issues. Further please make sure that no other 2.4 GHz radios like WiFi or Bluetooth are too close to ConBee II.

alex3305, 2019-05-27:
@manup Yes, I even tried multiple extension cables to exclude possible cable issues. When pairing I've held the Conbee at about 50cm. Maybe that's not close enough still?

manup, 2019-05-27:
50cm is fine, usually it should also work if the sensor is 10 meters away.
Can you please try with version 2.05.64 or if possible 2.05.65.

It would also be helpful to have the log while the sensor is to be joined, maybe it yields some message on whats going on.

    deCONZ --http-port=80 --dbg-info=1 --dbg-zcl=1 --dbg-aps=1

alex3305, 2019-05-27:
After about 5 tries of three minutes one of the two sensors just connected out of the blue. I didn't change anything and the distance to pair was about 3 meters in open space.

The second sensor was way more stubborn. I gave up after about 30 minutes of trying to pair the device. Than I tried pairing through the old webinterface - which I've never used. Which worked the first time around...

There was nothing obvious for me in the log, but for completion I've attached some logging for @manup .

First sensor
```
12:20:34:155 send permit join, duration: 59
12:20:37:042 no button map for: lumi.sensor_motion ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
12:20:37:042 ZCL attribute report 0x00158D0001539DAC for cluster 0x0406, ep 0x01
12:20:38:299 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
12:20:38:299 ZCL attribute report 0x00158D0001E525BB for cluster 0x0400, ep 0x01
12:20:38:309 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
12:20:38:309 ZCL attribute report 0x00158D0001E525BB for cluster 0x0406, ep 0x01
12:20:39:212 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
12:20:39:212 ZCL attribute report 0x00158D0001E54496 for cluster 0x0400, ep 0x01
12:20:39:229 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
12:20:39:229 ZCL attribute report 0x00158D0001E54496 for cluster 0x0406, ep 0x01
12:20:39:403 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
12:20:39:403 ZCL attribute report 0x00158D0001E55748 for cluster 0x0400, ep 0x01
12:20:39:413 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
12:20:39:413 ZCL attribute report 0x00158D0001E55748 for cluster 0x0406, ep 0x01
12:21:11:372 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
12:21:11:372 ZCL attribute report 0x00158D0001E524D5 for cluster 0x0400, ep 0x01
12:21:11:383 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
12:21:11:383 ZCL attribute report 0x00158D0001E524D5 for cluster 0x0406, ep 0x01
12:21:14:041 APS-DATA.indication from unknown node 0x00158D0001873510
12:21:14:073 new node - ext: 0x00158d0001873510, nwk: 0x092D
12:21:14:090 device announce 0x00158D0001873510 (0x092D) mac capabilities 0x80
12:21:14:090 set fast probe address to 0x00158D0001873510 (0x092D)
12:21:14:090 FP indication 0x0000 / 0x0013 (0x00158D0001873510 / 0x092D)
12:21:14:090                       ...     (0x00158D0001873510 / 0x092D)
12:21:14:090 device announce 0x00158D0001873510 (0x092D) mac capabilities 0x80
12:21:14:101 FP indication 0x0104 / 0x0000 (0x00158D0001873510 / 0x092D)
12:21:14:101                       ...     (0x00158D0001873510 / 0x092D)
12:21:14:101 ZCL attribute report 0x00158D0001873510 for cluster 0x0000, ep 0x01
12:21:14:877 FP indication 0x0104 / 0x0000 (0x00158D0001873510 / 0x092D)
12:21:14:877                       ...     (0x00158D0001873510 / 0x092D)
12:21:14:877 ZCL attribute report 0x00158D0001873510 for cluster 0x0000, ep 0x01
12:21:14:955 [1] get node descriptor for 0x00158d0001873510
12:21:15:384 CTRL skip polling while permit join is set
12:21:15:589 CTRL skip polling while permit join is set
12:21:15:634 don't close database yet, keep open for 900 seconds
12:21:15:634 FP indication 0x0000 / 0x8002 (0x00158D0001873510 / 0x092D)
12:21:15:634                       ...     (0x00158D0001873510 / 0x092D)
12:21:15:634 ZDP indication search sensors 0x00158D0001873510 (0x092D) cluster 0x8002
12:21:15:634 ZDP indication search sensors 0x00158D0001873510 (0x092D) clear timeout on cluster 0x8002
12:21:15:640 [2] get active endpoints for 0x00158d0001873510
12:21:17:318 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
12:21:17:318 ZCL attribute report 0x00158D0002E96D44 for cluster 0x0402, ep 0x01
12:21:17:331 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
12:21:17:331 ZCL attribute report 0x00158D0002E96D44 for cluster 0x0405, ep 0x01
12:21:17:336 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
12:21:17:336 ZCL attribute report 0x00158D0002E96D44 for cluster 0x0403, ep 0x01
12:21:17:671 Current channel 15
12:21:17:682 Device TTL 1793 s
12:21:18:640 CTRL skip polling while permit join is set
12:21:18:843 CTRL skip polling while permit join is set
12:21:18:849 FP indication 0x0000 / 0x8005 (0x00158D0001873510 / 0x092D)
12:21:18:849                       ...     (0x00158D0001873510 / 0x092D)
12:21:18:849 ZDP indication search sensors 0x00158D0001873510 (0x092D) cluster 0x8005
12:21:18:849 ZDP indication search sensors 0x00158D0001873510 (0x092D) clear timeout on cluster 0x8005
12:21:18:855 [3] get simple descriptor 0x01 for 0x00158d0001873510
12:21:19:069 don't close database yet, keep open for 900 seconds
12:21:19:069 SensorNode 29: Temperature 29 added
12:21:19:070 FP indication 0x0104 / 0x0402 (0x00158D0001873510 / 0x092D)
12:21:19:070                       ...     (0x00158D0001873510 / 0x092D)
12:21:19:070 no button map for: lumi.sensor_ht ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
12:21:19:070 ZCL attribute report 0x00158D0001873510 for cluster 0x0402, ep 0x01
[... 109 lines cut ...]
19:49:20:941 New websocket 192.168.178.241:29543 (state: 3) 
19:49:24:739 Bind response success for 0xd0cf5efffecb6fe2 cluster 0x0006
[... cut ...]
19:49:28:017 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
19:49:28:017 ZCL attribute report 0x00158D0001E55748 for cluster 0x0400, ep 0x01
19:49:28:030 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
19:49:28:030 ZCL attribute report 0x00158D0001E55748 for cluster 0x0406, ep 0x01
```

I will close this issue for now, since my issue is solved. But it still baffles me why these sensors are so hard to pair, while the others (Xiaomi and IKEA) were a breeze.

closed 2019-05-27 by alex3305