dresden-elektronik/deconz-rest-plugin #1491: Aqara 2-gang relay lumi.relay.c2acn01

andyjenkinson, 2019-05-02:
I just received this new Aqara LLKZMK11LM 2 channel relay for testing: 

https://community.hubitat.com/t/new-xiaomi-aqara-relay-2-channel/13335
https://www.banggood.com/XIAOMI-Aqara-LLKZMK11LM-Two-way-Control-Module-Wireless-Relay-Controller-2-Channels-Work-For-Mijia-APP-Home-Kit-Control-Module-p-1441658.html?rmmds=buy

The zigbee modelid is lumi.relay.c2acn01.

It adds fine and responds quite well, seems to work with groups just fine. The device even has switch inputs so can be used to retrofit lights to existing switches. Fills an important niche IMO.

Only a couple of issues with this one:

- the basic endpoint needs to be copied to the second cluster in order for the manufacturer etc to show on the second light, guess this is the same as for the ctrl_neutral switches mentioned in #1487 ?
- the version is just shown as 00000000 in the UI. Looks like there is no swversion reported; I suggest to get it from date code?
- these devices report power measurement, i don't know if anything is needed for that? is it even exposed anywhere by deconz?
- there's an analog cluster, no idea what this is for

Screenshots attached. I have a debug log which I can share if necessary, it's quite long

[image]
[image]
[image]
[image]
[image]

ebaauw, 2019-05-02:
> the basic endpoint needs to be copied to the second cluster in order for the manufacturer etc to show on the second light, guess this is the same as for the ctrl_neutral switches mentioned in #1487 ?

It would seem so.  The same change should fix this, as they also use manufacturer code 0x1037 for this switch.

> the version is just shown as 00000000 in the UI. Looks like there is no swversion reported; I suggest to get it from date code?

That should aleady happen (for the first endpoint).  Both version  and date code are in the _Basic_ cluster, so that should work for the second endpoint with the change to copy that.

> these devices report power measurement, i don't know if anything is needed for that? is it even exposed anywhere by deconz?

The _Electrical Measurement_ cluster should be exposed as a ZHAPower sensor, but this is a new model that needs to be whitelisted before sensor resources are created.  Could you press _Read_ on the GUI and post another screenshot?  We need to figure out the scale of the values for _RMS Voltage_, _RMS Current_ and _Active Power_ (assuming the relay actually reports anything useful for these).

> there's an analog cluster, no idea what this is for

Typically Xiaomi uses that for life-time consumption, exposed as ZHAConsumption.  Again, a screenshot (after reading the attributes) would be needed to figure out the scale.  It might report a change only after some significant consumption, and update the value only when the relay is switched.  Best attached a know load (e.g. 60W traditional lamp) and leave that on for an hour or two.

andyjenkinson, 2019-05-02:
I must have attached the wrong shots, doh! The values weren't different on read though there were no scale attributes but I don't have a load attached yet anyway. Will post back after I've done that.

For the version, if this is supposed to happen already, any idea why it doesn't show up in the UI?

ebaauw, 2019-05-02:
> The values weren't different on read though

That would mean the cluster is useless.  The 65536 and -32768 values are the defaults defined in general.xml, not values returned by the switch.

> there were no scale attributes

These are the multiplier and divisor attributes in the 0x0600 range.  Not all devices support these, and not all that do return meaningful values, though.

> but I don't have a load attached yet anyway

That might make a difference, but I'm not sure.  I haven't seen _Electrical Measurement_ on a Xiaomi device before, and it looks almost to good to be true.

andyjenkinson, 2019-05-03:
Other reports on the hubitat forum report it works so I am hopeful, but I
only see it on the first endpoint...

ebaauw, 2019-05-03:
> but I only see it on the first endpoint...

Yes, it only measures overall consumption/power/current/voltage, not per output wire.

ebaauw, 2019-05-03:
I've whitelisted the switch, which should cause the ZHAPower and ZHAConsumption sensors to be created (you need to re-pair the switch).  You need to compile the plugin from this commit, or wait for it to be included in the next version of deCONZ.

Could you run deCONZ with `--dbg-aps=2` and capture the Xiaomi special report for this sensor?  We'll be looking for something like (this is for a vibration sensor):
```
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:375 APS-DATA.indication srcAddr: 0x79f6, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 215, rssi: -64
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:375     asdu: 1c5f11640a01ff422e0121bd0b03281a0421a8430521470106240100010000082108030a21653598212800992125009a252900ffffdc04
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:376 no button handler for: lumi.vibration.aq1 ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 001
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:376 ZCL attribute report 0x00158D0002A52124 for cluster 0x0000, ep 0x01
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:376 0x00158D0002A52124 extract Xiaomi special attribute 0xFF01
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:376     01 battery 3005 (0x0BBD)
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:376     03 temperature 26 °C
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:376     04 unknown 17320 (0x43A8)
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:376     05 RSSI dB (?) 327 (0x0147)
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:376     06 LQI (?) 4295032832 (0x0100010000)
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:376     08 unknown 776 (0x0308)
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:376     0a unknown 13669 (0x3565)
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:377     98 unknown 40 (0x0028)
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:377     99 unknown 37 (0x0025)
May  1 08:03:23 pi2 deCONZ[7202]: 08:03:21:377     9a unknown 45084271696900 (0x2900FFFFDC04)
```

imbaimbaimba, 2019-06-19:
Can the switch inputs on this relay be used in deCONZ? Would be great if they could be used in the switch editor to use this relay as a zigbee transmitter.

ivan-leschinsky, 2019-07-30:
@manup @ebaauw  Any updates on this? I also have Aqara wall switch with neural line and it doesn't shows in the web UI, but through VNC I also see all clusters with switches and lights.

Any thoughts how to add them to UI to use then in HomeAssistant for example?
[image]

[image]

esseki, 2019-08-04:
> Could you run deCONZ with `--dbg-aps=2` and capture the Xiaomi special report for this sensor? We'll be looking for something like (this is for a vibration sensor):

Hi @ebaauw, in case you missed it, @Oliviakrkk provided you with the capture of the Xiaomi special report for this sensor.

See : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1411#issuecomment-485724957

Hope this can help moving forward witht eh support for this new device.

Mikha495, 2019-08-16:
Hi! I've a strange behavior of this relais. Can't delete it completely from Phoscon (using headless). When deleting one of the two lights, the other one seems to sit still in deconz and creates an error:
[image]

Anyone an idea how to delete it clean?

ebaauw, 2019-08-16:
> though HomeKit keeps telling me the battery is low (it’s connected to the powerline)

The REST API reports `null` for `config.battery`, so homebridge-hue exposes a _Battery_ service to HomeKit.  `state.battery` shouldn't be exposed as the device has no battery.  The value of `-1` for `state.power` is odd, as is the value `0` for `state.voltage`.  I wonder if this is because you deleted the light or because the device doesn't support power measurement, even though it exposes the 0x0b04 cluster.

> When deleting one of the two lights, the other one seems to sit still in deconz and creates an error:

The `/sensors` resource is giving the error, not the second `/lights` resource.  homebridge-hue expects a `/lights` resource to accompany a `ZHAPower` `/sensors` resource.  Makes me think that both `/lights` resources have been deleted, but the `/sensors` resource is left stale.  Maybe that causes the funny values?

Best delete the `/sensors` resource using the API, e.g. through `ph delete /sensors/4`.

Mikha495, 2019-08-17:
Thanks for the analysis. As far as I can remember, HomeKit told me straight from the first connection the battery would be at 0%. 

Regarding your suggestion for a clean removal: could you please explain it step by step for a bloody beginner?
I’m using deCONZ headless latest version on homebrigde.

thecem, 2019-08-19:
You have to connect to the REST-API:
https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/
and then delete the sensor via REST-API, I use the REST client as addon in Firefox.

Mikha495, 2019-08-20:
Thanks! Did that. Now no errors in Homebridge any more. 
Though the unit is still present in HomeKit (as light) and bothering me with the message of low battery on the main screen... 
What's now the best way to get completely rid of it? Would be very thankful for step-by-step instructions.

thecem, 2019-08-21:
Did you restart homebridge?
Did you restart IOS device?
If yes there is still a /lights resource in deconz.

Mikha495, 2019-08-21:
Yes, restarted everything. Just looked up in the REST API: no lights from Aqara there, but then checked the sensors and the LUMI entry is still there, despite I deleted sensors/4 yesterday. 
[image]

What to do?

Mikha495, 2019-08-23:
Anyone?

thecem, 2019-08-25:
Did you:

DELETE /api/key/config/light(or sensor)/4

?

Mikha495, 2019-08-26:
I did 
DELETE http://192.168.x.xxx:xx/api/mypersonalkey/sensors/4

Mikha495, 2019-08-26:
Sorted. Had again a deeper look and noticed, there was still a `sensors/3` as LUMI relay sitting in the config. 
Deleted it, and it's now gone from HomeKit.

lukicsl, 2019-09-02:
I Phoscon I get after troubles 2 lights light55 and light 56.
I put both lights into a group and assigned s switch for the toggle. Unfortunately only one relay switches. 
listing them API I get :
[image]

[image]
[image]

There are statements about copying something over, not having any clue how to do that.

alextud, 2019-09-14:
Relay switch is exposed as a dimmable light, is it normal as it cannot be dimmed from xiaomi ?
Can we changed it to a simple switch? Thanks

"1": {
            "etag": "9688c38429a2d7eaa730e95a0de35782",
            "hascolor": false,
            "manufacturername": "LUMI",
            "modelid": "lumi.relay.c2acn01",
            "name": "Kitchen light strip",
            "state": {
                "alert": "none",
                "on": true,
                "reachable": true
            },
            "swversion": null,
            "type": "Dimmable light",
            "uniqueid": "00:15:8d:00:03:92:a6:ae-01"
        }

Mikha495, 2019-09-25:
I've repaired the relay, as I've found a use case for it. It is working well, but what drives me crazy is the message in HomeKit, that the battery of the relay is low.
Does anyone else has this behavior? Any ideas how to solve it?

rcremasco, 2019-09-28:
> In Phoscon I get after troubles 2 lights light55 and light 56.
> I put both lights into a group and assigned s switch for the toggle. Unfortunately only one relay switches.
> listing them API I get :
> [image]
> 
> [image]
> [image]
> 
> There are statements about copying something over, not having any clue how to do that.

I have same issue, only second switch is working as expected. The first switch works just few seconds when added to group. It seems it loose group membership after few seconds has been added.
Any option to fix this behavior?

esseki, 2019-12-03:
> I have same issue, only second switch is working as expected.
The first switch works just few seconds when added to group. It seems it loose group membership after few seconds has been added.
> 
> more over, if i send command directly to light instead of group via home assistant switch will work.

I can report the same issue. Group command for on/off does not work for 1 of the relay. But if i send the command directly to the light it will switch ON/Off correctly.

Any chance this issue get investigated ?

xawme88, 2020-01-22:
Hi everybody!,

I'm using DeConz V2_05_71_stable using RaspBee with a Aqara LLKZMK11LM 2 channel relay and I read all 0.0 from Voltage, Current, Power and Consumption with LLKZMK11LM powered on with load into 2 channels.  

I read this values from openhab but deConz API only show power and consumption channel with a 0.0 values. Is it possible to read non zero values from power and consumption and publicate voltage and current channels from deConz API in order to read all the values from openhab.

Thank you very much for your time!!!

huuub, 2020-02-11:
I would also want to see this. Will this become available?

d3rw1n, 2020-03-11:
Will this model be officially supported? It’s been almost a year now.

Mimiix, 2020-06-05:
@d3rw1n is it supported already?

ivan-leschinsky, 2020-06-05:
@Mimiix I tired to wait and added 4$ CC5231 stick to my home assistant system with zigbee2mqtt for this Aquara switch

closed 2020-07-03 by stale[bot]

ThomSwiss, 2021-01-24:
I have the same issue on openhab. Can not see the current state of the device. In deconz directly it works as expected. In the phoscon app it works wrong.

go4anand-fedex, 2021-02-14:
Hey folks, I wanted to pass some leanings from my quest and mistakes. 
1. Before you conceal the relay switch in the junction box, please pair the relay with Aqara hub and update the firmware to the latest version. I went from 001 to 0046 version.  I don't know if there is a way to push updates to the switches using phoscon app. I was running into the known issue, where external toggle used in edge mode(dumb switches) were not reliable. 
2. If you are using homebridge, read the API documentation, phoscon keeps the device as a "Consumption XX" even after you delete the device.  Use Eve to determine the sensor ID and delete it to remove it from homekit.

psaggiomo, 2021-09-03:
> I can report the same issue. Group command for on/off does not work for 1 of the relay. But if i send the command directly to the light it will switch ON/Off correctly.
>
> Any chance this issue get [...]

I have the same issue

doreks, 2022-10-30:
I have the same problem here, none of the switches work .

ThomSwiss, 2022-10-30:
I got it working. I switched from openhab to Home Assistent (on the same raspbery 4). When you follow this description (https://www.home-assistant.io/installation/raspberrypi/), you should be able to add your conbee 2 stick (in my case). It apears automatically after the first boot. If you search for devices after, it works out of the box.

Another solution on Home Assistent.
Later I switched to Zigbee2MQTT in combination with Mosquitto broker Add On.