dresden-elektronik/deconz-rest-plugin #1492: Xiaomi pir , not reconnecting?  & osram plug is it worse?

pergolafabio, 2019-05-02:
ok, i have running firmware : 0x26320500 , running on HA (marhdoc addon)

all my xiaomi pir motion sensors were linked directly to my conbee, normal always green lines

see  here for mesh : https://www.dropbox.com/s/22wg7ls6207inql/xiaomi-osram.JPG?dl=0

2 questions/issues: 
1) my xiaomi end devices lost connectivty to my conbee? how can i bring it back? what is the cause? Pressing the reset button on pir, doesnt help..., just see blue light flashing, restarting deconz doesnt help... reboot the conbee stick doesnt help...

2) a few months before i had osram smart plugs connected, then sometimes my pir sensors were using the osram as router, that was very unstable, is there a newer firmware out for osram plugs to make them better?

thnx

Smanar, 2019-05-02:
1- you need to re-par them, without deleting them, just restart the procedure to detect them "Add new sensor"
2 - I have same problem for another manfacture that make realybad router, nothing to do, but good news, one time the xiaomi device have choose a router, they change rarely, so unplug the osram plug, make a detection (to make a mesh without it) and replug it.

pergolafabio, 2019-05-02:
1) ok, but is it normal? because i glued them on the wall , dont like it taking it off :) , what is the cause? how to prevent it? bad firmware on conbee/xiaomi ?
2) indeed xiami stick to the conbee, but eventually they change, so what can we do about ut

Smanar, 2019-05-02:
Even if you glued it, you still can acess to small button ?.
1 - No it's rare, I never have this problem. I have this problem every time a sensor try to connect to a switch without neutral, it works 24h, then the sensor leave the network.
It's hard to repair because to find the problem you need to sniff the zigbee transmition to find the problem.
If you have green link, directly on conbee on trought better router, it normaly not happen.
2 - Yeah, same for me, but all here will say the xiaomi device don't change router, and on my side it only happen on first hours, the device connect trought a router first, and after 24h, I see it direcly on conbee. But it never change good router for bad one.
It's the better solution I have, diconnect your badest router, make a correct mesh and reconnect them.

pergolafabio, 2019-05-02:
ah ok, yes i can still press that button
so i puth the conbee in search/pair mode, just press the button on it
.... i always thought that if you pair, you need to hold them verry close to the conbee... but maybe thats only the first time then

thnx for info

Smanar, 2019-05-02:
Nope, you need to put it close to the coordinator to use the touchlink feature, it's a feature for philips bulb for exemple, you don't have it on xiaomi device.

pergolafabio, 2019-05-02:
hmm ok, so i need to unglue them :(
but how can you pair the at their definitive place if you need them close to the conbee? :)

Smanar, 2019-05-02:
No, you don't need to unglue them ^^.
You don't need to put it close, keep it at it's definitive place.

The touchlink feature is not obligatory, and the Presence detector can't use it.

pergolafabio, 2019-05-02:
ok, gonna test it later !

thnx in advance

pergolafabio, 2019-05-02:
ok, had to unglue them, the xiaomi sensors needed close to the conbee
so repaired them again, they changed also name in phoscon, so needed to rename them
i hope this s not going to happen frequently, always thought deconz was a stable system

bdickhaus, 2019-05-03:
deCONZ is a stable system. Zigbee can be difficult sometimes. And the xiaomi stuff has its own quirks. You need to pair the sensors where they are going to be installed. If where they are installed is not within normal Zigbee ranges of the Conbee, you will need additional routers. The best routers out there are Hue bulbs. I have not found anything else that works reliably. I am struggling now trying to get a xiaomi sensor working through a Sylvania plug and it works for a short time and then drops off. Other devices, though, work just fine through the Sylvania plug. Almost all of my xiaomi sensors are working reliably through a Hue bulb. I have some ZWave bulbs that work just fine, but I am seriously considering replacing them with Hue bulbs to get better coverage for the sensors.

Smanar, 2019-05-03:
> ok, had to unglue them, the xiaomi sensors needed close to the conbee

No, they need to be at their definitive place, they can use the maximum range and router for pairing. Like have said @bdickhaus, if you can't pair them without unglue them, I don't think you have a good network.

On my side I m using ikea bulb wtih good result too.

pergolafabio, 2019-05-03:
@Smanar @bdickhaus 
1) ok gonna try the pairing again next time on location itself when it fails again, there are indeed 2 floors between xiaomi and conbee stick, but i always thought, green lines are GOOD ? so i have a good mesh...
2) if not, gonna invest in a router to place on each floor, i have osram smartplugs, but after reading, they are not reliable either  :(
the problem is, i already have a domotica system for my lights, so i dont actually need zigbee/hue lights. A light can only work as a router if its powered on right? so in my case thats a no-go , that means i also need to change my wall switches, sounds a but stupid...
So only smart plugs is a go for me? but after reading your comment, sylvania is a no-go either

its still strange, that sensors, was actually working perfectly fine for 2 months, and the other one also for 1 month prox

Smanar, 2019-05-03:
1- yep, it's just an indication, but green mean generaly ok. It's strange you can't do it if you have a green line before it leave the network ... You press > 10s the pairing button, and nothing happen in phoscon/deCONZ ?

You can try to, to improve the range by adding an USB extension cable. Else it will be easier to move the  conbee than your sensor, but the pairing works in definitive location. And 2 floors between conbee/device, for me it's too much, not possible without repeater.

I don't know the better plug, but I know some users with the osram one, that don't have problems, perhaps they have a better mesh. There is too ikea stuff, or using a CC2530 but I haven't never tried myself.

But if at your first test, you have put all yours devices close the conbee, it could be the problem, try with all your devices at good places and with plug power on.

pergolafabio, 2019-05-03:
i pressed 5 seconds like always when i pair :) not 10 seconds
usb extonsion is already in place
ok, gonna test some further again with those osram
or maybe this one as a repeater : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1428

Smanar, 2019-05-03:
Yep, I have thinked to that too, It works realy well with zigbee2mqtt, but I don't know for deCONZ.

pergolafabio, 2019-05-03:
ok, some stuff to read the for me, but gonna try osram again first, since i already have them
maybe there is also firmware for those osram devices

bdickhaus, 2019-05-05:
Your situation is similar to mine. Half of my house is zwave smart bulbs, and I really do not want to change them out, that is why I am trying to use the Sylvania plugs. Sylvania is osram. I hear the Smartthings plugs work well, so I ordered a couple. I might also change out one zwave bulb to Hue to provide a router in that part of the house. It could also be osram firmware, but I don't know an easy way to upgrade my plugs.

kugelkopf123, 2019-05-15:
Only for the info: Never had Problems with the Osram Smart+ Plugs and Xiaomi PIR Sensors. All of Mine PIR Sensors running through multiple osram Plugs als routers.

Smanar, 2019-05-16:
Thx, good to know.
Perhpas the authors problem is solved by
https://github.com/dresden-elektronik/deconz-rest-plugin/commit/31ef3962a4507e6310c23e4c497278f1fedb050b

pergolafabio, 2019-05-16:
Aha, really interesting ;)

closed 2019-09-20 by stale[bot]