dresden-elektronik/deconz-rest-plugin #1493: ConBee firmware update failed

ferby09, 2019-05-02:
Hi,
I just tried to update my ConBee to 26490700, but it fails

`GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
reset FTDI DM01H1J8 (dev: 0)
action: update firmware after 2 ms
flashing 158985 bytes: |==============================|
verify: ....
Flash update failed, invalid CRC. Please try again.`

This is the command I used.
`sudo GCFFlasher_internal -d 0 -f /home/fhem/deCONZ_ConBeeII_0x264907                                   00.bin.GCF`

Is it possible that the new firmwares are not compatible to ConBee as it says "ConBeeII" in the name? If that's true, where do I find the last compatible firmware for my ConBee?

Version 26480700 fails as well.

manup, 2019-05-02:
For ConBee I you need to select the latest AVR firmware version which is deCONZ_RPi_0x26330500.bin.GCF

ferby09, 2019-05-03:
Worked, thanks.
My missing lights reappeared as well.
And I totally forgot that I didn't use an override config for the custom port so this was obviously reset with the update. Maybe I'll look into it again, but it didn't work the first time I tried. Anyways this issue is solved. Thank you!

closed 2019-05-03 by ferby09