dresden-elektronik/deconz-rest-plugin #1495: Switch toggle & scene issue

lenisko, 2019-05-04:
There is small issue with buttons assigned like that:

[image]

**Typical Case:**
[Lights OFF]
Pressed ON/OFF once
[Lights ON]
Pressed ON/OFF once
[Lights OFF]

**Incorrect Cases:**
[Lights OFF]
Pressed Scene once
[Lights ON]
Pressed ON/OFF once
**[Lights ON]**
Pressed ON/OFF once
[Lights OFF]

Most likely SCENE is not flipping boolean value which is used for TOGGLE key.

daaaaa, 2019-05-07:
Same behavior happens to me.
It would be nice if this behavior would be fixed.

carloscae, 2019-05-07:
Yeah, this also works like this with Aqara Switches. Probably all switches.
Made me move away from switch automation on Phoscon to Home Assistant.

Switch will not pool current state of light, just change from last click. Weird.

manup, 2019-05-07:
The issue will be fixed soon (hopefully by 2.05.66)

Currently toggle is implemented via a extra flag which isn't synchronized with the groups actual on/off state. The fix will just use the `group.any_on` flag which reflects the real state no matter how it was applied.

Related issues: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1360 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1181

carloscae, 2019-05-07:
@manup you deserve a huge pitcher of beer for this.

DOliana, 2019-06-19:
@manup Version 2.05.66 was released a few days ago. I guess this fix did not make it, right?

manup, 2019-06-20:
>@manup Version 2.05.66 was released a few days ago. I guess this fix did not make it, right?

Sadly not yet, hope to get id done asap.

mike43525, 2019-06-26:
I am also experiencing this problem. When can I expect a fix?

mike43525, 2019-07-28:
Hello @manup,
what's the current status? Do you even plan to fix this bug?

I am seriously considering switching to zigbee2mqtt...

lenisko, 2019-11-25:
go away stale bot! ;-)

closed 2020-03-31 by stale[bot]