dresden-elektronik/deconz-rest-plugin #1496: Blinking dot top left of node display

tenholde, 2019-05-05:
I have 10 Keen Vents that are now working as far as OPEN/CLOSE   (ON/OFF) control.

I've been reading a lot to understand better how the ZigBee network is working.

I have one vent that the indicator at the top left is blinking green every 3-4 seconds.  Most of the other vents blink a few times blue infrequently.

Can someone explain what the different colors of the blinking indicator mean?  Any idea why this one vent might be continually active?

Smanar, 2019-05-07:
From what I know, but I m not pro.
- Red : error, timeout
- Yellow : incomplete traffic, wait for a return
- Blue normal traffic.

The blinking blue is something normal, on a device with report enabled it can blink every minuts for exemple, but on other device, the gateway pool realy fast the device, so it can blink all the time.

If you have same devices, they normaly blink at same rythm.

tenholde, 2019-05-12:
Thanks for the responses

closed 2019-05-12 by tenholde