dresden-elektronik/deconz-rest-plugin #1497: Device Limit or force connection

sragas, 2019-05-06:
Hi,
I didnt't find any suitable information. So I am asking if I understand everything correctly.
Per Coordinator (conbee) there is a device limit of 32 end devices. Also I'm using two tradfri repeater who seems to have a limit of 8 end devices (tested, not found any documentation).
So my zigbee network is actually mostly used for sensors and switches and therefore only the two tradfri routers and the conbee as actively repeating devices. 
The issue is that most devices try to connect to the tradfri repeater, even though the conbee is physicly much more closer and the device limit is quickly reached. There are only the router connected to the coordinators (according to GUI). Is there a way to force an end device to us a specific router/coordinator?

Thank you very much!

manup, 2019-05-06:
> Is there a way to force an end device to us a specific router/coordinator?

No that's not possible, end-devices manage parent selection on their own. When routers don't have anymore space left in their child table they won't be considered for joining. Therefore the establishment and maintenance of the mesh network is done automatically.

Please also make sure to use the latest firmware version 0x26330500 ConBee; or for ConBee II version 0x26490700.

sragas, 2019-05-06:
Thank yout for this information! May I should look deeper into mesh network and find out what causes the connection issues.

closed 2019-05-06 by sragas