dresden-elektronik/deconz-rest-plugin #1498: please add heiman water leak (HS1-WL-E) support

salopette, 2019-05-06:
Can you please add heiman water leak (HS1-WL-E)?

[image]
[image]
[image]
[image]

ebaauw, 2019-05-06:
What's the _Manufacturer Code_ in the _Node Info_ panel?

salopette, 2019-05-07:
[image]

salopette, 2019-05-09:
Hello, has something been done? How can it test?
Should I contribute something else?

Ronoon, 2019-05-09:
You have to look in the code section for updates in e-g- Bindings etc . Maybe it is integrated in the next version if you are lucky.

salopette, 2019-05-25:
Hello, I have made today on V2_05_65 update.
The sensor is connected, but it does not appear in the list.
I only see it at the nodes.

BottlecapDave, 2019-09-14:
I too am on the latest version and can't see the item in the list. I am using this in conjunction with Home Assistant, and it seems to have picked it up in there, however when the sensor is wet it's not reporting it.

I've migrated this device from Smartthings, where it was working as a "SmartSense Moisture Sensor".

opechunka, 2019-10-18:
I'm using Domoticz. Heiman Water sensor works fine, but no battery reports. Can it be fixed? The same problem with Heiman Smoke sensor. In contrary Heyman Temperature and humidity sensor reports battery well.

BottlecapDave, 2019-10-30:
I've taken another look at the traffic for the phoscon app. When I look at the "sensors" traffic, I can see the traffic

>  {
			"config": {
				"battery": 100,
				"on": true,
				"reachable": true
			},
			"ep": 1,
			"etag": "---",
			"manufacturername": "Heiman",
			"modelid": "WaterSensor-EM",
			"name": "WaterSensor-EM",
			"state": {
				"lastupdated": "none",
				"lowbattery": null,
				"tampered": null,
				"water": null
			},
			"swversion": "1.1.1",
			"type": "ZHAWater",
			"uniqueid": "---"
		},

Water always appears null even when the sensor is submerged in the water.

Interestingly, the api request to "sensors/new" is always reporting the water sensor back. I don't know if this means it hasn't finished setting up?

BottlecapDave, 2020-01-10:
I've just deleted the node and attempted to add it again (pushing the tamper button after joining), and it's now picking up the wet/dry state in HA.

However like @opechunka I'm also not geting battery information, which is odd as it looks like in my last post this was populated.

I've got another water sensor to setup, so I guess we'll see what that provides.

closed 2020-05-16 by stale[bot]