dresden-elektronik/deconz-rest-plugin #1501: deconz on debian

truidix, 2019-05-07:
Hey,

I'm trying to run deconz-dev under debian, with a conbee II.
I tried the repoway via the ubuntu packages --> no deconz package can be found.
A manual install with https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-dev-2.05.64.deb ws successfull, but i have to install 2.05.20 before.
Dev says it need deconz installed.
Anyways after installing dev, when I start deconz. -> systemct start deconz
My conbee II is not found, and the version number of phoscon app is still 2.05.20. 
It seems like I'm missing something. Anyone a Idea what is missing? 
I changed the dev via udev an added this to the service config, wihtout success..

Thanks

trudix

ebaauw, 2019-05-07:
You need to install the runtime deCONZ package, https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.05.64-qt5.deb, not the development support package.

truidix, 2019-05-07:
> You need to install the runtime deCONZ package, https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.05.64-qt5.deb, not the development support package.

Thanks .. My fault. So now 5.64 ist installed and starting.  
 you made my day.. :-)

closed 2019-05-07 by truidix