dresden-elektronik/deconz-rest-plugin #1502: Xiaomi Aqara Multi sensor battery level

ispiropoulos, 2019-05-09:
I have several Xiaomi Multi Sensors and Door / Window sensors paired using deConz.

The Door / Window sensors report the battry level, but the Multi sensors do not.

Is there something I can do to help integrate?

[image]

kugelkopf123, 2019-05-09:
As far as i know it takes some time till it get displayed. How long have you paired them? You can try to pair them once again without erasing them befor. Sometimes this helps.

ispiropoulos, 2019-05-09:
I have them for at least 4 days. I've also tried re-pairing them but no battery level.

kugelkopf123, 2019-05-09:
> I have them for at least 4 days. I've also tried re-pairing them but no battery level.

Strange. Idk. perhaps its some problem with the latest FW. I paired mine month ago. 
[image]

tunip, 2019-05-09:
Same issue here with the battery for Xiaomi multisensors. All other sensors (leak, motion, etc.) shows the battery Info. This is a new deconz setup which runs now 5 days.

bdickhaus, 2019-05-11:
I am seeing the same thing with one Xiaomi multisensor that was added about a week ago and re-paired 4 days ago. But I have several others that are reporting battery. The one that is not reporting battery in Phoscon is returning "battery": null in the REST api.

hubermic, 2019-05-11:
Same here with 3 Aqara multisensors. All of them reported Battery levels before - it took them an hour until the value was available in the Phoscon App as they report the vaue once every hour. After a fresh setup of my Pi however, the battery values are not available in the Phoscon App anymore. Waited 3 days, did several re-pairings, ... without success. Aqara door sensors and some other sensors still show battery values as before.

My-userid, 2019-05-12:
Same issue for me. Also updated to newest firmware, so it seems that something changed in the firmware.

Spirituss, 2019-05-15:
Same for me with Mi multi sensors (round ones).

manup, 2019-05-15:
The above commit fixes the issue. This will be available in deCONZ 2.05.65 arriving very soon.

DurrM, 2019-05-27:
So deCONZ 2.05.65 is listed in releases but there is no update option in de Phoscon WebApp. Is this a normal behaviour?

hubermic, 2019-05-27:
> So deCONZ 2.05.65 is listed in releases but there is no update option in de Phoscon WebApp. Is this a normal behaviour?
> Is there maybe a manual update option? I really want to get rid of those "low battery" warnings. #firstworldproblems

You can flash the firmware manually as described here:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

DurrM, 2019-05-27:
> You can flash the firmware manually as described here:
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Unfortunately the update (2.05.65) for the deCONZ package ist not listed here: https://www.dresden-elektronik.de/rpi/deconz/beta/

bubffm, 2019-05-28:
Having the same issue with wrong Batterie display on Aqara Multisensor.  I hope 2.05.65 will arive very soon...

@manup  any news on this one?

bubffm, 2019-05-30:
I have installed 2.05.65.  Deleted the Multisensors that had the battery issues and reconnected them again.

I get the "Version" of the sensor now displayed (which was showing as "unknown" before), but the battery level is still showing as "0%" in EVE.

Any ideas?

ebaauw, 2019-05-30:
Wait an hour or so for the Xioami special attribute report.  The sensor only reports the battery through there.

bubffm, 2019-05-30:
Thank you Erik. I just checked again and now Phoscon reports battery at 100%, EVE has picked this up now as well.  So this issue looks resolved.

robertlandes, 2019-06-03:
@manup I updated to 2.05.65 BETA about two hours ago. I can confirm, that after probably 1-1,5h all Xiaomi Multisenors (WSDCGQ11LM) now report battery properly (100% and not `null`).

Thanks!

ispiropoulos, 2019-06-07:
This is resolved indeed!

closed 2019-06-07 by ispiropoulos

muhkuhmcjones, 2020-11-16:
Hi There,

I have just configured a new system and I don't see the battery with firmware 26660700 version 2.05.88 / 15.10.2020. Any idea why this issue still occurs?

Mimiix, 2020-11-16:
@muhkuhmcjones You probably need to re-pair a few times. Sometimes it takes a bit.

muhkuhmcjones, 2020-11-16:
Thanks Mimiix,
just to double check:

1. Waiting after pairing for some time 
or
2. retry pairing again and again until I see the battery level immediatly after pairing succesful?

Thanks in advance.

SwoopX, 2020-11-16:
The battery level comes only with the recurring 50 mins report sent by the sensor. Can't recall if it was also transmitted by the physical button press. Note that those reports eventually get not received, so best wait a couple of hours.