dresden-elektronik/deconz-rest-plugin #1505: End device removal problem

lenisko, 2019-05-10:
I have some trouble removing devices from network, steps I took:

- Add new device
Device is successfully added
- Remove it via `Delete switch` from Phoscon App
Device disappears from network
- Removed battery from switch
- Wait a few seconds
After some time it's getting re-added.

Problematic devices:
* WXKG11LM
* Xiaomi Motion Sensor (same problem)

Few screens below

---

1. Successfully added device
[image]

2. Removal step
[image]

3. Device is removed
[image]

4. And it's re-added?
[image]

---

Product RaspBee
Version 2.05.63 / 4/3/2019
Firmware 26330500

Smanar, 2019-05-10:
Its something that can happen, it's because "neighbour table"
More info here, just wait some days before retrying to remove it.

lenisko, 2019-05-10:
@Smanar how can I remove it, after it was removed from the page? also `here` is missing an URL ;-)

Smanar, 2019-05-10:
Oups sorry, the link is here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1375
In fact ATM you need to wait some time for the "cached data" be cleared, and after that just make like you have done previously. Delete it in phoscon, and check in deconz if it's ok, else you can try to select the node in deconz and press delete key on keyboard.

lenisko, 2019-05-11:
Thank you @Smanar !

closed 2019-05-11 by lenisko