dresden-elektronik/deconz-rest-plugin #1506: Lights don't react after overnight / Debug possibility?

rendgeor, 2019-05-10:
Hello,
this morning the three lights assigned to a room do not react anymore.

I press the switch button and see in deconz-gui the blue note between the switch and the conbee II. But there is no update triggered in the lights.

Setup: one group with 3 lights, assigned is a switch.

I did not restart the deconz-gui service so far.

**Q: Is there a debug possibility/trace to find out/send to you why the lights became not reactive?**

rendgeor, 2019-05-10:
Update: Modyfing groups + temporary adding an already known light led to the fact that for a short time the light was responsible, but also with a big delay. At some point in time the light was not responsible anymore.

Also in deconz-gui:
- the green connection lines were missing
- the non-respondable lights showed for some time the red status lamp

Solution: I did edit->reboot device. Afterwards light are respondable (and without delay).

Question: Is there a way how to find out the reason next time why the usb stick hangs?

rendgeor, 2019-05-10:
Unfortunately happens again, and now even the reboot does not solve.

<pre><font color="#4E9A06">[184305.925523] </font><font color="#C4A000">usb 2-3</font>: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
<font color="#4E9A06">[184305.925530] </font><font color="#C4A000">usb 2-3</font>: New USB device strings: Mfr=1, Product=2, SerialNumber=3
<font color="#4E9A06">[184305.925534] </font><font color="#C4A000">usb 2-3</font>: Product: ConBee II
<font color="#4E9A06">[184305.925538] </font><font color="#C4A000">usb 2-3</font>: Manufacturer: dresden elektronik ingenieurtechnik GmbH
<font color="#4E9A06">[184305.925542] </font><font color="#C4A000">usb 2-3</font>: SerialNumber: DE1962502
<font color="#4E9A06">[184305.928790] </font><font color="#C4A000">cdc_acm 2-3:1.0</font>: ttyACM0: USB ACM device
<font color="#4E9A06">[184313.857881] </font><font color="#C4A000">usb 2-3</font>: USB disconnect, device number 24
<font color="#4E9A06">[184313.858062] </font><font color="#C4A000">cdc_acm 2-3:1.0</font><font color="#CC0000">: failed to set dtr/rts</font>
<font color="#4E9A06">[184314.207441] </font><font color="#C4A000">usb 2-3</font>: new full-speed USB device number 25 using xhci_hcd
<font color="#4E9A06">[184319.607138] </font><font color="#C4A000">usb 2-3</font><font color="#CC0000">: unable to read config index 0 descriptor/all</font>
<font color="#4E9A06">[184319.607159] </font><font color="#C4A000">usb 2-3</font><font color="#CC0000">: can&apos;t read configurations, error -110</font>
<font color="#4E9A06">[184319.735084] </font><font color="#C4A000">usb 2-3</font>: new full-speed USB device number 26 using xhci_hcd
</pre>

manup, 2019-05-11:
Hi, can you please provide some details about your setup:

* deCONZ and firmware version: *deCONZ >  Menu > Help > About deCONZ*
* Operating system
* Type of lights and switch

The deCONZ version is 2.05.64 and firmware version 0x26490700.
Do you already use a USB extension cable? It prevents interference with other periphery.

rendgeor, 2019-05-11:
@manup 
yes, The deCONZ version is 2.05.64 and firmware version 0x26490700.
Ubuntu
Ikea Tradfri Gu10 W + different switches/ switching via Phoscon (has the same result).

Btw: I assigned the same lights directly and via light group to different switches for testing recently.
Is it possible that the light getting confussed and therefore the problem happens? Can the Ikea tradfri lights handle the fact that two switches are assigned to them directly?

manup, 2019-05-12:
>Btw: I assigned the same lights directly and via light group to different switches for testing recently.
Is it possible that the light getting confussed and therefore the problem happens? Can the Ikea tradfri lights handle the fact that two switches are assigned to them directly?

Shouldn't be a problem, only if a lights group limit is been exceeded. I'm not sure how many groups a IKEA light supports, the GU10 firmware claims a capacity of 255 but I doubt this is correct since the count doesn't go down when the light is put in a group.

[image]

Minimum group count should be 8 for all lights. Would be interesting to check when the limit is exceeded and what happens in this case.

>The question is why in the beginning it worked fine and after some time it gets unstable...

How large is your network in total?

manup, 2019-05-12:
Also you'd said that the switch isn't trigger the lights anymore, are they also not responding when you trigger commands from the Phoscon App?

rendgeor, 2019-05-12:
> > The question is why in the beginning it worked fine and after some time it gets unstable...
> 
> How large is your network in total?

In actual test phase just 4 gu10 lights + 4 kind of switches.
Therefore it shouldn't touch the limits....

rendgeor, 2019-05-12:
>  are they also not responding when you trigger commands from the Phoscon App?

Yes, exactly. In the unrespondable or slow (light reacts delayed) phase it was the same if I use the hw remote or phosconn.
The same for no reaction at all afterwards.

I see in the deconz-gui also that the status lamp of the lights went red blinking. Does it mean communication problem?

BTW: I did reset the lights yesterday and attached them freshly.

closed 2019-09-16 by stale[bot]