dresden-elektronik/deconz-rest-plugin #1507: Websocket and light return

zoic21, 2019-05-10:
Hello,

I wonder why when in websocket we have a message from sensor I have uniqueid but when it's light I don't have uniqueid. It's little anoying because I used this uniqueid to find which command I need to update.

Another question, for light uniqueid always end with -03 ? 

Thank in advance.

Smanar, 2019-05-10:
???
You don't have uniqueid for light ? On my side I have
`2019-05-10 16:14:51.013 (deCONZ) ###### WebSocket Data : {'state': {'on': False, 'reachable': True, 'bri': 254, 'xy': [0.4962, 0.3527], 'hue': 0, 'x': 32520, 'y': 23116, 'sat': 0, 'alert': None, 'colormode': 'xy'}, 'r': 'lights', 'e': 'changed', 't': 'event', 'id': '2', 'uniqueid': '90:fd:9f:ff:fe:83:1d:17-01'}`
And nope, on my exemple the uniqueid end with "-01"
The last part in uniqueID is the endpoint.

zoic21, 2019-05-10:
No I only have : 
`{"e":"changed","state":{"y":15609},"id":"1","t":"event","r":"lights"}` or 
`{"e":"changed","id":"1","r":"lights","state":{"on":false},"t":"event"}`

Only deconz id not uniqueid....

Smanar, 2019-05-10:
On the config try to check the option called "websocketnotifyall", set it to "true".
But I never have same socket return than you, IDK if it's something normal.
Which one version have you ?

manup, 2019-05-11:
@zoic21 which deCONZ version are you using?

zoic21, 2019-05-12:
It's 2.5.25 so pretty old, I don't see that, I will ask user to update it.

manup, 2019-05-12:
Yes it was added in later version 2.05.44 so updating should fix the issue.

zoic21, 2019-05-13:
Ok after update all it's ok, thank for your reply, sorry for this stupide question...

closed 2019-05-13 by zoic21