dresden-elektronik/deconz-rest-plugin #1509: Support for Luzense devices

PrimaryInc, 2019-05-11:
Hi,

I've been testing a 2 luzense switches, but not been able to pair them,
As far as I know thy are not supported in deconz at the moment.

What can information can I provide for support?

wvuyk, 2019-05-11:
Check this page? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

corvy, 2019-05-13:
Yes they are ZigBee, the only info I have is here: https://www.unilamp.no/produkt/zig-b-veggbrytere/

These remotes are meant to be used with some "acutators" to dim traditional lights and make them smart. The great thing about them is that they fit directly into the 55mm frames from Elko in Norway. I am trying to chase the importer to find out who they buy their products from and see if they have more detailed product documentation. This is bought from a sub supplier (whitelabled) and rebranded for the Nordic market. I will see if I can get more information and share when I get it.

corvy, 2019-05-13:
Here is another one of the same source, but with another brand name. 

https://en.robbshop.nl/wallremote-8v-zigbee-robb-smarrt

corvy, 2019-05-13:
I think this might be the source: https://sunricher.en.alibaba.com/product/60812056361-0/Sunricher_wireless_Zigbee_3_0_single_color_keypad_smart_switch.html

corvy, 2019-05-16:
[ZigBee Shenzen Synricher.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3185560/ZigBee.Shenzen.Synricher.zip)

The only information I got from the producer is this. I could send them another question to get more information if you could help me formulate something. I think this is not detailed enough. They only state that they are "ZigBee certified" but still we are not able to join the switch to the gateway.

manup, 2019-05-16:
They look like the iCasa switches:

https://www.amazon.de/iCasa-Pulse-8S-ZigBee-Keypad/dp/B07L2SLY29

Looks like these are branded version of Sunricher OEM products.

http://www.sunricher.com/zigbee-series/zigbee-remotes.html

@ebaauw added supported for iCasa switches in 2.05.58.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_58

PrimaryInc, 2019-05-16:
Hi @manup @corvy ,

Perhaps our attempts to pair switch to DeConz has been wrong? 
I've uses pair "other" switch and set switch in pair mode, pairing times out with no switch found.
Any ideas?

manup, 2019-05-16:
Can you please provide some screenshots of the Basic cluster with the model identifier as described in:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Perhaps due the branding these are different from the iCasa model identifiers and need to be added.

ebaauw, 2019-05-16:
> They look like the iCasa switches:

But with different labels on the buttons, suggesting different commands (level control and multiple on/off instead of scene recall).

corvy, 2019-05-18:
@ebaauw could you share the pairing steps for the iCasa switch? Following the documentation on the luzense does not work. It never pairs. If it is fact the same hardware perhaps I could reset it and pair the same way.

ebaauw, 2019-05-18:
I didn’t have to do anything special, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1124.

I would try the usual remedies against the pairing-battery-powered-devices blues:
- Make sure the device has been reset;
- Make sure the device has a fresh battery (don’t trust blindly the battery shipped with the device);
- Power down all your routers and bring the device close to the gateway.  Make sure the gateway still has room for new end devices;
- Search for devices in the Phoscon app (open the network in the old web app).  Set the search time to 10 minutes, to make sure the network remains open during next steps;
- Wait for the device to appear in the deCONZ GUI.  Check the _Node List_ panel, sometimes new nodes are hidden behind existing ones or way off screen.  If needed, remove and re-insert the battery to make the device announce itself to the network;
- If the endpoints and clusters (behind the right of the two circles to the upper right of the node) don’t appear, read the node descriptor and simple descriptors from the left of the two circles, while forcing the device awake by pressing a button;
- If the REST resources aren’t created, read the attributes of the _Basic_ cluster from the _Cluster Info_ panel, while pressing a button to force the device awake.

If the device has different firmware with a different _Model Identifier_, it needs to be whitelisted, before the REST resources will be created.  Please provide the info from: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support.

ebaauw, 2019-05-18:
Loking at the manual, you need to follow steps 6 (Factory Reset Manually) and 3 (Zigbee Network Pairing through Coordinator or Hub).

While looking similar, they seem to have different firmware from the iCasa Pulse keypads, with a slighly different reset and pairing sequence (see manuals at http://www.icasa.io/downloads/).  Also the iCasa have additional output clusters, not mentioned in the Luzense manual: _Groups_, _Scenes_, _ZLL Commissioning_.  To be seen if that's an omission in the manual or really the case, though.

The battery power monitor function (under 15) looks interesting.  I haven't seen anything like this on the iCasa Pulse switches, and had to resort to polling the battery power.  The Pulse 8S seems to report battery when pressing S1 and S2 simultaneously, but the Pulse 4S doesn't.  The Pulse 2 only has On and Off buttons...

ebaauw, 2019-05-18:
Bloody hell, all three seem to report battery slightly under every 4 hours.  I totally missed that when integrating these.
```
May 17 06:37:12 pi1 deCONZ[929]: 06:37:07:556 no button handler for: ICZB-KPD14S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 08:51:38 pi1 deCONZ[929]: 08:51:35:296 no button handler for: ICZB-KPD12 ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 10:12:18 pi1 deCONZ[929]: 10:12:17:402 no button handler for: ICZB-KPD18S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 10:33:08 pi1 deCONZ[929]: 10:33:05:201 no button handler for: ICZB-KPD14S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 12:47:31 pi1 deCONZ[929]: 12:47:27:410 no button handler for: ICZB-KPD12 ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 14:04:55 pi1 deCONZ[929]: 14:04:32:671 no button handler for: ICZB-KPD14S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 14:06:15 pi1 deCONZ[929]: 14:06:12:523 no button handler for: ICZB-KPD12 ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 14:08:32 pi1 deCONZ[929]: 14:08:30:973 no button handler for: ICZB-KPD18S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 14:34:16 pi1 deCONZ[929]: 14:34:07:007 no button handler for: ICZB-KPD14S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 17:03:23 pi1 deCONZ[19468]: 17:03:23:272 no button handler for: ICZB-KPD12 ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 18:04:43 pi1 deCONZ[19468]: 18:04:42:285 no button handler for: ICZB-KPD18S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 18:12:22 pi1 deCONZ[19468]: 18:12:22:635 no button handler for: ICZB-KPD12 ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 18:30:14 pi1 deCONZ[19468]: 18:30:12:201 no button handler for: ICZB-KPD14S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 21:10:10 pi1 deCONZ[19468]: 21:10:08:793 no button handler for: ICZB-KPD12 ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 22:01:03 pi1 deCONZ[19468]: 22:01:03:228 no button handler for: ICZB-KPD18S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 17 22:26:29 pi1 deCONZ[19468]: 22:26:29:517 no button handler for: ICZB-KPD14S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 18 01:06:56 pi1 deCONZ[19468]: 01:06:53:869 no button handler for: ICZB-KPD12 ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 18 01:57:16 pi1 deCONZ[19468]: 01:57:15:328 no button handler for: ICZB-KPD18S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 18 02:22:49 pi1 deCONZ[19468]: 02:22:48:617 no button handler for: ICZB-KPD14S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 18 05:03:40 pi1 deCONZ[19468]: 05:03:40:132 no button handler for: ICZB-KPD12 ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 18 05:53:31 pi1 deCONZ[19468]: 05:53:30:493 no button handler for: ICZB-KPD18S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
May 18 06:18:54 pi1 deCONZ[19468]: 06:18:54:169 no button handler for: ICZB-KPD14S ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
```

corvy, 2019-05-19:
[image]

corvy, 2019-05-19:
[image]

corvy, 2019-05-19:
I think I got them working kindof. The first one is a four button, the last one is a 8 button. The GUI in phoscon does not recoginize that the device is joined, but I find them on the VNC. 

[image]
[image]

corvy, 2019-05-19:
Node info for the 8 button device: 

[image]

corvy, 2019-05-19:
Node info 8 button device: 
[image]

Node info 4 button device:
[image]

ebaauw, 2019-05-19:
Can you please double-check the _Model Identifier_ in the _Basic_ cluster of each of the switches?  I hope they use a different value of each type, or it will be very hard to distinguish them.

Can you also please manually read all the attributes (double click an attribute for the popup window, press a key on the switch, and press _Read_ in the popup window)?  I'm particularly interested in the _SW Build ID_.

The 8-button switch is exactly what I expected: four switch endpoints, one for each row.  I suspect the switch sends _Level Control_ commands when holding the buttons, and _On/Off_ commands when pressing/releasing them, just like the iCasa Pulse switches.  We need to double-check which commands, to expose the right `buttonevent` values.  
I think, for now, it needs to be exposed as four ZHASwitch resources, at least until `config.groups` can handle multiple values.  @manup?

The 4-button switch is a bit puzzling.  I suspect it doesn't send anything different for pressing vs holding the _On_ and _Off_ buttons.  I wonder if it does something on press of the _Dim Down_ and _Dim Up_ buttons.

Can you run deCONZ with `--dbg-aps=2` and, while pressing/releasing, holding, or releasing the different buttons search the logfile for messages like (these are for the iCasa Pulse 2):
```
May 19 15:50:28 pi1 deCONZ[21370]: 15:50:28:073 APS-DATA.indication srcAddr: 0xbc03, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -55
May 19 15:50:28 pi1 deCONZ[21370]: 15:50:28:073     asdu: 012801
May 19 15:50:28 pi1 deCONZ[21370]: 15:50:28:078 button 2002 On

May 19 15:50:30 pi1 deCONZ[21370]: 15:50:30:584 APS-DATA.indication srcAddr: 0xbc03, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -58
May 19 15:50:30 pi1 deCONZ[21370]: 15:50:30:584     asdu: 0129050032
May 19 15:50:30 pi1 deCONZ[21370]: 15:50:30:587 button 2001 Move up (with on/off)

May 19 15:50:32 pi1 deCONZ[21370]: 15:50:32:052 APS-DATA.indication srcAddr: 0xbc03, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -58
May 19 15:50:32 pi1 deCONZ[21370]: 15:50:32:052     asdu: 012a07
May 19 15:50:32 pi1 deCONZ[21370]: 15:50:32:058 button 2003 Stop_ (with on/off)

May 19 15:50:35 pi1 deCONZ[21370]: 15:50:34:455 APS-DATA.indication srcAddr: 0xbc03, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -55
May 19 15:50:35 pi1 deCONZ[21370]: 15:50:34:455     asdu: 012c050132
May 19 15:50:35 pi1 deCONZ[21370]: 15:50:34:459 button 1001 Move down (with on/off)

May 19 15:50:36 pi1 deCONZ[21370]: 15:50:36:603 APS-DATA.indication srcAddr: 0xbc03, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -55
May 19 15:50:36 pi1 deCONZ[21370]: 15:50:36:603     asdu: 012d07
May 19 15:50:36 pi1 deCONZ[21370]: 15:50:36:606 button 1003 Stop_ (with on/off)

May 19 15:50:37 pi1 deCONZ[21370]: 15:50:37:254 APS-DATA.indication srcAddr: 0xbc03, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -55
May 19 15:50:37 pi1 deCONZ[21370]: 15:50:37:254     asdu: 012e00
May 19 15:50:37 pi1 deCONZ[21370]: 15:50:37:259 button 1002 Off
```
The `srcAddr` should match the NWK address of the switch, the `profile` should match 0x0104 (ZHA), and the `cluster` probably matches 0x0006 (_On/Off_) or 0x0008 (_Level_).  Never mind the _Basic (0x0000) and _OTAU_ (0x0019) clusters.  The `asdu` contains the (binary) ZigBee command.  You won't see the third lines, until we've whitelisted the switches and created the buttom map.

corvy, 2019-05-19:
Correct identifier for 8 button switch, seems I did it wrong first time :)

[image]

corvy, 2019-05-19:
Not sure I 100% understand what you ask me to do with regards to reading but I have now double clicked all the lines from the screenshot above and clicked read to get more info. Then I get the following: 

[image]

Also I changed the options to be like this (running on hassio / home assistant):

```
{
  "deconz_device": "/dev/ttyAMA0",
  "web_port": 8080,
  "websockets_port": 8443,
  "debug_info": 1,
  "debug_aps": 1,
  "debug_zcl": 0,
  "debug_zdp": 0,
  "debug_otau": 0,
  "vnc": {
    "active": true,
    "port": 5900,
    "password": "changeme"
  }
}
```
I tried setting the debug_aps to 2 but then I get an error. So I changed it to 1 for now. Hope that gives you what you need. I tried pushing all 4 buttons (this is the 4 button one) and get the following output. Is this what you are looking for? 

```
19:03:58:627 APS-DATA.confirm id: 153, status: 0x00 SUCCESS
19:03:58:627 APS-DATA.confirm request id: 153 -> confirmed, timeout 41294112
19:03:58:659 APS-DATA.indication srcAddr: 0x001788010352e2f4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 143, rssi: -73
19:03:58:660 APS-DATA.indication request id: 153 -> finished
19:03:58:660 APS-DATA.request id: 153 erase from queue
19:03:58:808 APS-DATA.indication srcAddr: 0x000d6f000f950e9c, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
19:03:58:870 APS-DATA.indication srcAddr: 0x000d6f000f950e9c, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 199, rssi: -66
19:03:59:052 APS-DATA.request id: 161, addrmode: 0x03, addr: 0x001788010352e2f4, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
19:03:59:099 APS-DATA.confirm id: 161, status: 0x00 SUCCESS
19:03:59:099 APS-DATA.confirm request id: 161 -> confirmed, timeout 43437864
19:03:59:146 APS-DATA.indication srcAddr: 0x001788010352e2f4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 143, rssi: -73
19:03:59:146 APS-DATA.indication request id: 161 -> finished
19:03:59:146 APS-DATA.request id: 161 erase from queue
19:03:59:525 APS-DATA.request id: 165, addrmode: 0x03, addr: 0x001788010352e2f4, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
19:03:59:582 APS-DATA.confirm id: 165, status: 0x00 SUCCESS
19:03:59:582 APS-DATA.confirm request id: 165 -> confirmed, timeout 43106768
19:03:59:620 APS-DATA.indication srcAddr: 0x001788010352e2f4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 143, rssi: -73
19:03:59:620 APS-DATA.indication request id: 165 -> finished
19:03:59:620 APS-DATA.request id: 165 erase from queue
19:04:00:013 APS-DATA.request id: 170, addrmode: 0x03, addr: 0x001788010352e2f4, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
19:04:00:067 APS-DATA.confirm id: 170, status: 0x00 SUCCESS
19:04:00:067 APS-DATA.confirm request id: 170 -> confirmed, timeout 43679864
19:04:00:108 APS-DATA.indication srcAddr: 0x001788010352e2f4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 143, rssi: -73
19:04:00:109 APS-DATA.indication request id: 170 -> finished
19:04:00:109 APS-DATA.request id: 170 erase from queue
19:04:00:254 APS-DATA.indication srcAddr: 0x000d6f000f950e9c, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0300, lqi: 143, rssi: -73
19:04:00:484 APS-DATA.request id: 177, addrmode: 0x03, addr: 0x001788010352e2f4, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
19:04:00:541 APS-DATA.confirm id: 177, status: 0x00 SUCCESS
19:04:00:541 APS-DATA.confirm request id: 177 -> confirmed, timeout 43411200
19:04:00:573 APS-DATA.indication srcAddr: 0x001788010352e2f4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 143, rssi: -73
19:04:00:573 APS-DATA.indication request id: 177 -> finished
19:04:00:573 APS-DATA.request id: 177 erase from queue
19:04:00:965 APS-DATA.request id: 184, addrmode: 0x03, addr: 0x001788010352e2f4, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
19:04:01:032 APS-DATA.confirm id: 184, status: 0x00 SUCCESS
19:04:01:032 APS-DATA.confirm request id: 184 -> confirmed, timeout 43996264
19:04:01:064 APS-DATA.indication srcAddr: 0x001788010352e2f4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 143, rssi: -73
19:04:01:064 APS-DATA.indication request id: 184 -> finished
19:04:01:064 APS-DATA.request id: 184 erase from queue
19:04:01:446 APS-DATA.request id: 189, addrmode: 0x03, addr: 0x001788010352e2f4, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
19:04:01:503 APS-DATA.confirm id: 189, status: 0x00 SUCCESS
19:04:01:503 APS-DATA.confirm request id: 189 -> confirmed, timeout 43218784
19:04:01:540 APS-DATA.indication srcAddr: 0x001788010352e2f4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 135, rssi: -74
19:04:01:540 APS-DATA.indication request id: 189 -> finished
19:04:01:540 APS-DATA.request id: 189 erase from queue
[... 6 lines cut ...]
19:04:05:285 APS-DATA.request id: 216, addrmode: 0x03, addr: 0x000b57fffe39f23b, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
19:04:05:329 APS-DATA.confirm id: 216, status: 0x00 SUCCESS
19:04:05:330 APS-DATA.confirm request id: 216 -> confirmed, timeout 43106768
19:04:05:408 APS-DATA.indication srcAddr: 0x000b57fffe39f23b, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 215, rssi: -64
[... cut ...]
19:04:06:361 APS-DATA.confirm id: 227, status: 0x00 SUCCESS
19:04:06:361 APS-DATA.confirm request id: 227 -> confirmed, timeout 43124088
19:04:06:378 APS-DATA.confirm id: 230, status: 0x00 SUCCESS
19:04:06:378 APS-DATA.confirm request id: 230 -> erase from queue
19:04:06:404 aps request id: 230 finished, erase from queue
19:04:06:413 APS-DATA.indication srcAddr: 0x000b57fffe39f23b, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 207, rssi: -65
19:04:06:413 APS-DATA.indication request id: 227 -> finished
```

corvy, 2019-08-09:
Hello again, is there someone I could send these switches to for inclusion? Right now I just have them lying in a box and I think they could be very useful and affordable. 

I think these switches are the same as this: 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1286

I could send them to whomever wants to try and include them in Deconz :D

ebaauw, 2019-08-09:
> I tried setting the debug_aps to 2 but then I get an error.

I'm afraid that's hassio not supporting a value of 2 here.

> So I changed it to 1 for now. Hope that gives you what you need.

Unfortunately, no.  We need the payload of the message (the `asdu` part) to see which command (and parameters) they send.

> I could send them to whomever wants to try and include them in Deconz

I'll be happy to have a look at them.

corvy, 2019-08-09:
Great @ebaauw. Send me an email on stian at barmen dot nu and we can arrange the details. :)

velzend, 2019-08-17:
Would be really nice if these switches will be included/ supported. I ordered them including the raspbee with the assumption due to the compatibility with Philips Hue (according to some other fora) they would work out of the box.

Thanks for all the work so far... If I can deliver some debugging output let me know, I will be more than happy to share and get these switches to work.

ebaauw, 2019-09-11:
I received the switches by mail today, thanks @corvy.

Indeed, they're very similar to the iCasa pulse switches (the 8-button switch even has the same firmware image).  They did flip the position of the _On_ and _Off_ buttons, though, so I had to create new button maps.

The eight-button switch is straightforward: four _On_ and four _Off_ buttons, each pair controlling a separate group.  The buttons support press/release, hold, and release after hold, sending the usual _On_, _Off_, _Move up (with on/off)_, _Move down (with on/off)_, and _Stop (with on/off)_ commands.  So that leads to `buttonevent` values 1002, 1001/1003; 2002, 2001/2003; ...; 8002, 8001/8003.  Until we fix `state.group` (so it supports multiple groups), it's exposed as four ZHASwitch resources (one for each group):
```json
{
  "config": {
    "battery": 100,
    "group": "11009",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "4f739c696f64633242a08209e44daf83",
  "manufacturername": "Sunricher",
  "mode": 1,
  "modelid": "ZG2833K8_EU05",
  "name": "ZG2833K8_EU05 11",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2019-09-11T20:07:55"
  },
  "swversion": "2.2.3_r11",
  "type": "ZHASwitch",
  "uniqueid": "00:0d:6f:00:0f:95:0e:e0-01-1000"
}
```
```json
{
  "config": {
    "battery": 100,
    "group": "11010",
    "on": true,
    "reachable": true
  },
  "ep": 2,
  "etag": "4f739c696f64633242a08209e44daf83",
  "manufacturername": "Sunricher",
  "mode": 1,
  "modelid": "ZG2833K8_EU05",
  "name": "ZG2833K8_EU05 12",
  "state": {
    "buttonevent": 3002,
    "lastupdated": "2019-09-11T19:48:59"
  },
  "swversion": "2.2.3_r11",
  "type": "ZHASwitch",
  "uniqueid": "00:0d:6f:00:0f:95:0e:e0-02-1000"
}
```
```json
{
  "config": {
    "battery": 100,
    "group": "11011",
    "on": true,
    "reachable": true
  },
  "ep": 3,
  "etag": "4f739c696f64633242a08209e44daf83",
  "manufacturername": "Sunricher",
  "mode": 1,
  "modelid": "ZG2833K8_EU05",
  "name": "ZG2833K8_EU05 13",
  "state": {
    "buttonevent": 5002,
    "lastupdated": "2019-09-11T19:49:00"
  },
  "swversion": "2.2.3_r11",
  "type": "ZHASwitch",
  "uniqueid": "00:0d:6f:00:0f:95:0e:e0-03-1000"
}
```
```json
{
  "config": {
    "battery": 100,
    "group": "11012",
    "on": true,
    "reachable": true
  },
  "ep": 4,
  "etag": "4f739c696f64633242a08209e44daf83",
  "manufacturername": "Sunricher",
  "mode": 1,
  "modelid": "ZG2833K8_EU05",
  "name": "ZG2833K8_EU05 14",
  "state": {
    "buttonevent": 7002,
    "lastupdated": "2019-09-11T19:49:00"
  },
  "swversion": "2.2.3_r11",
  "type": "ZHASwitch",
  "uniqueid": "00:0d:6f:00:0f:95:0e:e0-04-1000"
}
```

The four-button CCT switch is somewhat a-typical.  The _On_ and _Off_ buttons only support press/release (_On_ and _Off_ commands).  The _Dim_ button only supports hold and release after hold (alternating between _Move up (with on/off)_, _Stop (with on/off)_ and _Move down (with on/off)_, _Stop (with on/off)_.  The _C/W_ button supports press/release, hold, and release after hold.  On press/release, it sends _Move to color temparature_, cycling through five different `ct` values.  On hold it alternates between _Move color temperature up_ and _Move color temperature down_.  On release after hold it sends _Stop Move Step_.  This leads to `buttonevent` values 1002, 2002, 3001/3003, 4002, 4001/4003 for a single ZHASwitch resource:
```json
{
  "config": {
    "battery": null,
    "group": "22529",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "9993867cdcae5b3c73c05f4bcaaa07c0",
  "manufacturername": "Sunricher",
  "mode": 1,
  "modelid": "ZGRC-KEY-002",
  "name": "ZGRC-KEY-002 10",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2019-09-11T20:08:30"
  },
  "swversion": "2.2.3_r11",
  "type": "ZHASwitch",
  "uniqueid": "00:0d:6f:00:0f:95:0e:9c-01-1000"
}
```

@manup there's some unholy `buttonMap` hack for the dresden elektronik lighting switch in handling _Move color temperature_.  Luckily I managed to find the `param` values for the Sunricher, so I haven't touched that code.

ebaauw, 2019-09-11:
@corvy, your picture above of the four-button switch looks different from the one you sent me.  What _Model Identifier_ does it report?  It also looks different from the four-button switch at ROBBshop.nl.

corvy, 2019-09-11:
It is the same one but only got a few different options for buttons.

corvy, 2019-09-12:
In the box there is some more button covers like this one.

[image]

ebaauw, 2019-09-13:
That's not just the cover; the electronics are in that module.  Different labels typically correspond to different commands (requiring a different button map to support the switch).  The other labels suggests _DimUp_ and _DimDown_ whereas this one has _Dim_ (alternating between _Up_ and _Down_) and _Color Temperature_.  The _RGB_ switch (from the website) is probably different as well.

corvy, 2019-09-13:
Ahhhh I see!

corvy, 2019-11-17:
Hello @ebaauw, I see now your changes has been merged into the latest release but I still do not understand how I can configure and set them up. Could I use some kind of GUI to pair them with lights and configure the use of the buttons?

ebaauw, 2019-11-17:
On pairing, the REST API plugin should configure the switches to send commands to a group (specified in `config.group`).  Simply add lights to this group to control them directly.

Alternatively, create rules based on the buttonevent values using the REST API.  Phoscon is not open source, so I cannot change that web app.

Did you ever receive the switches I sent back to you?  I never got a confirmation by the postoffice.

corvy, 2019-11-18:
Ok thanks @ebaauw. I will see if I can get my head around this :) Yes I got the switches back, thanks alot. Please if I can reimburse you for the freight I will. Just send me the details in email.

corvy, 2019-11-18:
I found a way with the single switch. It works fine now controlling 4 lights. I could configure the group in the wireless light control interface but the 8 button one only shows in VNC and not in the wireless light control. Can I edit the config.group directly somehow? Do you have a step-by-step way to do it in REST API? I have a REST API plugin but I am seriously on deep water there...

Here is the one-zone one (4 button) working fine: 
[image]

ebaauw, 2019-11-18:
> the 8 button one only shows in VNC and not in the wireless light control

Did the REST API plugin create the (four) resources for it (one for each button pair/endpoint)?  It should also have created four groups (one for each resource).  That might actually confuse the old web app - not sure.

> Can I edit the config.group directly somehow?

I don't think you can.  It's updated when the REST API plugin sees a group command from the corresponding endpoint.  To change the group, use the GUI to create a binding to a different group in the GUI.

corvy, 2019-11-19:
Thanks for responding @ebaauw 

No only the 4 button one created one group. The 8 button (four groups) did not show up in the APWeb interface. In VNC I can see all four just fine like this: 

[image]

In the Web API I get these groups. The bottom left one is the 4-button one, just renamed it. The 4 zone, 8 button one, does not show up. 
[image]

I have also connected over REST API with a REST client. Did a GET towards http://localhost:40850/api/<key>/groups and got this response: 
https://pastebin.com/ydL5Lpb3

Is there any way I could debug this further?

ebaauw, 2019-11-19:
Looks like the REST API plugin didn’t create anything for the 8-button switch.  Could you open the network from the old web app (settings tab) and then in the GUI read the _Basic_ cluster of the switch?  The label in the node should change as the REST resources are created.

corvy, 2019-11-19:
Yes I can try this tonight. I am not at home right now and I need to push the buttons to wake the switch up. But just to be clear on what you want me to do. 

1. go to the web interface and click "open network" under settings. 

[image]

2. Go to VNC, click one of the switches on the 4 zone/8 button swtich. Click Basic cluster and then "read" button under "Cluster Info". 

[image]

Potentially I would need to repeat for each of the four switches I guess?

ebaauw, 2019-11-19:
Yes and no, you should not have to repeat for each endpoint.  If you have multiple switches, you might have to do that for each switch.  Tip: set the _Duration_ on the web app to 10 min, so you have some more time to wake the switch(es) and read the _Basic_ cluster(s).

corvy, 2019-11-20:
[image]

It worked :) I had to repeat the "read" for each switch a few times before I got them all into groups. But in the end it worked!

corvy, 2019-11-20:
Another quick question for you @ebaauw. The 4 button one have no way to dim up-down for the lights assosciated. It will only turn on-off. But when I listen to deconz_events I can see when I hold in the on or off button it will fire a new event every half-a-second with a new ID. The events look like this: 

> Hendelse 7 avfyrt 19:06:
> {
>     "event_type": "deconz_event",
>     "data": {
>         "id": "switch_24",
>         "unique_id": "00:0d:6f:00:0f:95:0e:e0",
>         "event": 1002
>     },
>     "origin": "LOCAL",
>     "time_fired": "2019-11-20T18:06:35.278362+00:00",
>     "context": {
>         "id": "f21aba6a16a84b1cb8717eeb4a5d047a",
>         "parent_id": null,
>         "user_id": null
>     }
> }
> Hendelse 6 avfyrt 19:06:
> {
>     "event_type": "deconz_event",
>     "data": {
>         "id": "switch_24",
>         "unique_id": "00:0d:6f:00:0f:95:0e:e0",
>         "event": 1003
>     },
>     "origin": "LOCAL",
>     "time_fired": "2019-11-20T18:06:34.335304+00:00",
>     "context": {
>         "id": "d3d6e56c76604825afc887329bfdc713",
>         "parent_id": null,
>         "user_id": null
>     }
> }
> Hendelse 5 avfyrt 19:06:
> {
>     "event_type": "deconz_event",
>     "data": {
>         "id": "switch_24",
>         "unique_id": "00:0d:6f:00:0f:95:0e:e0",
>         "event": 1001
>     },
>     "origin": "LOCAL",
>     "time_fired": "2019-11-20T18:06:34.318269+00:00",
>     "context": {
>         "id": "a0a43bf7e8de4e92af151c725433211f",
>         "parent_id": null,
>         "user_id": null
>     }
> }
> Hendelse 4 avfyrt 19:06:
> {
>     "event_type": "deconz_event",
>     "data": {
>         "id": "switch_24",
>         "unique_id": "00:0d:6f:00:0f:95:0e:e0",
>         "event": 1003
>     },
>     "origin": "LOCAL",
>     "time_fired": "2019-11-20T18:06:33.081583+00:00",
>     "context": {
>         "id": "86a42f7726bc40e9843559cbdbe87bf3",
>         "parent_id": null,
>         "user_id": null
>     }
> }
> Hendelse 3 avfyrt 19:06:
> {
>     "event_type": "deconz_event",
>     "data": {
>         "id": "switch_24",
>         "unique_id": "00:0d:6f:00:0f:95:0e:e0",
>         "event": 1001
>     },
>     "origin": "LOCAL",
>     "time_fired": "2019-11-20T18:06:33.061875+00:00",
>     "context": {
>         "id": "399b71f195ff4f3fb2d2bc612bbf57c6",
>         "parent_id": null,
>         "user_id": null
>     }
> }
> Hendelse 2 avfyrt 19:06:
> {
>     "event_type": "deconz_event",
>     "data": {
>         "id": "switch_24",
>         "unique_id": "00:0d:6f:00:0f:95:0e:e0",
>         "event": 1003
>     },
>     "origin": "LOCAL",
>     "time_fired": "2019-11-20T18:06:31.837922+00:00",
>     "context": {
>         "id": "8549e969449448ab8753846b682baf28",
>         "parent_id": null,
>         "user_id": null
>     }
> }
> Hendelse 1 avfyrt 19:06:
> {
>     "event_type": "deconz_event",
>     "data": {
>         "id": "switch_24",
>         "unique_id": "00:0d:6f:00:0f:95:0e:e0",
>         "event": 1001
>     },
>     "origin": "LOCAL",
>     "time_fired": "2019-11-20T18:06:31.824760+00:00",
>     "context": {
>         "id": "57563b4b44a64d2da7340b6cb53573dd",
>         "parent_id": null,
>         "user_id": null
>     }
> }
> Hendelse 0 avfyrt 19:06:
> {
>     "event_type": "deconz_event",
>     "data": {
>         "id": "switch_24",
>         "unique_id": "00:0d:6f:00:0f:95:0e:e0",
>         "event": 1002
>     },
>     "origin": "LOCAL",
>     "time_fired": "2019-11-20T18:06:30.609955+00:00",
>     "context": {
>         "id": "3374a47feb4e40b18e16fd20ed1d962c",
>         "parent_id": null,
>         "user_id": null
>     }
> }

You can see the ID changes underways, and also the event changes from 1002 to the occational 1003 (to indicate that it is being held?). Is it any way to make it so that when you hold the button the lights actually gets brighter and brither or dimmer and dimmer?

ebaauw, 2019-11-20:
Looks like the binding from the client _Level Control_ cluster to the group hasn’t been setup.  For each of the four endpoints, it should be to the same group as the client _OnOff_ cluster.

Are you on deCONZ v2.05.70?  I think that version has serious issues with pairing devices.  I haven’t tried .71 yet.

corvy, 2019-11-20:
Yes I am on .70. I will update and try again. 👍

corvy, 2019-11-22:
Updated to .71 one but still it does not dim up and down. How should I attack this? Do I need to remove the remote and add it again?

ebaauw, 2019-11-22:
Should be enough just to reset the switch/dimmer and re-pair it.

corvy, 2019-11-22:
Thanks again @ebaauw. I have tried reset and pairing twice but still the dimming does not work. I get the groups, can pair the lamps and everything but long-press does not work. I only see it in deconz_event log. I have also tried in VNC to read the cluster-info. If you have any ideas I would be happy to try them out :)

ebaauw, 2019-11-22:
The switch sends a _Move (with on/off)_ command, when holding the button.  That part works, because deCONZ sees this as it creates the button events.  What type of lights did you add to the group?  Do they accept the _Move (with on/off)_ command?  Could you check in the deCONZ GUI, in the _Cluster Info_ panel of the _Level Control_ cluster of one of those lights, that they actually react to this command?

Note that the REST API creates a x002 button event when pressing/releasing the button; a x001 when holding the button, and a x003 when releasing the button after holding it.

I double-checked .71 with my iCasa switches.  Still had pairing issues with one of them (the other two worked fine), in that deCONZ wouldn't read the _Simple Descriptor(s)_ while the network was open (so the right drop-down circle for the endpoints/clusters wouldn't appear).  After fixing that (and for the other two) the groups and bindings were created alright.  Of course these switches only have one endpoint.  Did you force the switch awake during the entire pairing process by repeatedly pressing one of its buttons?

If the lights react to the _Move (with on/off)_ command, most likely the switch doesn't send the command to the corresponding group, but to group 0x0000 instead.  In that case, the binding from the client _Level Control_ cluster (on the switch) to the group hasn't been setup.  Kinda weird, as the binding from the client _OnOff_ cluster has been setup correctly (or the lights in the group wouldn't react to the press/release).  You can set up the binding manually in the _Bind Dropbox_ panel in the GUI.

corvy, 2019-11-23:
Thanks again for your time and effort @ebaauw :) 

I have GU10 Philips Hue second generation for two button pairs and  two IKEA lamps, E27 for the two others. All of which dimmed fine with the 4 button switch. 

I tried to check the _Level Control_ on one of the buttons but there is nothing. Is this related? 

[image]

corvy, 2019-11-23:
Now I found the Bind Dropbox 👯‍♂ But I cannot understand how I enable the binding. I read somewhere that I could pull the 008 (Level Control) from the switch onto the bind dropbox, but how do I add the group? 

[image]

[image]

ebaauw, 2019-11-23:
Select the radio button and enter the group ID (in hex) in the field.

corvy, 2019-11-23:
Ok just tried this. Took the first part of the switch and dragged the Level Control cluster onto the source, now it looks like this:

[image]

What do I pull onto destination? Do I select one lamp in the group? Or how do I find the group ID?

ebaauw, 2019-11-23:
That's looking good, so far.  You need to select the _Group_ radio button and enter the group ID.  This is listed as `config.group` in the ZHASwitch `/sensors` resource of the corresponding endpoint (check the `uniqueid`).  You need to convert the group ID to hex.

So e.g. for my iCasa switch, group 60929 (0xee01) corresponds to endpoint 0x01 (`uniqueid` ends with `-01-1000`).

```
  {
    "config": {
      "battery": 62,
      "group": "60929",
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "aa2c8d35a0e0616b0d0a3d1b5c4ff4a4",
    "manufacturername": "icasa",
    "mode": 1,
    "modelid": "ICZB-KPD18S",
    "name": "ICZB-KPD18S 13",
    "state": {
      "buttonevent": 4002,
      "lastupdated": "2019-11-22T15:42:47"
    },
    "swversion": "2.2.3_r10",
    "type": "ZHASwitch",
    "uniqueid": "00:0d:6f:00:10:65:8a:63-01-1000"
  }
```

corvy, 2019-11-23:
Almost there :P 

[image]

>   },
>   "24": {
>     "config": {
>       "battery": 33,
>       "group": "1025",
>       "on": true,
>       "reachable": true
>     },
>     "ep": 1,
>     "etag": "9dd2910f33f4ff704ca3ab68cc7dd2eb",
>     "manufacturername": "Sunricher",
>     "mode": 1,
>     "modelid": "ZG2833K8_EU05",
>     "name": "ZG2833K8_EU05 24",
>     "state": {
>       "buttonevent": 1002,
>       "lastupdated": "2019-11-23T10:00:43"
>     },
>     "swversion": "NULL",
>     "type": "ZHASwitch",
>     "uniqueid": "00:0d:6f:00:0f:95:0e:e0-01-1000"
>   },
 

I think this should work. Found the group ID in the API Web interface in the URL and also in the REST client but when I bind I get a timeout. Could that be why it did not work by itself when joining?

ebaauw, 2019-11-23:
> when I bind I get a timeout

Make sure to press a button to wake the switch.

corvy, 2019-11-23:
Got it paired now, stupid to forget to press a button. 😂 But still no dimming. The first group/switch is the top pair on the remote right?

ebaauw, 2019-11-23:
Yes.

Not sure why it doesn’t work.  Could you double-check that `config.group` doesn’t change between a press/release (x002) vs hold/long release (x001/x003) buttonevent?

Could try and bind the _Level Control_ cluster endpoint to a different group, and check that `config.group` does change after hold/long release?  What happens if you bind the _OnOff_ cluster to the same different group?

corvy, 2019-11-23:
Thanks for the suggestions, let me play a bit with it and get back to you 👌

corvy, 2019-11-24:
I found the solution 💃 I checked the battery level on the remote and found it to be low. I noticed since the remote seemed sluggish. Replaced the battery, re-paired it and now it all just works. Dimming and all! Thanks for the unlimited patience @ebaauw

corvy, 2019-11-30:
Hey @PrimaryInc you can close this issue. Solved by Mr @ebaauw 👍

[1 more reply]

closed 2019-12-02 by PrimaryInc