dresden-elektronik/deconz-rest-plugin #1512: Schedules do not get executed

philipptrenz, 2019-05-12:
Hi there, 

I'm trying to create timers to have a delayed switch off of smart sockets, but created schedules via the API do not get executed. 

Here is the command:

```
curl --header "Content-Type: application/json" \
  --request POST \
  --data '{
  	"name": "switch off smart plug",
  	"description": "switch off smart plug after 30 seconds",
  	"command": {
  		"address": "/api/<api_key>/lights/4/action",
  		"method": "PUT",
  		"body": {
  			"on": false
  		} 
  	}, 
  	"time": "PT00:00:30", 
  	"autodelete": true 
  }' \
  http://phoscon.local:80/api/<api_key>/schedules
```

Which results in: 

```
...
"10": {
        "activation": "start",
        "autodelete": true,
        "command": {
            "address": "/api/<api_key>/lights/4/action",
            "body": {
                "on": false
            },
            "method": "PUT"
        },
        "created": "2019-05-12T11:54:29",
        "description": "switch off smart plug after 30 seconds",
        "etag": "74cee03da9cf90dedb5c1821af9a7de0",
        "name": "switch off smart plug 10",
        "starttime": "2019-05-12T11:54:29",
        "status": "enabled",
        "time": "PT00:00:30"
    },
...
```

I also created a timer within the Phoscon app (1 minute), which works fine. The scheduler specifications look nearly identical. The only difference is, that the one created by the Phoscon app has a additional field `"localtime": "PT00:01:00"`

philipptrenz, 2019-05-12:
I now tried to use the same procedure as the Phoscon app does: Calling a scene of a group of lights.

```
curl --header "Content-Type: application/json" \
  --request POST \
  --data '{
    "name": "switch off smart plug scene",
    "description": "switch off smart plug scene after 30 seconds",
    "command": {
        "address": "/api/<api_key>/groups/2/scenes/1/recall",
        "body": {},
        "method": "PUT"
    },
    "autodelete": true,
    "time": "PT00:00:30"
	}' \
  http://phoscon.local:80/api/<api_key>/schedules
```

This works just fine, also when set via the API. So is setting timers for lights not supported or this an issue?

ebaauw, 2019-05-12:
There’s no `action` in a `/lights` resource.

philipptrenz, 2019-05-12:
My fault 😅
It would be great if the command gets validated by the API, though.

closed 2019-05-12 by philipptrenz