dresden-elektronik/deconz-rest-plugin #1513: Mi Smart Plug don't show power consumption in Home Assistant plugin

Spirituss, 2019-05-12:
None of my Mi Smart Plug show the power consumption in Home Assistant integration. Integration shows status and power correctly but for consumption it shows 'unknown' always.

Ronoon, 2019-05-12:
Actual Version:

The deCONZ version is 2.05.64 and firmware version 0x26490700.

Maybe you should update the firmware first an check again.

My-userid, 2019-05-12:
> Actual Version:
> 
> The deCONZ version is 2.05.64 and firmware version 0x26490700.
> 
> Maybe you should update the firmware first an check again.

That firmware is only for Conbee II.

Spirituss, 2019-05-13:
Hello,

 

Don’t understand the point. This is the screen of my Conbee I:

 

It has actual version, isn’t it?

Best regards

nunoary, 2019-05-18:
Yes. Firmware 26490700 is for Conbee 2.

Try to rescan the plugs. On Phoscon start search for new lights go to the plug and press the button for aprox. 5 seconds (until the blue light starts to flash). Don't need to reset or put the plug near the dongle.

Spirituss, 2019-05-18:
I reconnected plugs without deletion of existing ones. The same situation. NO ONE of my Mi plugs shows the consumption while HA shows the Power. Moreover, I noticed that for some of the plugs the Power updates very rarely or doesn't update at all. For example, I see of of the plug with permanent 100Watt power in HA.

Spirituss, 2019-05-18:
May be it will be useful. I found the next messages in the HA log for one of my plugs:
```
DEBUG (MainThread) [pydeconz] {'config': {'UTC': '2019-05-18T12:27:48',
            'apiversion': '1.16.0',
            'backup': {'errorcode': 0, 'status': 'idle'},
            'bridgeid': 'ххх',
            'datastoreversion': '60',
            'dhcp': True,
            'factorynew': False,
            'fwversion': '0x00000000',
...
'lights': {},
 'resourcelinks': {},
 'rules': {},
 'scenes': {},
 'schedules': {},
 'sensors': '74': {'config': {'on': True,
                               'reachable': False,
                               'temperature': 4000},
                    'ep': 3,
                    'etag': 'ххх',
                    'manufacturername': 'LUMI',
                    'modelid': 'lumi.plug',
                    'name': 'Consumption 74',
                    'state': {'consumption': None, 'lastupdated': 'none'},
                    'swversion': '04-20-2018',
                    'type': 'ZHAConsumption',
                    'uniqueid': '00:ххх'},
...
```

```
DEBUG (MainThread) [pydeconz.deconzdevice] Consumption 74 created as 
{'_async_callbacks': [],
 '_battery': None,
 '_consumption': None,
 '_device_id': '74',
 '_ep': 3,
 '_etag': 'xxx',
 '_lowbattery': None,
 '_manufacturername': 'LUMI',
 '_modelid': 'lumi.plug',
 '_name': 'Consumption 74',
 '_on': True,
 '_reachable': False,
 '_sensor_class': None,
 '_sensor_icon': None,
 '_sensor_unit': None,
 '_swversion': '04-20-2018',
 '_tampered': None,
 '_type': 'ZHAConsumption',
 '_uniqueid': '00:xxx'}
```

```
DEBUG (MainThread) [pydeconz.deconzdevice] Consumption 74: update temperature with 4000
DEBUG (MainThread) [pydeconz.deconzdevice] Consumption 74: update reachable with True
```

Spirituss, 2019-05-23:
Hi, any ideas re. the issue?

Spirituss, 2019-06-07:
Installed deconz 2.0.65 - the same situation, no any consumption value.

Spirituss, 2019-06-18:
I have installed 2.0.66. Now I can see Mi Plug's temperature in HA, but still can't see the consumption.

Hedgehog57, 2019-08-27:
Same here with latest Deconz.

ElBartoRUS, 2019-08-27:
Same thing, sad..

Sergiozzzz, 2019-08-27:
I have the same problem. Please fix asap. It would be highly appreciated indeed.

Spirituss, 2019-09-19:
I have checked in the last DeConz version (2.05.69) - still consumption sensor is unavailable in HA.

Spirituss, 2019-10-07:
Still waiting for the fix. 0.99.3

mihalski, 2019-10-24:
Running 0.100.3 and just got a ConBeeII and installed Deconz add-on for Hassio.
I had previously been running zigbee2mqtt and everything was fully featured and worked perfectly (except of course for range which was mediocre).

I have 6 outlets (not a fan of how they come up as lights, but it doesn't affect their functionality in and of itself):
[image]
[image]

 A few popped up with no power usage data:
[image]

This is a best case scenario where everything comes up (the naming could be much more useful if prefixed rather than requiring identification via the integrations page):
[image]

Some come up named after one of the power sensors instead:
[image]

It's all very random.. The things they have in common is that they either don't exist or do not update (or update once or twice).

Also, I can see that power is in Watts because there has been an update or two between all the 6 outlets, but what is consumption measured in? None of those reading have ever updated.

Spirituss, 2019-11-05:
Guys, do you plan to fix the missing features?

aksvenk, 2019-12-14:
I have the same issue as well and keen on seeing this fixed. Let us know if you need more information.

Spirituss, 2020-01-06:
Still have the same functional omission in 0.103.5.

wuppi83, 2020-01-28:
Hi Guys, first i thought it was a good choice to buy the Conbee 2 Stick. After a week, i'm i little 
disappointed. So many ToDo's and often no answers from the support. I'm sure you have to do sooo much ... but a short answer would be great :0)

You wrote on the compability list "Xiaomi Mi | Smart Plug | Zwischenstecker mit Strommessung", and in real you can use it only as an light :0( Please can you fix it !!! Thanks !

mike9011, 2020-03-10:
Is this suppose to work? I have the same issue running the latest firmware and deconz.

jeanclode76, 2020-03-22:
Same here ! Are those messages read by deconz guys ????

marine1988, 2020-04-02:
Im facing problem too xiaomi plux reconized as light LOL I wish I don't buy this crap and staid with good old xiaomi gateway :( any fix

Nuvolari360, 2020-04-21:
Having same problem. Only shows as light. Can turn on/off. But no power consumption.

samuellinde, 2020-04-24:
I just bought a Mi Smart Plug because power reading was listed as a supported feature on the deconz wiki page. Would love to see this fixed as well.

Frazco-s, 2020-04-25:
Same problem. I've just bought a Conbee and the Mi smart because it is the only one which have the consumption feature. Deconz 2.05.75, firmware 264A0700, Mi smart found in lights and no consumption yet. Very disapointed to see the Conbee is not fully compatible.
Please Phoscon, improve your software. Thanks

febalci, 2020-04-28:
When i first installed this plug in deconz; it was showing on/off and instant power value but no consumption. Deleted the plug in deconz and rescanned it. Now it shows total consumption value under 03 Home Automation Interface-000C Analog Input-0x0055 Present Value in total kW. 

However it almost never updates this value. The only way to update is to go to Phoscon, and from Cluster Info press Read button under 03-000C. Otherwise it always shows the value from the last Read button process.

Spirituss, 2020-05-11:
Hi guys! More than one year with no-solution looks like Deconz software is gone by the vendor.

burok88, 2020-05-15:
Same problem here.
I have updated the firmware of conbeeII (26580700) hoping that this will fix this. 
Unfortunatley still the same problem as above.
- Total Consumption never updates automatically in deconz. (Entity in HA changes sometimes also to: Not Available )
- Power (W) is not showing any value in Home Assistant (Entity in HA shows: Not Available)

I see that in the deconz wiki-page for supported devices shows the xiaomi plug with the following marking:

Vendor | Name                      | Measured Values | Issues
Xiaomi | lumi.plug.maeu01 3 | C, P                      | 2267

Maybe this issue should also be linked in the wiki page?
Or is it some problem of HA?

Spirituss, 2020-05-15:
> Or is it some problem of HA?

It is NOT the problem of HA, since there is no update of the parameters above mentioned in Phoscon which is not connected with HA.

mihalski, 2020-05-26:
Out of curiosity does zigbee2mqtt also have this problem?

I can't remember. I moved from that to the ConBee II and it feels like all problems with Deconz are simply ignored and it's been stuck in place without any noticeable improvements (at least from my point of view).

At the very least there seems to be a communication problem with customers being ignored.

jeanclode76, 2020-05-26:
> Out of curiosity does zigbee2mqtt also have this problem?
>
> I can't remember. I moved from that to the ConBee II and it feels like all problems with Deconz are simply ignored and it's been stuck in [...]

Exactly, no communication, no answers, same old problems never fixed for months .... That's really disappointing ....

febalci, 2020-05-26:
in my case, seems like deconz 2.05.76 solved Automatic updating of total consumption (HA deconz addon 5.3.3)

Mimiix, 2020-06-05:
Is this issue resolved?

samuellinde, 2020-06-05:
The latest update fixed it for me at least!

Den Fre 5 jun 2020 kl 14:41 skrev Dennis D:

closed 2020-06-12 by Mimiix