dresden-elektronik/deconz-rest-plugin #1514: Xiaomi Multi Sensor update schedule, battery and version

mtjoernelund, 2019-05-13:
Just got a Raspbee to try and set up some sensors from Xiaomi. Installation and pairing works effortlessly. However in the Phoscon webapp updated measurements only come in at very irregular intervals, mostly several hours apart. No battery level is reported and version is reported as unknown. (Round multi sensor). 

I’ve seen other threads with similar issues. Is this a known bug that is being worked on, or do I have a problem with the sensor I have got. I am holding off on buying more until this works reliable.

Spirituss, 2019-05-15:
I have the same issues. Some of my round Mi multi sensors don't show their battery level and behave very unstable.

Spirituss, 2019-05-16:
Here is the example of the problem case in Home Assistant with deCONZ integration:
[Some xiaomi sensors have no battery status via integrations with DeConz #23347](https://github.com/home-assistant/home-assistant/issues/23347)

thowi, 2019-06-15:
I've got similar issues with a square Xiaomi temp sensor.
It sends an update right after pairing in Phoscon, but then I'm not getting any updates from the sensor.
No updates even if I change the temp significantly (warm it up in my hands) or press the button (which should force an update).
Battery in the sensor is new and it's just one meter away from the Conbee stick.

Other Xiaomi sensors (window, motion) work perfectly fine, even 10m away from Conbee.
So either something is wrong with my particular sensor (broken?) or there is some software issue.

Xiaomi Temperature / Humidity Sensor
WSDCGQ11LM
CMIT ID: 2016DP6443

Gateway: ConBee
Version 2.05.64 / 22/04/2019
Firmware 26330500
deCONZ 2.05.64

Node info:
[image]

mtjoernelund, 2019-06-18:
Any updates on this issue? Can I install the commit referenced and help testing?

manup, 2019-06-18:
Please try with latest version 2.05.66.

thowi, 2019-06-18:
I'm using the Ubuntu/APT setup and the latest version is only 2.05.65.
That one still doesn't work for me.
Happy to try once it's available.
LMK if I can provide any debug information.

mtjoernelund, 2019-06-20:
@manup How do I run 2.05.66? Phoscon app reports 2.05.64 as latest version and does not allow me to update?

mtjoernelund, 2019-07-07:
Found the process for manual updating. With version 2.05.66 I now have the multi sensor reporting the firmware version, but still no battery percentage. Is there any additional debugging info I can help with to troubleshoot?

Smanar, 2019-07-07:
If you have just updated your version, you need to wait for some hours, to update the batterie value.

mtjoernelund, 2019-07-07:
I spoke to soon. Although Deconz had received an update from the sensors after the update (which did not give battery status), a manual refresh (i.e. push the button of the sensors to refresh) updated with battery state. However weirdly enough I still only have version info on one of the multi sensors, and nothing on the one Xiaomi door sensor I have (not that this is a real concern).

closed 2019-11-11 by stale[bot]