dresden-elektronik/deconz-rest-plugin #1516: Update Tradfri firmware through cli

renegade2k, 2019-05-13:
Hi.
It's really nice, that there are finally firmware updates (for tradfri devices) possible, but everyone talks about using USB devices and windows or the linux GUI ...
I'm running deconz on RPi without any gui (just openHAB as main system).
I tried downloading the Tradfri firmwares and placing it in ~/otau, then powercycle the Tradfri devices, but nothing happens ... 
Is it possible to update the firmware without gui at all?
Can't find anything about it ...

manup, 2019-05-13:
Lights should update automatically after a power cycle.

For end-devices like sensor and switch currently the gui is required.

phdelodder, 2019-05-15:
Where can the ota be downloaded? Is there a guide somewhere on how to do it?

@renegade2k the docker container from https://github.com/marthoc/docker-deconz can enable to gui and be accessed using vnc.

manup, 2019-05-15:
By executing the following script in the home directory, the ota files will be downloaded into `~/ota` 

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/ikea-ota-download.py

There is no guide available yet. Basically after ota download is done Ikea lights should start the update shortly after a power-cycle.

phdelodder, 2019-05-15:
@manup is this the same ota for the motion detector?

manup, 2019-05-16:
Yes the script downloads files for all IKEA products. However the ota update for battery powered sensors and switches won't start automatically, in order to not drain the battery without user permission.

For battery powered devices the update needs to be started manually in the deCONZ GUI OTA panel. We're working on more user friendly way to start updates per devices from the Phoscon App.

phdelodder, 2019-05-16:
Sorry to hijack the issue, still thanks for the info!

MickL, 2019-06-28:
Would be awesome to have a option to update firmware's in Phoscon :)

LightningRhino, 2019-07-01:
Sorry, can't find the option. Where do I have to look?

timbru31, 2019-07-01:
> Sorry, can't find the option. Where do I have to look?

In the deCONZ GUI. IIRC you might need to enable the plugin in the menu bar first.

LightningRhino, 2019-07-01:
Ok, found it. But it's as sophisticated as it was before. Still no Phoscon way to update the firmwares.

FearlessSpiff, 2019-07-03:
Thanks for this. Just found out I can update the Tradfri Motion Sensor firmware with deconz!

@LightningRhino : Just enable the Plugin, then check  "STD OTAU Plugin" Tab in the left panel. Now click the correct address (you have to check in the overview for the wanted device address). Then select the correct file and press "Update" button. 
Hope this helps...

Adje, 2019-08-19:
I've used the ikea-ota-download.py and i get some firmware files. The problem is that my lightbolb e27 600lu still has firmware 1.3.002 where there is a firmware file 1.3.009. 

i've switched te lightbolb on and off and even have reset the lightbolb but no succes.

closed 2019-12-24 by stale[bot]