dresden-elektronik/deconz-rest-plugin #1521: Unable to add Hue Motion sensor to network

Oliviakrkk, 2019-05-18:
Hi

I am trying to add new Hue motion sensor to my deconz instance. 

All steps:
0. Upgraded sensor on Hue gateway to use newest firmware.
1. Removed senso from the Hue hub. Unplugged Hue hub from power. 
2. Started "add new sensor" in Phoscon
3. Pushed button "setup" on the sensor.

After a couple of retires I can see sensor in deconz GUI, but not in phoscon. Pairing was not completed.

I tried to remove sensor from deconz and restart pairing on phoscon. Still not working.

Logs from docker container:
```
19:33:41:799 APS-DATA.indication from unknown node 0x0017880104B4E37A
19:33:41:799 new node - ext: 0x0017880104b4e37a, nwk: 0x8407
19:33:59:372 void zmController::nodeKeyPressed(deCONZ::zmNode*, int) node 0x0017880104b4e37a deleted
19:34:06:324 APS-DATA.indication from unknown node 0x0017880104B4E37A
19:34:06:326 new node - ext: 0x0017880104b4e37a, nwk: 0x8407
19:34:16:013 void zmController::nodeKeyPressed(deCONZ::zmNode*, int) node 0x0017880104b4e37a deleted
19:34:39:380 APS-DATA.indication from unknown node 0x0017880104B4E37A
19:34:39:382 new node - ext: 0x0017880104b4e37a, nwk: 0x8407
```

[image]

Also I am working on Conbee II.

Any idea what's wrong?

manup, 2019-05-18:
Did you use the Philips Hue motion sensor setup wizard in the Phoscon App?
The setup button should be pressed for about 10 seconds until it blinks.

Which versions are you running? Should be deCONZ 2.05.64 and ConBee II firmware version 0x26490700.

Oliviakrkk, 2019-05-19:
No. I did not see any setup button. I go to sensors --> add sensor --> push setup button on the sensor --> new device appear in deconz GUI...and then nothing happens. Only button "back" is active. 

I had to do reboot today. After reboot I can see more about sensor in deconz:

[image]

But still nothing on phoscon webui. 

Verion info:
Version 2.05.64 / 22/04/2019
Firmware 26490700

Oliviakrkk, 2019-05-19:
Hmm. I think I have bigger problem after migrating from Conbee I to II. 
Just found in logs:
```
14:57:06:624 ZCL attribute report 0x0000000000000000 for cluster 0x0000, ep 0x01
14:57:06:625 0x0000000000000000 extract Xiaomi special attribute 0xFF02
14:57:06:625 	64 on/off 1
14:57:06:625 	01 battery 3035 (0x0BDB)
14:57:06:625 	03 unsupported index (data type 0x21)
14:57:06:625 	04 unsupported index (data type 0x24)
14:57:06:625 	05 unsupported index (data type 0x21)
14:57:06:625 	06 unsupported index (data type 0x20)
```

Some multiple xiaomi devices have to be repaired. Otherwise address 0x0000000000000000 is shown and they do not work.

manup, 2019-05-19:
>No. I did not see any setup button. I go to sensors --> add sensor --> push setup button on the sensor --> new device appear in deconz GUI...and then nothing happens. Only button "back" is active.

I mean the Philips Motion sensor wizard under: *Sensors > Add sensor > Philips > Motion Sensor*
The button on the backside of the sensor needs to pressed for about **10 seconds** with a paper clip.

[image]

I've just testet it with ConBee II and the same versions, here it works.

One important note, if you have multiple tabs or browsers open with the Phoscon App, perhaps on different devices, you need to close these while joining new devices.

>Some multiple xiaomi devices have to be repaired. Otherwise address 0x0000000000000000 is shown and they do not work.

Hmm strange, with latest firmware deCONZ should always have the full mac address on incoming commands. Not sure how to reproduce this.

Oliviakrkk, 2019-05-19:
```
Vendor dresden elektronik
Product ConBee II
Version 2.05.64 / 22/04/2019
Firmware Not connected
```

Now I have "Firmware Not connected".  I will try with Conbee I....

Oliviakrkk, 2019-05-20:
Ok. I cofirm it was issue with Conbee II. Sensor is working fine on Conbee I.

closed 2019-05-20 by Oliviakrkk