dresden-elektronik/deconz-rest-plugin #1524: Since 1 weeks: Can't find any lights, sensors or switch

HeyJonny, 2019-05-19:
Hey there,

last weekend I discovered, that all lamps, sensor and switches are unreachable. I guessed, there was an update - I had problems before, that I have to restore a backup after reboot or update of the firmware/software.
However, the backup was loaded successfully but sensors/lamps/switches are still missing. I then tried to find everything new: Does not work ... I don't find anything ... I have no idea what to do.
 
Here are some steps I did:
- Rebooted everything
- updated the firmware and software via console
- updated the linux system
- tried to add switches and lamps and sensors

Some informations about the system I am using:
- Raspberry Pi 3 with RaspBee
- Linux raspberrypi 4.14.98-v7+
- Ikea Tradfri Bulbs, IKEA Switch (not the dimmer), Xiaomi/Aquara Temperatur sensor
- deCONZ/Phoscon:
         - Version 2.05.64 / 22.4.2019
         - Firmware not connected

Anyone has an idea? Is this issue known?
I hope, you can help me...

manup, 2019-05-19:
>Firmware not connected

Can you please check if the serial interface is still configured properly.

This is described in step 1) at https://phoscon.de/raspbee/install#raspbian

1968-Larry, 2019-05-19:
Try reset gateway before restore backup, I hade a similar problem and I had to reset first.

HeyJonny, 2019-06-17:
Hey guys,
sorry for the late answer, was really busy last weeks.
- Checked the serial interface: Everything according to the installation guide. Firmware still not connected.
- I have tried to reset the gateway and restored the backup: Didn't work
- Reset the gateway: Can't find lights.

I have no idea, what's the problem... Do you have any idea?!

HeyJonny, 2019-07-24:
Hello everybody,
I want to give an update on that issue. I bourght a new SD-Card, reinstalled the system (followed the User manual, using the pre-configured image and everything worked: I could find everything and it worked)
Today, I installed openhab2 and now the gateway in phoscon shows:
Hersteller       unknown
Product
Version          2.05.66 / 16.06.2019
Firmware       nicht verbunden
And i cannot find any lights ... However, i can control the lamps with Alexa, and also with the switches, thus ZigBee is woking, I guess... But why does Phoscon "fail"? 
I have no idea, if the installation of openhab is the problem. However: I have no idea what to do now .... Can please someone help me?

PS: Maybe this is a different issue ...

closed 2019-11-28 by stale[bot]