dresden-elektronik/deconz-rest-plugin #1527: Xiaomi Multisensor reports each value as a seperate Sensor

JoschaMiddendorf, 2019-05-19:
My Xiaomi Multisensors are reporting each value as a separate Sensors via API.
That means the API shows 3 Sensors for every Multisensor
Temperature, Humidity and Pressure.
That means I have to subscribe to 24 Sensors to get all the Values for just 8 Physical Multisensors. Not very Elegant! 
Is there any option to put all measured values in one API reported Sensor as it is shown in Phoscon and as the Xiaomi Vibration sensor realizes it?

Smanar, 2019-05-19:
Nope, because you have 1 device sensor by cluster (and 1 by endpoint)
IDK what you are doing with the API, but for exemple you can manage the 3 sensors to make only one with the application that use the API.

JoschaMiddendorf, 2019-05-19:
I don’t really understand what you mean with „1 device sensor by cluster (and 1 by endpoint)„. In Phoscon the sensor is shown as one. And on the API another sensor, the Xiaomi Vibration Sensor is also shown as one sensor with multiple values. 
battery
orientation
reachable
temperature
tiltangle
vibration
vibrationstrength

Why can’t Temperature, Humidity and Pressure be reported together as the vibration sensor is doing it?

I’m using the API to integrate the Sensor Values in my Smart Home Systen (FHEM).

chimpy, 2019-06-23:
Does your sensor look like this?

[image]

[image]

JoschaMiddendorf, 2019-06-23:
Yes it does. Besides the picture and an additional Pressure value. Hence I’m using the Aqara type Mulisensor.

closed 2019-10-28 by stale[bot]