dresden-elektronik/deconz-rest-plugin #1528: Xiaomi Aqara Multi Sensor - Battery value is no longer shown

holdestmade, 2019-05-20:
Used to have battery level shown and battery_level attribute in Home Assistant.

Other Aqara sensors still have the battery shown

```
Vendor
Xiaomi
Product
Multi Sensor
Versionunknown
Air pressure
1007
 hPa
Temperature
20
 °C
Humidity
56
 %
Refreshed
05/20/2019 11:29:02 AM
```

Smanar, 2019-05-20:
Hello, there is already a correctif (to test), just wait for it https://github.com/dresden-elektronik/deconz-rest-plugin/commit/31ef3962a4507e6310c23e4c497278f1fedb050b

JoschaMiddendorf, 2019-07-06:
Will that be implemented any time?

Smanar, 2019-07-06:
It is. Still not working on your side ?

JoschaMiddendorf, 2019-07-06:
I'm sorry, it works fine. Thank you.

bratanon, 2019-10-02:
Same issue here with the temperature sensor. Running 2.05.69

```
Vendor  Xiaomi
Product  Multi Sensor
Version  unknown
Air pressure  1003 hPa
Temperature  25 °C
Humidity  43 %
Refreshed  10/03/2019 12:40:39 AM 8 minutes ago
``

govido, 2020-01-18:
Hi, im running deconz 2.05.72 and added 3 Aqara multisensors today (new pressure version).
I'm missing version and battery as well, also 1 of my 2 Aqara buttons (gyro) doesn't report the battery level. The other one does...
Is there a problem with the pairing process?

closed 2020-05-24 by stale[bot]