dresden-elektronik/deconz-rest-plugin #1530: How to reinitialize Conbee II?

Oliviakrkk, 2019-05-20:
Hi

I tried to clone my network from Conbee I to Conbee II. After that I had tons of issues with Conbee II. 

How I can reinitialise it ?  I have previous mac written on the box, but how can I separate it from current network?

manup, 2019-05-20:
The most important part is to never run two coordinators with the same mac address side by side. Meaning if you have cloned the ConBee I mac address to ConBee II the former should no longer be powered (in signal range).

We're working on an easier way to restore the original mac address of a device, till then you can only write it via deCONZ settings dialog `custom mac address` + Save + Leave + Join.

Oliviakrkk, 2019-05-20:
Do I need to change other settings? Or just mac address?

Oliviakrkk, 2019-05-20:
How about TC address/network key and tc link key. They are the same as original conbee. Won't it mix up?

manup, 2019-05-21:
Oops good catch, the TC address must be same as the coordinator MAC address.

The  TC link key should be set to `0x5a6967426565416c6c69616e63653039` which is the default HA trust center link key.

Oliviakrkk, 2019-05-30:
Ok, thanks.

closed 2019-05-30 by Oliviakrkk