dresden-elektronik/deconz-rest-plugin #1531: I've added a new Xiaomi Smart Switch (WXKG11LM) but I cannot find it in Home Assistant

loz7777333, 2019-05-20:
I had an existing existing Xiaomi Smart Switch (WXKG11LM) set up (called switchKitchen). I've added another identical switch (switchNew) in Phoscon but I cannot find the new switch in Home Assistant. I notice the original switch has an entry called "CLIP-Sensor TOOGLE-ON-OFF" but the new one doesn't instead it only contains a battery level sensor. 

Product ConBee
Version 2.05.64 / 22/04/2019
Firmware 262F0500

This is how the two switches look under Integrations in Home Assistant

[image]

tefinger, 2019-05-20:
Switches just fire events in HA when they are pressed. Check the documentation here for details: https://www.home-assistant.io/components/deconz/#remote-control-devices

> Remote controls (ZHASwitch category) will not be exposed as regular entities, but as events named deconz_event with a payload of id and event. Id will be the device name from deCONZ and Event will be the momentary state of the switch. However, a sensor entity will be created that shows the battery level of the switch as reported by deCONZ, named sensor.device_name_battery_level.

Iskenof, 2019-05-25:
You need in HA the service "deconz.device_refresh" or you can remove/add the deconz integration new

[image]

closed 2019-09-29 by stale[bot]

mbriney, 2021-04-09:
@loz7777333 did you ever figure this out?  I'm running into the same issue.

loz7777333, 2021-04-10:
> @loz7777333 did you ever figure this out? I'm running into the same issue.

as one of the replies mentioned, you can't see the buttons in HA. I use node red (deconz_event) to pick up on the events that the buttons produce