dresden-elektronik/deconz-rest-plugin #1532: Tradfri remote / single key pressed in a row problem

rendgeor, 2019-05-21:
When the tradfri remote is attached to a light group the keys can be configured, e.g. scene switch forward.
Problem is that when the keys are pressed to quickly (below one second) no action is done and the key need to pressed again.

Is this a issue from the remote or the handling within deconz?

alex3305, 2019-05-21:
@rendgeor Same here. But when I double press the buttons they do work once every two clicks.

alex3305, 2019-09-19:
Not stale. Is there any update on this?

manup, 2019-09-19:
To my knowledge that's how the remote works, it has a small dead time (1 second?) after pressing the button before it will enable pressing again and fire the event.

rendgeor, 2019-09-19:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  No, Not true. It's how deconz handles key presses.Using direct communication to the light or even the Hue gateway there is no delay!Btw For all these issues would be nice to give some hints how the photon created rules could be modified using the rest-api. Then USER could try themself to find a good solution!                                                                                                                                                                                                                                                                                                                 Von: notifications@github.comGesendet: 19. September 2019 10:49An: deconz-rest-plugin@noreply.github.comAntworten: reply@reply.github.comCc: github.com@hypeo.de; mention@noreply.github.comBetreff: Re: [dresden-elektronik/deconz-rest-plugin] Tradfri remote / single key pressed in a row problem (#1532)  To my knowledge that's how the remote works, it has a small dead time (1 second?) after pressing the button before it will enable pressing again and fire the event.

manup, 2019-09-19:
>Using direct communication to the light or even the Hue gateway there is no delay!

Interesting, I think we can check if the logs indicate faster events and perhaps some debounce function gets activated in which case we could deactivate it for the IKEA switch.

> For all these issues would be nice to give some hints how the photon created rules could be modified using the rest-api. Then USER could try themself to find a good solution!

How do you mean this? You can inspect the rules which are created by the Phoscon App just like any other rules via the REST-API. It's not recommended to modify them directly because they will be rewritten every time the configuration of a switch assignment is changed in the switch editor.

To modify them it would be the easiest to create new rules based on the Phoscon App rules with the desired modifications.

closed 2020-01-24 by stale[bot]