dresden-elektronik/deconz-rest-plugin #1534: Support for Develco smoke sensor

SwoopX, 2019-05-21:
Hi there,

I'm hoping th get the smoke sensor integrated in deconz. Interestingly, it was quite a hassle to have it paired so the clusters became visible but what you can pull out of it seems quite useless (similar to [#1294](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1294)). I hope that's equally good.

[Technical manual](https://www.smarthome-store.de/media/documents/smoke-alarm-technical-manual.pdf)

Thanks!

ebaauw, 2019-05-22:
Interesting, this looks like the first smoke detector whose siren can be activated remotely.  Where did you buy it?

SwoopX, 2019-05-22:
:) Always looking for interesting stuff. I actually bought it as new at ebay Kleinanzeigen for 40€ after some research regarding zigbee smoke detectors. Just found 2 shops really selling them. The one associated to ubisys in GER (78€) and Cozify in FI (~49€)

b2un0, 2019-05-22:
You cal also activate the siren on Xiaomi MiJia Honeywell smoke detector

https://github.com/Koenkk/zigbee2mqtt/issues/420

SwoopX, 2019-05-22:
> you can also activate the siren on Xiaomi MiJia Honeywell smoke detector

Might bentrue but as farmasmI know, those do not have necessary certifications to be officially operated in Europe. The Develco‘s do ;)

ebaauw, 2019-05-22:
> you can also activate the siren on Xiaomi MiJia Honeywell smoke detector

That's not what I read.  Apparently, they managed to activate the sensor's self-test remotely (causing it to beep), but it doesn't support the _IAS_WD_ cluster.

ma-ca, 2019-05-23:
The Bitron 902010/24 smoke detector with siren has an _IAS_WD_ cluster that can be used to remotely start the siren.

ebaauw, 2019-05-23:
Cool, I totally missed that.  Does the REST API expose the _IAS WD_ cluster as a `/lights` resource (as it does for the Heiman Siren)?  I didn’t see that in your PR from January.

SwoopX, 2019-05-27:
@manup Mind taking a look at this and flag it for device integration? Much appreciated!

ma-ca, 2019-06-14:
I have created PR #1593 for siren support on smoke sensors. I also added the Develco smoke sensor to supported devices.

closed 2019-10-19 by stale[bot]

Smanar, 2021-09-16:
@SwoopX, sorry to make the undertaker, but this device is fully supported ATM or partially ?

SwoopX, 2021-09-16:
Fully, why?

Smanar, 2021-09-16:
I m reading your comment about the 2 clusters, this device create both ? the ZHAFire and the one for the siren ?

SwoopX, 2021-09-16:
The stuff here is 2 years old and I now got 6 of them 😂

You get 2 sensors (fire and temperature) + 1 light for the siren.

labaland, 2022-02-17:
The on/off switch for siren is not working correctly :( I just bought this as my wake up alarm :( i can start alarm but disable is not easy :/ any help ?

SwoopX, 2022-02-17:
Erm, you're using the smoke detector's siren as a wakeup alarm? You must be even more desperate than I am to not oversleep :)

Regardless, which on/off switch, the device doesn't have any? You can silence any alarm giving the device a firm press.

labaland, 2022-02-17:
> Erm, you're using the smoke detector's siren as a wakeup alarm? You must be even more desperate than I am to not oversleep :)
> 
> Regardless, which on/off switch, the device doesn't have any? You can silence any alarm giving the device a firm press.

i have so hard to wake up  my alarm clocks, i can sleep 1 hour to my "sonic boom" alarm clock :(

I have this entity on the siren: 
[image]

labaland, 2022-02-18:
When i turn on the switch it instantly turn it self off, but the siren starts, where can i report so they can fix it ?

dc911x, 2022-04-06:
@labaland Did you find a solution for this? My Frient smoke alarm (rebrand of the develco) also does the same.

When turning on the alarm, the switch turns off immediately and the alarms starts beeping. Only way to stop it is to press the "test" button. Also after a while it starts beeping again. It always needs 2 presses.

I wouldn't use it as a wake up alarm but was looking into using it as a burglar alarm.

Smanar, 2022-04-06:
Are you sure it s not a problem from the third app ?
If I m right you can trigger the siren using "alert":"select" and dtop it using "alert":"none".

dc911x, 2022-04-07:
> Are you sure it s not a problem from the third app ? If I m right you can trigger the siren using "alert":"select" and stop it using "alert":"none".

Haven't tried it with those settings only with the build-in toggle from HA. I'll try when I got a moment.

[image]