dresden-elektronik/deconz-rest-plugin #1537: Missing Color Temp Controls for Sylvania/OSRAM/LIGHTIFY 72569

SiliconAvatar, 2019-05-24:
I have a [Sylvania Smart Home 72569 LIGHTIFY](https://assets2.sylvania.com/media/bin/asset-1377955/SMART%2B%20Tunable%20White%20Edge-Lit%20Undercabinet%20Luminaire.pdf) under-cabinet lighting which I added using the Phoscon web app.

The on/off control works fine, however it's showing up as a RBG color control light, not a color temperature controlled light. This light *only* supports color temp.

It does, however, accept `put`ting a color temperature, and the light reacts correctly. Changing the RGB color does nothing.

In Home-Assistant is shows as `supported_features: 61`, and also only shows an RGB selector, however manually setting color temp via a service call also works just fine.

The album linked below shows all the information I have on these lights from deCONZ-
https://imgur.com/a/JXMKXnT

manup, 2019-05-24:
This is planned to be fixed in the version 2.05.66. We have some Sylvania lights now for testing and also noticed these issues.

hgelpke, 2019-05-26:
Perfect! I just came for this very issue.

manup, 2019-06-17:
The SYLVANIA tunable white lights are added now in version 2.05.66.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_66

SiliconAvatar, 2019-06-18:
Issue solved in 2.05.66.

Closing.

closed 2019-06-18 by SiliconAvatar