dresden-elektronik/deconz-rest-plugin #1539: ConBee "read-only", suddenly cannot send any packets

0ff, 2019-05-24:
Hi there,

starting this morning, my Odroid with ConBee (1) stopped sending zwave packages - it seems to be able to receive any package without any problems, but whenever I try to send a package (for example Read Attributes), the node will flash red and then switch into zombie mode. I guess this is because for some reason the packages are not sent out and run into a timeout.
Whenever any node sends information, this is received fine (say I click a switch, this is received + all changes are reflected as well). So it seems to be "just" sending that's not working.

My configuration is as below:

- Version 2.05.64 / 4/22/2019
- Firmware 26330500
- Armbian 5.85 stable Debian GNU/Linux 9 (stretch) 4.14.111-odroidxu4
- latest docker image from https://hub.docker.com/r/marthoc/deconz/

To troubleshoot this, I tried:

- restarting deconz
- restarting the whole sbc
- connecting the ConBee via a long extension cord
- resetting the gateway + restoring a backup
- restored config as per https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1447#issuecomment-487259124
- changing the channel from 15 to anything else and then back again (as no other node seemed to follow to the new channel)
- tried most of the above in different order and waited a couple of minutes with each try

At one point the controller suddenly connected successfully after a restart and was able to control all nodes just fine. I changed nothing except for a reboot to make sure it would persist, yet after the reboot I'm back to read-only.

<details>
<summary>Logs</summary>

```
deconz        | 11:27:09:098 ZDP skip fetch, node 0xB57FFFE469372 total unconfirmed zdp request count 2 [2]
deconz        | 11:27:09:168 APS-DATA.confirm id: 110, status: 0xA7 NO_ACK
deconz        | 11:27:10:682 ZCL read cluster: 0x0000, attribute: 0x0000
deconz        | 11:27:10:682 ZCL read cluster: 0x0000, attribute: 0x0001
deconz        | 11:27:10:682 ZCL read cluster: 0x0000, attribute: 0x0002
deconz        | 11:27:10:682 ZCL read cluster: 0x0000, attribute: 0x0003
deconz        | 11:27:10:682 ZCL cmd-req nwk: 0x1A04, profile: 0xC05E, cluster: 0x0000 cmd: 0x00
deconz        | 11:27:10:683 APS-DATA.request id: 223, addrmode: 0x02, addr: 0x1a04, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 3 len: 11 tx.options 0x00
deconz        | 11:27:10:683 ZCL read cluster: 0x0000, attribute: 0x0004
deconz        | 11:27:10:683 ZCL cmd-req nwk: 0x1A04, profile: 0xC05E, cluster: 0x0000 cmd: 0x00
deconz        | 11:27:10:683 APS-DATA.request id: 224, addrmode: 0x02, addr: 0x1a04, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 4 len: 5 tx.options 0x00
deconz        | 11:27:10:683 Delay APS request diff = 0
deconz        | 11:27:10:683 ZCL read cluster: 0x0000, attribute: 0x0005
deconz        | 11:27:10:683 ZCL cmd-req nwk: 0x1A04, profile: 0xC05E, cluster: 0x0000 cmd: 0x00
deconz        | 11:27:10:683 APS-DATA.request id: 225, addrmode: 0x02, addr: 0x1a04, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 5 len: 5 tx.options 0x00
deconz        | 11:27:10:683 Delay APS request diff = 0
deconz        | 11:27:10:683 Delay APS request diff = 0
deconz        | 11:27:10:683 ZCL read cluster: 0x0000, attribute: 0x0006
deconz        | 11:27:10:683 ZCL cmd-req nwk: 0x1A04, profile: 0xC05E, cluster: 0x0000 cmd: 0x00
deconz        | 11:27:10:683 APS-DATA.request id: 226, addrmode: 0x02, addr: 0x1a04, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 6 len: 5 tx.options 0x00
deconz        | 11:27:10:683 Delay APS request diff = 0
deconz        | 11:27:10:683 Delay APS request diff = 0
deconz        | 11:27:10:683 Delay APS request diff = 0
deconz        | 11:27:10:683 ZCL read cluster: 0x0000, attribute: 0x0007
deconz        | 11:27:10:683 ZCL read cluster: 0x0000, attribute: 0x0008
deconz        | 11:27:10:683 ZCL read cluster: 0x0000, attribute: 0x0009
deconz        | 11:27:10:683 ZCL read cluster: 0x0000, attribute: 0x000A
deconz        | 11:27:10:683 ZCL cmd-req nwk: 0x1A04, profile: 0xC05E, cluster: 0x0000 cmd: 0x00
deconz        | 11:27:10:683 APS-DATA.request id: 227, addrmode: 0x02, addr: 0x1a04, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 7 len: 11 tx.options 0x00
deconz        | 11:27:10:683 Delay APS request diff = 1
deconz        | 11:27:10:683 Delay APS request diff = 1
deconz        | 11:27:10:683 Delay APS request diff = 1
deconz        | 11:27:10:683 Delay APS request diff = 1
deconz        | 11:27:10:683 ZCL read cluster: 0x0000, attribute: 0x4000
deconz        | 11:27:10:683 ZCL cmd-req nwk: 0x1A04, profile: 0xC05E, cluster: 0x0000 cmd: 0x00
deconz        | 11:27:10:683 APS-DATA.request id: 228, addrmode: 0x02, addr: 0x1a04, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 8 len: 5 tx.options 0x00
deconz        | 11:27:10:683 Delay APS request diff = 1
deconz        | 11:27:10:683 Delay APS request diff = 1
deconz        | 11:27:10:683 Delay APS request diff = 1
deconz        | 11:27:10:683 Delay APS request diff = 1
deconz        | 11:27:10:683 Delay APS request diff = 1
deconz        | 11:27:10:683 ZCL read cluster: 0x0000, attribute: 0xFF0D
[... 110 lines cut ...]
deconz        | 11:27:15:092 APS-DATA.request id: 231, status: BUSY (counter: 2)
deconz        | 11:27:15:092 APS-DATA.request id: 231, set state: 0x05
deconz        | 11:27:15:092 emit artificial APSDE-DATA.confirm id: 231
deconz        | 11:27:15:092 APS-DATA.request id: 231, failed-status: APP_BUSY (0x02) 
deconz        | 11:27:15:099 aps request id: 231 failed, erase from queue
deconz        | 11:27:25:436 APS-DATA.confirm id: 186, status: 0xA7 NO_ACK
deconz        | 11:27:26:131 APS-DATA.request id: 38, addrmode: 0x03, addr: 0x000b57fffe1f82bb, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 8 len: 5 tx.options 0x00
deconz        | 11:27:34:231 APS-DATA.request id: 71, addrmode: 0x03, addr: 0x000b57fffe15363c, profile: 0x0104, cluster: 0x0004, ep: 0x01 -> 0x01 queue: 9 len: 4 tx.options 0x00
deconz        | 11:27:34:274 APS-DATA.request id: 71, status: BUSY (counter: 2)
deconz        | 11:27:34:274 APS-DATA.request id: 71, set state: 0x05
deconz        | 11:27:34:274 emit artificial APSDE-DATA.confirm id: 71
deconz        | 11:27:34:274 APS-DATA.request id: 71, failed-status: APP_BUSY (0x02) 
deconz        | 11:27:34:298 aps request id: 71 failed, erase from queue
deconz        | 11:27:34:435 APS-DATA.confirm id: 223, status: 0xA7 NO_ACK
deconz        | 11:27:34:739 APS-DATA.confirm id: 224, status: 0xA7 NO_ACK
deconz        | 11:27:35:299 APS-DATA.confirm id: 226, status: 0xA7 NO_ACK
deconz        | 11:27:35:507 APS-DATA.confirm id: 225, status: 0xA7 NO_ACK
deconz        | 11:27:36:116 APS-DATA.confirm id: 38, status: 0xD0 
deconz        | 11:27:36:372 APS-DATA.confirm id: 227, status: 0xA7 NO_ACK
```
</details>

Any ideas on what I could try to debug this?

Thanks so much in advance!

manup, 2019-05-24:
Please be careful when changing channels, some devices like IKEA lights tend to not properly support this.

Do you see any changes in the Phoscon App > Zigbee configuration page, which would indicate configuration isn't valid anymore?

If the configuration parameters, like PANID, channel and network key, are still the same please try the following:
1) unplug the ConBee for a few seconds
2) power-cycle the lights which don't respond

0ff, 2019-05-24:
Hi @manup, thanks for your response! Okay, I'll stop changing the channel then, thanks for the hint!
As for the configuration, everything looks good to me. You can easily follow my testing today ;)

[image]

I've now unplugged the ConBee for several minutes and also turned of 3 of my lamps (all on the same wall switch), when plugging the ConBee back in and restarting (lamps off) everything was as before, no communication.
*But:* Once I turned on the lamps, suddenly everything went back to working again.

Also restarting the Odroid completely did not break it again, so it seems to be fixed now - which I find very good but also very weird. Any ideas what might have caused this?
For your reference, the lamps in question pretty near to the controller (as in ~3m away) and it's these lamps:
[image]

A bit of further testing shows that I can get back into this state by down- and then upgrading the firmware for the ConBee. All I need to get out of it is _simply_ power-cycle the lamps and everything is back to normal.
If you want I'd love to debug this more (especially as I'm able to reproduce it)?

manup, 2019-05-24:
After a firmware update it might take a short while until the network routes are rebuild, but it shouldn't break the connectivity for long.

I can only guess that the lights went stuck in firmware, at least this have been seen before in other issues. Currently if it happens the only way to get the working state back is to power-cycle the related lights. Hopefully in future IKEA firmware updates this will be fixed.

closed 2019-09-28 by stale[bot]