dresden-elektronik/deconz-rest-plugin #1540: Support request for LifeControl MCLH-08

bastshoes, 2019-05-24:
I would like to ask for support of LifeControl MCLH-08. This enviroment sensor which provide information about temperature, humidity, eCO2 and VOC. Screen shots of clusters below.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Measured parameters located in Temperature measurement cluster.

0x0000 Measured Value - Temperature value. Should be divided by 100 
0x0001 Min Measured Value - Humidity value. Should be divided by 100
0x0002 Max Measured Value - eCO2 value in ppm
0x0003 Tolerance - VOC value in ppm. Should be divided by  10000.

bastshoes, 2019-08-23:
Trying to implement support of this sensor my self. Got a few questions.
1. There is no finggerprint for VOC sensor. Can I just add SensorFingerprint  fpVolatileOrganicCompoundsSensor?
2. How to deal with sensor which hold all his values in Humidity cluster under different endpoints?

closed 2019-12-28 by stale[bot]

MadMonkey87, 2020-05-05:
Hey @bastshoes 
Where did you buy this sensor, is it available anywhere?

bastshoes, 2020-05-05:
It's available in Russia and on aliexpress
LifeControl Store - https://a.aliexpress.ru/_eOyIy0