dresden-elektronik/deconz-rest-plugin #1541: Support request for LifeControl MCLH-03

bastshoes, 2019-05-24:
I would like to ask for support of LifeControl MCLH-03 - smart plug.
Actually it working for switching. But it also supports metering. Screen shots of clusters below:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

bastshoes, 2019-06-23:
Could you provide general guide line how to add sensors with power consumption for plug? I'll try to do it myself.

gonczarov, 2019-06-25:
make support please

bastshoes, 2019-08-22:
Support added via PR#1726

closed 2019-08-22 by bastshoes