dresden-elektronik/deconz-rest-plugin #1543: Xiaomi Door/window + Vibration + Temperature/humidity sensors go "Not reachable" - How to fix?

imbaimbaimba, 2019-05-24:
Hi!
My lights, motion sensors and switches seem to work, but the above mentioned sensors goes missing and does not come on again. Even pressing the button on them, or activating them by opening a door or similar does not get them reachable.

Is this a known bug at the moment? Or can I do something?

bdickhaus, 2019-05-25:
The Xiaomi sensors do not  handle losing their router and often do not automatically attempt to find another router if they do. If their main router is a bulb that might be powered off from time to time, this can cause this. Also there are some devices that are poor routers overall, and some that just don't work well with the Xiaomi sensors. Are you running the gui version of deCONZ? If so, you can see that the sensors no longer have a line connecting them to anything.

The only way to get them back is to open the network as if you were adding them for the first time and do a reset on the sensor (hold the button until it blinks repeatedly). 

What routers do you have in your system? Usually bulbs or mains connected switches, and deCONZ itself, if it is within close enough range. You can see all this with the deCONZ gui. I am using mostly Hue bulbs in my setup and they are excellent routers.

imbaimbaimba, 2019-05-25:
Thank you for your answer. I am using deconz on a Synology NAS in docker. I tried to install the GUI on my PC to see the mesh, but it seems like you cannot so that from another device. This seemed a bit weird to me, since I assume not many are using a Windows PC for deconz server.

I only have Hue bulbs as mains powered devices, nothing else. But a couple of them might have been powered off a few times. I just thought the mesh could heal itself since they never moved positions?

manup, 2019-05-25:
Are you using ConBee II and if so which firmware version? It looks like a known firmware bug of former releases. If you are running version 0x26420700, please update to the latest version 0x26490700 as described here:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

imbaimbaimba, 2019-05-25:
Nope it is the first one. I had it for a while. I have never done any manual updating to it, but I believe it happens automatically?

sragas, 2019-05-31:
Maybe the same issue discussed on #1488

kugelkopf123, 2019-06-06:
I think that's very strange. The Aqara / Xiaomi sensors are obviously always causing big problems. My question is, do these problems exist with other gateways like CU2530 / 31, the Xiaomi Gateway, Hue Bridge, etc. as well? Or is it explicitly a problem with deconz?
Although it is cheap China goods, I can not imagine that there are these problems even with the original Xiaomi Gateway.
Maybe there is someone here who has experience with other Gatways.
I'm not sure if my PIR and door sensors are working correctly. I have the feeling that sometimes they do not send their status.
There is only one Tradfri lamp in my household. The rest are Osram Smart + sockets.
My hardware (Raspbee) and software is on the most recent versions.
However, I have the feeling that all the sensors ran a few months ago with the much older firmware better.
The drama started when the duration of the PIR sensors was set to 90sec. No idea which FW that was.

Smanar, 2019-06-06:
No, I don't think, I have products from 5/6 manufacture,s I have tried 3 universal zigbee gateway, and I haven't more problem with xiaomi than other, their sensor are so basic and realy stable with deconz.
There isn't way with headless install to have information on mesh ? To see LQI or path ?

For Osram plug, I can see some people with lot of problem with and other without, so I realy can't say if it's a good router, but for ikea tradfri, I can confirm they are realy good with Xiaomi in my home.

closed 2019-10-11 by stale[bot]