dresden-elektronik/deconz-rest-plugin #1544: Light ca n only be switched on in a Phoscon group

peterlindeman, 2019-05-24:
I have a Hue light strip (and several Ikea bulbs) connected though Deconz (Raspbee). I use Home Assistant and nodered for switching via the rest api. So far so good. 

For some reason sometimes a light will not repsond anymore when I try to switch it directly but it responds on switching the light group in Phoscon it is member of. I have no clue at the moment of how to fix this issue. Anybody familiar with this problem?

Ps. Running latest release in docker, raspbee has latest firmware installed

Peter

bdickhaus, 2019-05-25:
A Zigbee group command is processed in the network as a "multicast". The members of the group share the command around. What that means is that a bulb may be out of direct range of the bridge, but in range of a group member. But a direct command should still be routed to the bulb. You probably have some connectivity issues. Unfortunately this can also be the result of router devices (like bulbs) not doing a good job of routing. If you are running deCONZ in the gui mode then you can see how everything is interconnected and that might help figure out where the problem resides.

Kane610, 2019-05-28:
There is also an issue with the home assistant component that sometimes fails sending signals. I have an improvement coming with next release. Beta should be built tomorrow. Would be good if you can verify if there is any improvement

peterlindeman, 2019-05-28:
Thanks for your answers. I have re-added the light that didn't work correct anymore. Couldn't see any network errors in VNC at the time it wasn't working. If there are any possibilities to debug it whenever the issue is back please let me know.

@Kane610 If there is anything I can do to do some tests let me know. I have indeed seen some problems here in the past that I could not switch lights in Home Assistant but could do it in Phoscon. Please let me know how to test.

Kane610, 2019-05-28:
It's as easy as upgrading to hass beta when it arrives later this week

SeraphimSerapis, 2019-06-02:
I am running into the same issue where one or two of many lights become unresponsive and eventually lose connectivity via HA and the deconz UI. Turning off still works via the Phoscon app but it won't accept turning on anymore.

closed 2019-10-07 by stale[bot]