dresden-elektronik/deconz-rest-plugin #1545: WXKG01LM Stating only battery level.

ispiropoulos, 2019-05-25:
On the latest version of deConz I pair a Mijia round switch (WXKG01LM) but it does not report state.
[image]

Only on Hass.io via the deConz addon there is an entity created just for battery level.
[image]

I have tried several switches with the same result (I actually bought 10 because I saw they were compatible and I’ve tried 4-5 of them)

bastshoes, 2019-05-25:
Switches do not have states. They are produce events.
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#wireless-switches

ispiropoulos, 2019-05-25:
@bastshoes that is totally my bad. I was keen to the previous Xiaomi gateway integration where it created stateful entities for the buttons as well :\

closed 2019-05-25 by ispiropoulos