dresden-elektronik/deconz-rest-plugin #1549: Deconz/Phoscon thermostat

wuast94, 2019-05-27:
Gibt es irgendwelche zigbee thermostate die kompatibel sind?

salopette, 2019-05-28:
Eurotronic  SPZB0001, eCozy 1TST-EU könne Zigbee, ich weis aber nicht ob Deconz.
Zigbee2mqtt wird aber unterstütz!

wvuyk, 2019-05-29:
it works fine here with deCONZ

Tobi0892, 2019-07-25:
@wvuyk or @salopette do you know any tutorial for the SPZB0001 with deCONZ? My search was rather unsuccessful

wvuyk, 2019-07-25:
You will find most of the inforamtion in that thread, but make sure to unhide the comments, for some reason a lot of comments are skipped, unless you choose to open it.

I do not have a tutorial. To join the thermostat there are only two steps needed:

- start sensor search
- Read basic cluster

To reset the device remove/reinsert batteries. Press minus+plus+boost and hold them until the device resets. Then you can do the sensor search again if needed.
You really do need to read the basic cluster through deCONZ, otherwise integration will not succeed.

choss38, 2019-08-07:
I can't join my SPZB0001 anymore. Deconz didn't find them and they displayed "Err" after the joining time.

> To join the thermostat there are only two steps needed:
    start sensor search

Could you confirm the node is displayed in Deconz at this step ?

>    Read basic cluster

And do you mean that we have to read the basic cluster of the new node next ?

You wrote:
> To reset the device remove/reinsert batteries. Press minus+plus+boost and hold them until the device resets. Then you can do the sensor search again if needed.

I reset a lot of time but it never join again. The only time I can join the Conbee is the first time I insert the batteries in a new thermostat. Do you have any idea about my problem ?

wvuyk, 2019-08-07:
The node should be displayed in the first step wil the oxfffff naming. Then select the node and read the basic cluster. 
If you are rejoining the device because the connection is lost, the existing device will start blinking in the deCONZ interface during the search - then read the basic cluster of the exising device.

choss38, 2019-08-07:
Sadly I can't see it anymore in Deconz. I don't know why. I don't know if I did it correctly or if the thermostat is not really reset or if I should look on the other side, Deconz and Conbee.

choss38, 2019-09-01:
I've all tried, I didn't understand what happens.
I started from scratch with the sd card image from Deconz with Raspbian and Deconz stable.
I reset a lot of time my spirit, restart Conbee, nothing worked.
And I didn't know what I've done, but the Spirit appeared sudently in the deconz windows. I didn't have to read basic cluster to identify it, it was already recognize. So I conclude it is stored on the Conbee Stick, am I wrong ?

Too happy with this result, I tried with my second Spirit I couldn't join again in deconz.
But this time, nothing new. I can't join it.

I tried to have debug information, but I really don't know which are necessary. 
I did it with this parameters: --dbg-info=2 --dbg-aps=1

I can identify the conbee (0x00212effff03b2ff 0x0000 ) and the first spirit (0x00158d000192d122 0x1984)

I don't know if there is a track of the second Spirit I placed in join mode ?
is this MAC Poll 0x02 0x01B5 ? (first track  at 10:24:45:438 in the logfile)

If somebody understand this logfile, please tell me if there is an indication, a clue to help me to join my Eurotronic Spirit zigbee again !

[deCONZlog.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3563373/deCONZlog.txt)

choss38, 2019-09-07:
I solved my problem and it was a silly problem : the batteries I was using on the Spirit were to weak !
With a new pair of battery the thermostat joined the network immediately !
I'm happy but what a waste of time...
So save your time, use new batteries !

closed 2020-01-12 by stale[bot]