dresden-elektronik/deconz-rest-plugin #1551: Periodic report, set sensor state to false.

Smanar, 2019-05-28:
Hello, I have a Xiaomi presence sensor in a room with lot of mouvement, and on log graph, I can see the sensor state is always "true" but every hour, it go to "False".

```
2019-05-28 19:30:39.128 (deCONZ) ### WebSocket Data : {'r': 'sensors', 't': 'event', 'uniqueid': '00:15:8d:00:02:b5:d2:ff-01-0406', 'e': 'changed', 'id': '16', 'state': {'lastupdated': '2019-05-28T17:30:39', 'presence': True}}
2019-05-28 19:30:39.128 (deCONZ) ### Update device (deCONZ - Presence bureau) : {'sValue': 'On', 'nValue': 1} 
2019-05-28 19:32:55.518 (deCONZ) ### WebSocket Data : {'r': 'sensors', 't': 'event', 'uniqueid': '00:15:8d:00:02:b5:d2:ff-01-0406', 'e': 'changed', 'id': '16', 'config': {'on': True, 'battery': 100, 'duration': 900, 'temperature': 

2600, 'reachable': True}}
2019-05-28 19:32:55.519 (deCONZ) ### Update device (deCONZ - Presence bureau) : {'BatteryLevel': 100, 'sValue': 'On', 'nValue': 1}, IGNORED , no changes !
2019-05-28 19:32:55.520 (deCONZ) ### WebSocket Data : {'r': 'sensors', 't': 'event', 'uniqueid': '00:15:8d:00:02:b5:d2:ff-01-0406', 'e': 'changed', 'id': '16', 'state': {'lastupdated': '2019-05-28T17:32:55', 'presence': False}}
2019-05-28 19:32:55.520 (deCONZ) ### Update device (deCONZ - Presence bureau) : {'sValue': 'Off', 'nValue': 0}
```

In fact just after all hourly report the api send "'presence': False", even it was set to true 2mn before and with a config.duration= 900.

And worst, if the hourly report happen 30s after the detection, it was set to false, an the sensor sleep for still 30s, so you can move like a zumba dancer, you will stay in the dark.

Smanar, 2019-05-29:
Ok, I m not sure but in the de_web_plugin.cpp around line 9341.
During the hourly report you are setting the presence value with 
`item->setValue(onOff);`

If I m right this sensor haven't on/off value, it just send report on presence detection, so every time it send hourly report, you are setting the bool value to false.

Smanar, 2019-06-01:
Ok so I have solved the problem with
```
            if (attrId == 0xff02)
            {
                // don't update Mija devices
                // e.g. lumi.sensor_motion always reports 1
            }
            else if (sensor.modelId().startsWith(QLatin1String("lumi.sensor_motion")))
            {
				// don't update Motion sensor
				// seemingly using the attribute 0xff02 to check is not enought
			}
            else if (sensor.modelId().startsWith(QLatin1String("lumi.sensor_wleak")))
            {
```

Like I have said in comment, IDK for other sensor but for this one checking for attribute 0xff02 don't work.

I can try a PR just for that, but not sure it will be usefull, if you have planned to refactoring the sensors code ?

Smanar, 2019-06-05:
Ok so, I have understand. It's something normal,
If I put a 100 s delay, the state returned by the API is true, but for the sensor after 60s, the state is false, so when it make the "periodic update" it force the false state.
We can update state timestamp if you are thinking it's usefull (not tested this one yet)

```
            else if (sensor.modelId().startsWith(QLatin1String("lumi.sensor_motion")))
            {
			// don't update Motion sensor state.
			// Imcompatibility with delay feature, and not really usefull
			sensor.updateStateTimestamp();
			enqueueEvent(Event(RSensors, RStateLastUpdated, sensor.id()));
			updated = true;
	    }
            else if (sensor.modelId().startsWith(QLatin1String("lumi.sensor_wleak")))
            {
```

manup, 2019-06-10:
The sensors hourly report has a invalid value of presence = false in case of motion?

I think it should be safe to ignore the presence value from the hourly report as in your PR, since the real presence signal already does the job right. We have done this for certain other Xiaomi devices and values as well, for the same reason.

Smanar, 2019-06-10:
Nope, the sensor have valid value, but this working mode is not compatible with duration feature (the problem is from the presence status in API).
If a set a 500s duration.
00:00 : Presence detection > sensor presence value is "True", duration timer start, API presence value is "True"
00:60 : No more presence so the value is "false" for sensor, but still "true" for API because of duration.
01:00 : Hourly report, the internal value "false" overwrite the "true" value in API.
So you will have end of presence detection after 60s instead of 500s.

closed 2019-06-20 by Smanar