dresden-elektronik/deconz-rest-plugin #1552: Innr SP120 smart plug sensors not named properly (ZHAConsumption & ZHAPower)

robertlandes, 2019-05-28:
Pairing _**Innr SP120**_ smart plugs creates three resources. One light and two sensor resource. Renaming the light resource in i.e. Phoscon app leaves the two sensors _ZHAConsumption_ and _ZHAPower_ unrenamed, defaulting to their _modelid_ (?). So they are named "_SP 120_", "_SP 120 (2)_" and so forth. This translates to several resources in i.e. Homebridge or Home Assistant with almost same naming and it is hard to differentiate them.

I would suggest to rename the two sensors accordingly the light resource. 
* Light: "Smartplug Livingroom"
* ZHAConsumption: "Smartplug Livingroom 2" or "Smartplug Livingroom Consumption"
* ZHAPower: "Smartplug Livingroom 3" or "Smartplug Livingroom Power"

Here is an example from `get /sensors` and `get /lights` through REST API:
```
Light resources

  "37": {
    "etag": "41f516523e441128aa5a652376549be4",
    "hascolor": false,
    "manufacturername": "innr",
    "modelid": "SP 120",
    "name": "Kaffeemaschine",
    "state": {
      "alert": "none",
      "bri": 254,
      "on": false,
      "reachable": true
    },
    "swversion": "2.0",
    "type": "On/Off plug-in unit",
    "uniqueid": "00:15:8d:00:03:61:a4:f3-01"
  },
  "38": {
    "etag": "3d3b4506f27306df2444255c39c5d75d",
    "hascolor": false,
    "manufacturername": "innr",
    "modelid": "SP 120",
    "name": "Küche Dunstabzug",
    "state": {
      "alert": "none",
      "bri": 254,
      "on": true,
      "reachable": true
    },
    "swversion": "2.0",
    "type": "On/Off plug-in unit",
    "uniqueid": "00:15:8d:00:03:44:28:40-01"
  }

Sensor resources
  "40": {
    "config": {
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "073b08f7c9430fa69c426ddc62614553",
    "manufacturername": "innr",
    "modelid": "SP 120",
    "name": "SP 120 (2)",
    "state": {
      "consumption": 110,
      "lastupdated": "2019-05-28T19:57:47"
    },
    "swversion": "2.0",
    "type": "ZHAConsumption",
    "uniqueid": "00:15:8d:00:03:44:28:40-01-0702"
  },

  "54": {
    "config": {
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "1fd83a67932b6ab94ca099213becfc08",
    "manufacturername": "innr",
    "modelid": "SP 120",
    "name": "SP 120 (2)",
    "state": {
      "current": 0,
      "lastupdated": "2019-05-28T19:57:57",
      "power": 0,
      "voltage": 225
    },
    "swversion": "2.0",
    "type": "ZHAPower",
    "uniqueid": "00:15:8d:00:03:44:28:40-01-0b04"
  },
---
    "37": {
    "config": {
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "64b475df340318e4b8365caa1d0627a5",
    "manufacturername": "innr",
    "modelid": "SP 120",
    "name": "SP 120",
    "state": {
      "consumption": 8900,
      "lastupdated": "2019-05-28T19:57:53"
    },
    "swversion": "2.0",
    "type": "ZHAConsumption",
    "uniqueid": "00:15:8d:00:03:61:a4:f3-01-0702"
  },

  
  "39": {
    "config": {
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "5a7365d6492ff6a81e3b603a2bb38bdb",
    "manufacturername": "innr",
    "modelid": "SP 120",
    "name": "SP 120",
    "state": {
      "current": 0,
      "lastupdated": "2019-05-28T19:57:25",
      "power": 0,
      "voltage": 227
    },
    "swversion": "2.0",
    "type": "ZHAPower",
    "uniqueid": "00:15:8d:00:03:61:a4:f3-01-0b04"
  },
```

ualex73, 2019-06-01:
You can rename them via the API, e.g.

curl -sS -X PUT -H "Content-Type: application/json" -H "Accept: application/json" -d '{"name": "sp1022_energy"}' http://192.168.0.1:80/api/0011223344/sensors/2

robertlandes, 2019-06-02:
@ualex73 I know, that I can rename them through API, still wanted to open an issue to maybe get it fixed in some later release.

Thanks for the hint! I guess I'll follow your suggestion and rename them through API for the time being.

ualex73, 2019-06-03:
@robertlandes I agree with you that this a bug in app, because it doesn't display the sensors of the SP120. The SP120 is long out there, so I am not sure if they are going this fix it (?)

robertlandes, 2019-06-03:
@ualex73 It is not super urgent for me, as it is working properly except the naming of the sensors.

robertlandes, 2019-06-06:
@manup I see that you actually included a fix for this in 2.05.65 ;) Just paired two new ones and saw, that they are now named as "Consumption `ID`" and "Power `ID`"... 

Would be awesome to see those two as sensors, but that's a different story. Closing this issue.

Thanks!

closed 2019-06-06 by robertlandes