dresden-elektronik/deconz-rest-plugin #1553: 2.05.65?

djwlindenaar, 2019-05-29:
It seems there is no .Deb file for the latest release on the beta site. Any reason for this?

manup, 2019-05-29:
Sorry for the delay they are uploaded now in the respective directories.
Note there are also new APT repositories which are updated for each release:

https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz

ebaauw, 2019-05-30:
The `stretch main` repository still hasn't been updated.

Maybe `zshark` could be published here as well?
```
Linux pi1 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l

$ sudo apt update
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Hit:3 http://phoscon.de/apt/deconz stretch InRelease      
Hit:4 https://deb.nodesource.com/node_10.x stretch InRelease
Fetched 15.0 kB in 1s (7,677 B/s)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
$ sudo apt list deconz
Listing... Done
deconz/unknown,now 2.05.64-raspbian-stretch-stable armhf [installed]
$ sudo apt list deconz-dev
Listing... Done
deconz-dev/unknown,now 2.05.64-raspbian-stretch-stable armhf [installed]
$ sudo apt list zshark
Listing... Done
$ cat /etc/apt/sources.list.d/deconz.list 
deb http://phoscon.de/apt/deconz             stretch main
```

bubffm, 2019-05-30:
ok, thats why I just couldnt get that update to install....

bubffm, 2019-05-30:
While I was checking for my updated deconz package, I noticed that there is an old jessie beta installed on my raspi which is unused. I tried to remove it using apt remove, but dont seem to be able to do so. Any idea?

`pi@raspberrypi:~ $ sudo apt list deconz -a
Listing... Done
deconz/unknown 2.05.64-raspbian-stretch-stable armhf [upgradable from: 2.05.64-raspbian-jessie-beta]
deconz/now 2.05.64-raspbian-jessie-beta armhf [installed,upgradable to: 2.05.64-raspbian-stretch-stable]

pi@raspberrypi:~ $ sudo apt remove 2.05.64-raspbian-jessie-beta
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package 2.05.64-raspbian-jessie-beta
E: Couldn't find any package by glob '2.05.64-raspbian-jessie-beta'
E: Couldn't find any package by regex '2.05.64-raspbian-jessie-beta'`

djwlindenaar, 2019-05-31:
Thanks! Installed it two days ago and things seem to have improved again. For one, the xiaomi temperature sensors started routing every hour again

closed 2019-05-31 by djwlindenaar