dresden-elektronik/deconz-rest-plugin #1556: DRESDEN ELEKTRONIK 600085 Funktaster Scene Switch

tomfritz1, 2019-05-29:
I have the Dresden Elektronik Scene Switch and it works perfectly. It turns groups on/off as it should. But: if I do not use the switch for several hours it seems to fall in a "sleep mode". In this sleep mode I have to press several times the button before it "awakes" again to work again perfectly.

closed 2019-10-03 by stale[bot]

trabantp60, 2020-01-13:
@tomfritz1: do you have found any solution therefore?

tomfritz1, 2020-01-15:
Not really. When use the switch after standby I click the button long enough too wake up....

trabantp60, 2020-01-28:
Alsooo (schreibe nun auf deutsch in der Hoffnung, trotzdem verstanden zu werden und die Aufmerksamkeit der Forenadmins zu wecken)!
Ich habe es nun geschafft, den Switch zu verzögerungsfreiem Schalten einer zurückgesetzten Hue-Lampe, die direkt am zurückgesetzten Switch neu angelernt wurde, zu bewegen. Funktioniert zuverlässig und reproduzierbar.
Nur schaffe ich es nun nicht mehr, dieses neue Subnet am zurückgesetzten ConBee II Stick einzubinden. Wie geht das, oder geht das überhaupt?