dresden-elektronik/deconz-rest-plugin #1558: Raspbian Buster release support

SeTeM, 2019-05-30:
Hello,

Today I've installed the latest Hassbian (based on Rasbian Buster):
```sh
system info:
  Distro......: Raspbian GNU/Linux buster/sid
  Kernel......: Linux 4.14.98-v7+
```
And wanted to install Phoscon, but faced an issue:
```sh
pi@hassbian:~ $ sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Ign:3 https://gitlab.com/hassbian/repository/raw/master buster InRelease
Get:4 https://gitlab.com/hassbian/repository/raw/master buster Release [3110 B]
Ign:5 https://gitlab.com/hassbian/repository/raw/master buster Release.gpg
Ign:6 http://phoscon.de/apt/deconz buster InRelease
Err:7 http://phoscon.de/apt/deconz buster Release
  404  Not Found [IP: 81.169.237.235 80]
Reading package lists... Done
E: The repository 'http://phoscon.de/apt/deconz buster Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
All three channels were tested: stable, beta and dev.

b2un0, 2019-05-30:
debian `buster` is still in `testing` so try to install the `stretch` version.

Klagio, 2019-06-28:
It took me 3 hours to setup a pi3 with buster (now is the only downloadable OS from raspberry site), just to find out this ..

lopf, 2019-08-15:
Same here with most recent buster version.

`Unpacking deconz (2.05.66-raspbian-buster-beta) ...
dpkg: error processing archive /var/cache/apt/archives/deconz_2.05.66-raspbian-buster-beta_armhf.deb (--unpack):
 unable to open '/usr/lib/systemd/system/deconz-homebridge.service.dpkg-new': No such file or directory
Errors were encountered while processing:
 /var/cache/apt/archives/deconz_2.05.66-raspbian-buster-beta_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)`

closed 2019-12-20 by stale[bot]