dresden-elektronik/deconz-rest-plugin #1559: No lightlevel/lux readings from Hue motion sensor

maguiresf, 2019-05-30:
Hi,

I don't seem to be able to get any readings for lightlevel or lux from Hue motion sensors connected to a Conbee2 gateway. The reading never changes in HomeAssistant, nor the Phoscon web UI nor the REST API. I've tried forcing a read from the deconz GUI and that always seems to return 0. This is the second motion sensor I've tried now and they've both had the same issue. If I connect them to the Hue bridge the light level works fine. Motion sensor is running firmware 6.1.0.18912. Gateway info:

Vendor dresden elektronik
Product ConBee II
Version 2.05.64 / 4/22/2019
Firmware 26490700

There is definitely light in the room, I've tested by placing the motion sensor right next to another one that is detecting a non-zero light level. Motion sensing and temperature sensing appear to work fine.

Any help would be greatly appreciated.

ebaauw, 2019-05-30:
Sounds like attribute reporting hasn't been setup.  Kinda odd that it would happen independently for two sensors, though.  Do the values update if you read the attributes of the _Illuminance Measurement_ cluster in the _Cluster Info_ panel in the deCONZ GUI?  You can verify the reporting setup by double-clicking on _Measured Value_ and press _Read Config_:
[image]

maguiresf, 2019-05-30:
Nope, don't get anything back from the deconz GUI when forcing a read from the cluster info (tried to explain that in initial post, sorry if it wasn't clear). I've tried the same thing for the temperature sensor and it seems to work fine.
[image]

ebaauw, 2019-05-30:
Actually, you get something back, the issue is that _Measured Value_ is zero.  You could double-check that the sensor actually reports 0, by running deconz with `--dbg-aps=2` and search the log for something like (substituting the NWK address of your sensor in `srcAddr`):
```
May 30 19:57:35 pi2 deCONZ[885]: 19:57:34:811 APS-DATA.indication srcAddr: 0x1ee9, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0400, lqi: 255, rssi: -47
May 30 19:57:35 pi2 deCONZ[885]: 19:57:34:811     asdu: 08090a0000216217
May 30 19:57:35 pi2 deCONZ[885]: 19:57:34:816 no button map for: SML001 ep: 0x02 cl: 0x0400 cmd: 0x0A pl[0]: 000
```
for the attribute report or
```
May 30 20:01:39 pi2 deCONZ[885]: 20:01:39:149 APS-DATA.indication srcAddr: 0x1ee9, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0400, lqi: 255, rssi: -47
May 30 20:01:41 pi2 deCONZ[885]: 20:01:39:149     asdu: 18750100000021621701000021010002000021feff
```
for the response to the _Read Attributes_.

The `asdu` contains the ZCL payload:
```
0809 0a 0000 21 6217
     |  |    |  + value 0x1762 = 5986.
     |  |    + data type 0x21 = u16
     |  + attribute 0x0000 = measured value
     + command 0x0a = report attributes

1875 01 0000 00 21 6217 0100 00 21 0100 0200 00 21 feff
     |  |    |  |  + value 0x1762 = 5986.
     |  |    |  + data type 0x21 = u16
     |  |    + status 0x00 = ok
     |  + attribute 0x0000 = measured value
     + command 0x01 = read attributes response
```
If you see these messages with value 0x0000, the sensor is indeed reporting 0.  I haven't seen that before.  Other than double-checking the battery, I wouldn't know what to do.

maguiresf, 2019-05-30:
Ok, now I'm really confused. The sensor has started reporting light levels since about an hour ago:
[image]
Even though the temperature sensor has been working since I installed it earlier on in the day:
[image]
Presumably there's no obvious explanation for that? I've got another 3 Hue motions sensors which are currently linked to a Hue bridge. I'll try linking those with the Conbee tomorrow and see if the same thing happens. I know for sure last time I tried linking one of them it didn't seem to report light levels, maybe I just didn't leave it long enough.

wvuyk, 2019-05-31:
I saw this same issue with one of my users. He has been rescanning and re-adding while keeping the device awake and it would not report temperatures in this case. 
He added the device originally using deCONZ 2.05.52. It would not add as this old version did not yet support the utside sensors.  I told him to upgrade for using the (outside)  motion sensors. But then this issue occurred. 
Then I told him to remove the device from deCONZ and then rescan, still the same issue occured, but then after some time all of the sudden reporting started working. No special actions done.

It definitily needed removing after adding to 2.05.52, but indeed it seems to need some time after adding it to the current deCONZ version...

ebaauw, 2019-05-31:
deCONZ checks periodically whether the reporting is working, re-applying the configuration if it isn't.  So reporting "suddenly" starting to work is not that strange (the fact that it didn't work before, obviously, is).

maguiresf, 2019-06-03:
Bit of an update on this, I tried adding another motion sensor to deCONZ over the weekend and this one was even worse. Neither the temperature nor the light levels were being shown. I left this alone for almost 24 hours to see if the issue would resolve without intervention but it didn't. Having a look in the GUI, I could force a read from the temperature sensor and get a value back but when I read the reporting configuration back all 3 values were set to 65535. I manually matched those up with the other sensor I added previously and that seemed to get the temperature reporting working. The light levels were doing this same thing as previously, reading a 0 value even with a forced read. Again with this sensor all three configuration variables were set to 65535. I've manually set those to match the other sensor and restarted the deCONZ process. I'll give it a while longer and see if that makes reporting kick in. Additionally I also had some strange issue where the sensor was holding onto the presence detected state for way to long. The "delay" attribute in the REST API was set to like 122 where as on the other sensor it was 0. I've tried setting that manually to 0 and it does seem to be a little quicker in changing states now. If the light levels don't start reporting after a few hours I'll attempt the debug steps posted earlier.

robertlandes, 2019-06-03:
@maguiresf Which firmware version do you have on the Hue motion sennsor? I've got 8 of them running with Conbee II, öatest firmware / deCONZ version (stable) and they are working perfectly fine incl. light and temperature values. I however had  a similar issue with on of them, but after 1-2h temperature and light level readings where there.

However before i connected them to deCONZ I paired them all to a Hue Bridge to do FW updates. I am running Firmware: 6.1.1.27575 on all of them.

ebaauw, 2019-06-03:
Hadn’t seen that firmware version yet, thanks.  Sure hope we can locate the firmware, so I don’t have to re-pair all my sensors to the Hue bridge to update them, see #270.  I wonder if we’ll be able to get the 0x01 endpoint to work with the new firmware, binding the sensor directly to (a group of) lights.

> The light levels were doing this same thing as previously, reading a 0 value even with a forced read.

That still puzzles me.  Maybe the device disables the hardware light level sensor when no attribute reporting (or supporting binding) has been configured for it?

maguiresf, 2019-06-03:
First one I tried was on 6.1.0.18912, second one was on 6.1.1.27575. I actually thought I'd updated the first one to the latest firmware, guess I'll have to connect it to the Hue bridge for a while. I think I may have added that second one to deCONZ when I first got it on whatever version of firmware it shipped with, noticed the light sensor wasn't working and added to Hue bridge thinking I'd mess with it when I had time. Wonder if that is why the settings were strange, the device was still visible from the web UI, just greyed out. I have another two on 6.1.1.27575 which I need to migrate over to deCONZ that have only ever been attached to the Hue bridge, I'll see what happens with those and update the ticket. Seems like the restart deCONZ thing seems to kick the light sensor into life so at least there's a workaround.

robertlandes, 2019-06-03:
Before connecting, make sure to press the reset button for 10 seconds, so it really resets.

ebaauw, 2019-06-03:
> Wonder if that is why the settings were strange.

The reporting configuration and binding table are lost when factory-resetting the device (which you need to do to re-connect the device to deCONZ, after updating its firmware on the Hue bridge).

> Seems like the restart deCONZ thing seems to kick the light sensor into life so at least there's a workaround.

Restarting deCONZ should trigger the checking and, when needed, re-applying of the reporting configuration.

ebaauw, 2019-06-08:
Updated one of my motion sensors to 6.1.1.27575 through the Hue bridge.  Other than different values for _Date Code_ and, obviously, _SW Build ID_, I don't see any differences.  _PIR Occupied to Unoccupied Delay_ is still not reportable, and creating a binding from the client _OnOff_ cluster on endpoint 0x01 to a light still doesn't do anything.

ebaauw, 2019-06-08:
Updated one of my motion sensors to 6.1.1.27575 through the Hue bridge.  After re-pairing it to deCONZ, I see the following differences:
- Different values for _Date Code_ and, obviously, _SW Build ID_ in the _Basic_ cluster on both endpoints;
- Manufacturer-specific attribute 0x0031 (_Configuration_ ?) is no longer supported on either endpoint;
- Manufacturer-specific attribute 0x0030 (_Sensitivity_ ?) is now supported on endpoint 0x02.
which wasn't supported on the previous firmware.

_PIR Occupied to Unoccupied Delay_ is still not reportable, and creating a binding from the client _OnOff_ cluster on endpoint 0x01 to a light still doesn't do anything.

ebaauw, 2019-06-08:
Writing 0x0030 fails.  The sensor reports back _Status_ 0x88 (read only).

maguiresf, 2019-06-10:
Apologies for the lack of updates on this, I've been having some really annoying issues with my lights/deCONZ network and I've been fighting with them for the last couple of days. Think I've finally gotten to the bottom of it though! So, I added my last two motion sensors a few days ago on the latest firmware and they worked straight away, light levels, motion, temperature all working perfectly so I think perhaps that older firmware was a little dodgy. On a slightly worse note, as part of the issues I was having I ended up taking the VM that deCONZ is running in offline for a while. When I brought it back, three of the motion sensors weren't working at all, no updates, reads were failing from the deCONZ GUI. All the lights, switches and Xiaomi sensors and stuff came back fine, just 3 of the Hue motion sensors, two on the latest firmware and that one on the slightly older one. Left them for a while to see if they came back, they didn't. Reset the one on the older firmware and when it came back all those read settings were at 65535 again which was fairly annoying. There's definitely something a bit iffy between Hue motion sensors on that firmware and deCONZ. I ended up pairing that with the Hue bridge and updating to latest firmware. It added back fine the second time and the other two sensors added back fine too. So, all in all seems my original issue with down to that version of firmware and the latest one seems fine. A little bit concerned about those 3 sensors dropping out when I took deCONZ down for a bit though :(

Luke15153, 2019-06-22:
Hi, thanks all for working this out. 
I have the same issue with this sensor. The reporting configuration shows 65535 only.
Can someone provide the reporting values for the temperature measurement?

Thanks, Luke

ebaauw, 2019-06-22:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/9f1054c74f9c8097e0ead894bc7826629b0a0fa5/bindings.cpp#L930-L938

Pirol62, 2019-09-23:
Hi, I already intalled 3 huemotion sensors. Two of them working great. One sensor provides zero for lightlevel and temperature. Why?

merdok, 2019-10-22:
Hi,

I have the same issue. I have added a Hue motion sensor to my Conbee 2 and it works well except the temperature and light level. After adding the sensor there was no temperature or light level reading. I had to go the GUI and press **Read** on the _Illuminance Measurement_ and _Temperature Measurement_ cluster. This got me the correct values once but they still do not update automatically. I always have to go to the GUI and read them manually... 
Can this be fixed? Is there maybe a fix for that?

lklinggaard, 2019-10-24:
Hi,
I also have the same problems with me outdoor sensor, the sensor provides zero for lightlevel and temperature. I have 3 normal sensor working fine.

maguiresf, 2019-11-06:
The Hue motion sensors are certainly the least stable device types I have. The worst issue I'm having at the moment is that whenever I have to take the server running deconz down for a short while, I always end up with at least one of the motion sensors which completely stops responding. I had to take it down today and I've ended up with 3 out of the 4 of them stuck in a state where the deconz GUI appears to show them not connected to the network, they don't generate events and none of the attributes update. Forcing a read of the cluster attributes times out. I suppose re-pairing them will resolve the issue but this does get a bit tiring having to do it every time. I've had times where they magically seem to come back to life and other times when they haven't. Should I open this as a separate issue? Anyone have any suggestions for anything that might kick them back into life that doesn't involve having to physically reset them? Happy to do anything needed to further debug the issue if required.

Vartkat, 2020-01-22:
Hi,

I'm in quite the same situation. I paired a Philips Hue Motion Sensor to Conbee II, it reports 1 lux in Home Assistant, 0 in Phoscon, and the same as you folks in Deconz 
[image]

I bought an old Hue V1 bridge, reset both and automagically they paired to each other. As the Hue bridge was on the Ethernet Network, a Home Assistant restart discovered automatically the Motion Sensor. Home Assistant tells me it's via the bridge even if I did nothing to integrate the bridge.
With the bridge the light level is coming in Home Assistant. Without verifying I thought the bridge did an update and it would now work with the Conbee II so I unpluged the bridge, reset the Motion sensor and paired it to Conbee II.... Back to square one, lux level is 1 in Home Assistant, 0 in Phoscon and so on.
How the bridge could get a measurement the Conbee cannot ?
I'm confused

scottie-ac, 2020-01-25:
Hi,

same thing here. I´ve just bought the sensor and the motion part is working properly. But I only get an update for light and temperature when I manually trigger the read event for this items via cluster info. It seems that the sensor is not updating the values by its own.

[image]

TuomasPakkanen, 2020-02-15:
I just got the Hue motion sensor yesterday and this has been driving me nuts. I'm running latest version of hass.io with the Deconz add-on on a RBPi4. I've tried to remove and setup the sensor a bunch of times. For me, at first it seems to work at least on the motion detection side, but then after a minute or so it's no longer reporting any values. Also manually reading the configs or attributes fail on Deconz GUI. I do see the green link between the motion sensor and a lamp, which connects to ConBee II. The blue lights also blink on the sensor when I'm initiating a read, but the read's always fail after about a minute. Same with manual writes that I have tried. Any ideas? I think I'd be fine for a while with even the motion sensing working, but can't get even that functioning. Maybe I'll just need to wait a bit and see if it works itself out after some time.

The version of the Hue motion sensor is 6.1.0.18912, so I guess there is a newer one available but unfortunately I already sold my Hue bridge, so I cannot update it. Maybe I need to ask if some colleague etc. could loan their bridge for doing the updating 😃 . Deconz is on the latest firmware from Dec 2019. 

[image]

meerios, 2020-02-24:
> Basically had to set up the correct config values for occupancy, illuminance measurement and I guess I do need to do temperature as well, but did not yet figure out the correct values. Then second step to bind the occupancy sensing cluster to ConBee II Configuration Tool or whatever it is.

...this finally fixed it for me. For temperature I used the Reporting Setting 10, 300, 20 as used by the SML001 internal Hue Motion sensors and wrote it to the sevice and created the binding for Temperature and illuminance.

TuomasPakkanen, 2020-02-25:
> > Basically had to set up the correct config values for occupancy, illuminance measurement and I guess I do need to do temperature as well, but did not yet figure out the correct values. Then second step to bind the occupancy sensing cluster to ConBee II Configuration Tool or whatever it is.
> 
> ...this finally fixed it for me. For temperature I used the Reporting Setting 10, 300, 20 as used by the SML001 internal Hue Motion sensors and wrote it to the device. Afterwards I created the binding for temperature and illuminance.

Thanks. Temperature reporting seemed to work with just the changes I did, so did not have to fiddle with it. Maybe it's enough to do the binding. Maybe a suggestion for Deconz to do these configs automatically in a future firmware version if possible? Was kind of difficult even with technical background. Not sure if it was because of using the Home Assistant Deconz add-on though.

Pedder007, 2020-04-02:
Hi all, as after dozens of attempts to re-pair and re-bind and re-configure my indoor Hue Motion Sensor, I’m more than frustrated that the illumination values are not send, I only see now two options:
- Send him back and get another outdoor Hue Sensor, which works absolutely fine in my setup.
- Buy a Hue bridge, which I normally would absolutely not need, and do a firmware update on the indoor Sensor to Vers.  6.1.1.27575
As by reading all above I‘m not really sure if the update really helps in the end. So, could one of you confirm that this would be the final answer to my problem?
Thanks in advance👍

ebaauw, 2020-04-02:
No need to buy a Hue bridge; you should be able to update the Hue motion sensor firmware through the STD OTAU plugin in deCONZ, see https://github.com/dresden-elektronik/deconz-ota-plugin.  I agree, I don't think it will resolve your issue.

As I mentioned before, if you do receive updates for motion and temperature, most likely the binding and/or attribute reporting for illumination haven't been setup.  You can correct this manually in the GUI.  You only need to do this once; the config will stick until you factory-reset the motion sensor.

Note that the sensor only reports changes in illumination periodically and when it also detects motion.  Just a (sudden) change in light level might take a couple of minutes to be reported.

Pedder007, 2020-04-02:
Thanks for your reply! I missed the aspect to do the update via STD OTAU, that I will test then in addition later on...
With respect to the configuration I already un-binded and binded the illumination cluster, which didn‘t help.
I will give it an additional try as following:
1. do the Update
2. assure the binding
3. reconfigure the attribute reporting

Respective your hint about when the sensors sends the Illumination data: My outdoor sensor continuously sends it, or maybe in short intervals. He doesn’t need a motion detected for that.
As I plan to steer my shutters with the illumi-measurement, I definitely need continuous data for that.
So if it doesn’t work in the end, I will install an outdoor sensor also within the house, for sure without the huge cave, it is originally mounted 😉

ebaauw, 2020-04-02:
deCONZ sets up the sensor to report periodically every 5 minutes, same as the Hue bridge.  That doesn’t need motion.  It also sets up the sensor to report when the illumination changes significantly.  That only works when the sensor detects motion.

This is probably good enough for blind, curtain, and light control, even during a day-time thunderstorm or solar eclipse, but I must admit that I still haven’t tested the latter case.

ebaauw, 2020-04-02:
You need a version of the OTAU plugin from March 14 or later to handle the `.sbl-ota` files.  It should be included in the next version of deCONZ.  If you don’t want to wait, you can compile the plugin yourself (just like the REST API plugin).

Pedder007, 2020-04-03:
Ok thanks again.
Question to installation: is this „std_otau_plugin.cpp“ the right file to install/compile?

ebaauw, 2020-04-03:
It's a separate plugin, don't mix REST API plugin and STD OTAU plugin sources.  You need to clone the entire library into a new directory and compile that.  After compilation, you get a `libstd_otau_plugin.so` which you need to copy to `/usr/share/deCONZ/plugins` (stop deCONZ first or it will crash).

Pedder007, 2020-04-03:
Ahh, so the first step/command here:
https://github.com/dresden-elektronik/deconz-rest-plugin

has to look like this:
 git clone https://github.com/dresden-elektronik/deconz-ota-plugin.git

right?
.... and the rest (steps 2-4) then as described in the mentioned link, resp. you also wrote above.

Sorry I‘m really not a developer or Linux expert, I‘m only trying to tackle these device problems ...

Stopping deCONZ before was clear for me 😉

ebaauw, 2020-04-03:
Yep.  In 2, you need to `cd deconz-ota-plugin`. And in 4, you need to copy `../libstd_otau_plugin.so`.

ebaauw, 2020-04-03:
Did you do the `sudo apt-get deconz-dev`?
And `sudo apt update` followed by `sudo apt install -f`?
I don't quite remember how I setup my Pi for development (a long time ago), but I think that should be enough.

Pedder007, 2020-04-03:
.. oh, good hint, I really missed that.
But I just bootet now in parallel my test PI and will do it first there, wherr for sure I also first had to save my iobroker developement status from the last 2 weeks ...
In additon I also already cloned the SD-Card of my Prod PI and if minor issues occur, also generated a copy of the libstd_otau_plugin.so file ...

Si but now first the test on my test PI.
I will update here soon ...

Pedder007, 2020-04-03:
nope ...

$ sudo apt-get deconz-dev (as you wrote)
returns: not allowed operation

$ sudo apt[-get] install deconz-dev (as given on the deconz-rest-plugin page)
returns:
reading packages
...
reading status
package not findable

Pedder007, 2020-04-03:
...hmhm read a bit further and now asking myself if I might have first to update my deCONZ firmware ...?
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

ebaauw, 2020-04-03:
Are you running deCONZ on this machine?  You need to add the deCONZ `apt` library to your system, as described in [Install deCONZ](https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz).

Pedder007, 2020-04-03:
yes it's all there, it has been a former clone from my prod system
and I'm able to start deCONZ with GUI

Pedder007, 2020-04-04:
OK, i did the ConBee firmware update, which was a bit tricky, but in the end worked.
So I was now able to connect my Xiaomi light sensor, which I still had here for some weeks.
The Hue's are also configured and now let's see how this develops.

The Hue sensor firmware update in the end became to complicated for me. I might spend some more time next week - before maybe sending them back ...

Pedder007, 2020-04-04:
Well, that looks quiet good up to now. All three sensors are still reporting. Up to now in minimum one of the Hues stopped in the morning ...
Maybe the stick firmware update might be the key ...

[image]

Pedder007, 2020-04-05:
Since all sensors are now more than 24h properly reporting, I guess the ConBee II firmware update seems having been the answer to my problems.
One former question now came back into my mind. The Hue **indoor** sensor seems only be able to measure up to 14444lux, whereas the others are showing realistic values with more than 70.000lux in pure sunshine. As the measurement itself seems to be correct (shows equal values for low illumination like the others), this seems to be a mater of the range.
Can this be changed simply by changing the value in the deCONZ GUI in the device parameter settings in ‚Max Measured Value‘? And if yes, what would I have to put in here?

ebaauw, 2020-04-05:
_Max Measured Value_ is read-only.  My SML001 indoor Hue motion sensors report 65534 for that.  Currently, the highest `lightlevel` I'm seeing is 33819, which the REST API plugin translates to 2369 lux.

See https://developers.meethue.com/develop/hue-api/supported-devices/#supportred-sensors, section 2.7 for the relationship between `lightlevel` and lux.

Pedder007, 2020-04-05:
Ok, but I‘m wondering, as mine shows up to 14444lux, also in the Phoscon App, so not only in my Pimatic house automatization.
 Then I guess, that I can not change the range ...

Pedder007, 2020-04-08:
Besides the now bigger issue #2273, now the indoor sensor seems to report in a stable manner. He took himself the longest time to re-connect, after the stick forgot about the entire network, but after some hours he came back by himself and restarts reporting.
So in minium this issue disappered by doing the firmware update, besides now the other occuring obstacles ...

bangcrash, 2020-04-15:
I have set up a Conbee II stick this afternoon, new out of the box and I am having similar problems with a Philips motion sensor as described in this issue.

I removed the sensor from my Hue hub and reset it to set up with Phoscon.  It was detected quickly and also picked up by Home Assistant.  The motion/presence detection is working perfectly but both the lux level and temperature sensors stay at zero.  I have started to look at the Deconz GUI and i have found that I can get a manual read of both sensors but the Min Report Interval, Max Report Interval and Reportable Change values are all 65535.  I have been able to write the correct values to these settings and they are retained if I go back and check again.  Since writing these values the sensors still do not update automatically.

The Philips sensor is on firmware version 6.1.1.27575 and the Conbee II is on version 2.05.75 / 08/03/2020 and firmware 264A0700.  I have tried removing the sensor, resetting it and re-adding it using Phoscon and it comes back in exactly the same state.

If anyone has any suggestions or can point me in the right direction to fix this then I would be really grateful.

Thanks

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

jensflorian, 2020-06-07:
No this issue is not resolved. @Mimiix could you please reopen this issue?

I'm having the same problem with a Phillips SML002 outdoor motion sensor. Motion and temperature works, but illuminance returns only 0. I attempted to bind as suggested in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/340#issuecomment-353648984 but this doesn't work. Philips sensor is on firmware version 6.1.0.25261 and the Conbee II is on version 2.05.77. Restarting Conbee/phoscon or re-adding the sensor doesn't bring the illuminance to life. Any other ideas how to kick that in?

Mimiix, 2020-06-07:
@jensflorian As OP's issue is resolved, this issue is closed. Please open another issue as your issue might be different.