dresden-elektronik/deconz-rest-plugin #1561: Conbee/DeCONZ looses connectivity with devices

ualex73, 2019-06-01:
I got my Conbee II last week and I am using the latest version (2.05.64) with also the latest FW. I got my Innr SP120 and Ikea lights working perfectly. Only after I powered off my pc and power it up 1 hour later, the problems began. At this moment de GUI does *not* show any lines between the Conbee and the devices (even after 1 hour).

Is there any way of getting my network back, without having to re-include the devices again? I know it is just 3 devices, but I won't be happy if this happens when I got 30+ devices.

ualex73, 2019-06-01:
A part of my log, and the "error APSDE-DATA.confirm: 0xD0 on task" seem to be linked to polling the devices?

16:03:07:074 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
16:03:07:074 GW firmware version: 0x26490700
16:03:07:074 GW firmware version is up to date: 0x26490700
16:03:07:273 Announced to internet
16:03:14:779 New websocket 192.168.0.1:64525 (state: 3) 
16:03:18:849 Incr. ZDP retry count 40981697 on item 1
16:03:34:268 0x00158D0003273CD0 error APSDE-DATA.confirm: 0xD0 on task
16:03:45:070 0x00158D0003284147 error APSDE-DATA.confirm: 0xD0 on task
16:03:55:864 0x000D6FFFFE40F7D6 error APSDE-DATA.confirm: 0xD0 on task
16:04:02:076 Current channel 11
16:04:02:084 Device TTL 573 s flags: 0x7
16:04:02:084 Device reset watchdog 7200 s
16:04:06:871 0x00158D0003284147 error APSDE-DATA.confirm: 0xD0 on task
16:04:13:563 scan skip host .200
16:04:16:162 scan finished
16:04:17:868 0x00158D0003284147 error APSDE-DATA.confirm: 0xD0 on task
16:04:17:889 Incr. ZDP retry count 40897010 on item 1

Kane610, 2019-06-01:
Do you have your conbee on an USB extension cord? If not, start there

Smanar, 2019-06-01:
He loose device after a power off.
I already have same with battery powered device but never with bulbs or plugs, even with a 24h off period.

Kane610, 2019-06-01:
I got a similar error APSDE-DATA.confirm: 0xD0 on task before adding an extension cord

ualex73, 2019-06-01:
Yes, it is already on an extension cord.

No, I didn't change anything to my configuration, just power-off, unplug, replug and power-on. The docker container and configuration also didn't change.
@Kane610 What's the reproduction/test scenario with your 0xD0?

Kane610, 2019-06-01:
Well it just didn't work reliably before adding the cord.

Bonus information, my issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/430

alinelena, 2019-06-02:
ok I have the same but strangely enough only for one light ikea one, replaced it with a xioami and seemed to be ok. I have replaced the ikea light builb with a new one, the same issue.
strangely enough this is the closest light to conbee, there are another 24 ikea bulbs in network and none has this issue, 2 sockets and 10+sensors on batteries.

ualex73, 2019-06-03:
@Kane610 - Interestingly the cord solved your issue.
@alinelena - Ok, seems some devices aren't 100% reliable ;-(

I have re-included 1 SP120, but not the second one. The second SP120 still doesn't work (reinclude will get it back), maybe i will increase the logging and see if something Zigbee messages are send or not. Also i will test some power-off/on scenario end of this week, see if it is more stable or not (before i start running 'production' ;-)).

alinelena, 2019-06-03:
thank you. I seem to have fixed my issue by removing the conbee from the computer and re-adding it. So I suspect the secret was rebuilding the network.

Alin

---Dr Alin Marin Elenahttp://alin.elena.space

 ---- On Mon, 03 Jun 2019 09:41:50 +0000 Alexander <notifications@github.com> wrote ----
 > @Kane610 - Interestingly the cord solved your issue.
 > @alinelena - Ok, seems some devices aren't 100% reliable ;-(
 > I have re-included 1 SP120, but not the second one. The second SP120 still doesn't work (reinclude will get it back), maybe i will increase the logging and see if something Zigbee messages are send or not. Also i will test some power-off/on scenario end of this week, see if it is more stable or not (before i start running 'production' ;-)).
 > —
 >

ualex73, 2019-06-17:
The problem went away with re-including all the devices, but i will keep it open for a while longer (I will do some more testing in the couple of weeks)

joch, 2019-09-30:
I’m having the same issue. Suddenly the entire zigbee network just stopped responding after having been solid for months. Now getting lots of these in the log. A few A7, but lots of D0:

```
deconz       | 19:51:28:275 0x000D6F001065708D error APSDE-DATA.confirm: 0xD0 on task
deconz       | 19:51:39:179 0x000B57FFFED965EA error APSDE-DATA.confirm: 0xD0 on task
deconz       | 19:52:03:791 0x086BD7FFFE2F6786 error APSDE-DATA.confirm: 0xA7 on task
```

Any way of debugging this further?

JensTimmerman, 2019-10-09:
same issue here, I can not connect to any zigbee device anymore and my log is filled with:
```
10:33:38:620 Current channel 25
10:33:38:629 Device TTL 2311 s flags: 0x7
10:33:44:357 0x90FD9FFFFE34F0ED error APSDE-DATA.confirm: 0xD0 on task
10:33:55:263 0x90FD9FFFFE34F0ED error APSDE-DATA.confirm: 0xD0 on task
10:34:06:168 0x90FD9FFFFE34F0ED error APSDE-DATA.confirm: 0xD0 on task
10:34:17:063 0x90FD9FFFFE34F0ED error APSDE-DATA.confirm: 0xD0 on task
10:34:27:864 0x90FD9FFFFE34F0ED error APSDE-DATA.confirm: 0xD0 on task
10:34:38:624 Current channel 25
10:34:38:635 Device TTL 2251 s flags: 0x7
10:34:38:769 0x90FD9FFFFE34F0ED error APSDE-DATA.confirm: 0xD0 on task
10:34:49:658 0x90FD9FFFFE34F0ED error APSDE-DATA.confirm: 0xD0 on task
10:35:00:463 0x90FD9FFFFE34F0ED error APSDE-DATA.confirm: 0xD0 on task
10:35:11:367 0x90FD9FFFFE34F0ED error APSDE-DATA.confirm: 0xD0 on task
```
This happened after I powered down the house for an hour to install some shelly's

JensTimmerman, 2019-10-19:
got this again today after a reboot of the machine this usb device is plugged into

r2ixuz, 2020-02-21:
What is wrong with this deconz application? If your host where deconz is running gets suddenly powered off due to an powerloss all your devices needs to be re-added? WTF can anybody fix this pls, thx

Mimiix, 2020-06-05:
@r2ixuz is this still a issue?

mhristache, 2020-06-09:
I have the same issue after a reboot of the host. Please don't tell me I have to re add all the devices. Any other workaround possible?

Mimiix, 2020-06-10:
@mhristache In you case:

- Try to remove the stick and put it back in. See if that works.

Otherwise i'd like you to ask to open a new issue for this as this is a fairly old thread and new updates bring new problems.

mhristache, 2020-06-10:
I opened https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2921

Mimiix, 2020-06-11:
As the OP isn't responding i consider this issue solved!

closed 2020-06-11 by Mimiix