dresden-elektronik/deconz-rest-plugin #1563: Please add Heiman HS1CA-E Carbon Monoxide Detector

robroll, 2019-06-03:
Hello Dresden-Elektronik team,
Please add support for the Heiman HS1CA-E Carbon Monoxide Detector.
Many thanks,
Rob

[image]

ebaauw, 2019-06-07:
What _Manufaturer Code_ does it use?  0x120b?

robroll, 2019-06-07:
Hello. Yes, 0x120b.

[image]

timbru31, 2020-01-10:
I hate to spam issues, but the issue is AFAIK not solved and should remain open, because they do not work correctly.

pattyland, 2020-03-29:
I bought this sensor because is it listed as a supported device, but when I add the sensor through deCONZ (which is displayed as successful) I don't see any new sensor. I'm using version 2.05.75

timbru31, 2020-03-29:
It's supported by deCONZ but not yet by the webapp (Phoscon) - the real API response lists the sensor and it's, e.g., usable with Home Assistant

pattyland, 2020-03-29:
Thanks for the clarification! Since using deCONZ I'm having a hard time time understanding why there are so many different places where devices can show up or not.
I didn't found a project for "Phoscon" at GitHub, should I open a separate issue for Phoscon here or does  "Please add ***" also imply adding it to Phoscon?

timbru31, 2020-03-29:
You could open one here I guess: https://github.com/dresden-elektronik/phoscon-app-beta

Mimiix, 2020-06-05:
@timbru31 , I will get this issue moved to that repo ;)